ducalex/retro-go #5: SD Card Error

32teeth, 2022-03-02:
## SD Card Error
> Mount Error

* Only building out launcher (no emulators)
* ESP-IDF Version 4.1 with applied SD Card Patches

On flashing (to offset 0x1000)

> First Screen

#### SD Card Error
*Mount Failed.*

**OK**

##### Monitor

```
========================================================
launcher 1.31-5-g63e2faee-dirty (Mar  2 2022 09:21:02)
 built for: ODROID-GO. aud=0 disp=0 pad=0 sd=0 cfg=0
========================================================

E (253) sdmmc_io: sdmmc_io_reset: unexpected return: 0x109
[error] rg_storage_init: SD Card mounting failed. driver=1, err=0x109
[warn] rg_storage_init: Failed to load settings from /sd/retro-go/config/retro-go.json.
[info] rg_input_init: Input ready. driver='GPIO'
[info] rg_display_init: Initialization...
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFBB200
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFBBC04
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFBC608
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFBD00C
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFBDA10
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFBE414
[info] ili9341_set_backlight: backlight set to 80%
[info] rg_display_init: Display ready.
[info] rg_alloc: SIZE=12800, CAPS=SPIRAM|8BIT, PTR=3F80001C
[info] rg_gui_set_font_type: Font set to: points=12, scaling=1.00
[info] rg_audio_init: Audio ready. sink='Speaker', samplerate=32000, volume=50
```

> Pressing A button (Next Screen)

#### System Panic!
*Unable to set boot app! App not found*

**OK**

> Again pressing A button (Next Screen)

Quick Hourglass and then previous screen with panic error

##### Monitor

```
========================================================
launcher 1.31-5-g63e2faee-dirty (Mar  2 2022 09:21:02)
 built for: ODROID-GO. aud=0 disp=0 pad=0 sd=0 cfg=0
========================================================

[error] rg_storage_init: SD Card mounting failed. driver=1, err=0x107
[warn] rg_storage_init: Failed to load settings from /sd/retro-go/config/retro-go.json.
[info] rg_input_init: Input ready. driver='GPIO'
[info] rg_display_init: Initialization...
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFBB1A8
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFBBBAC
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFBC5B0
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFBCFB4
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFBD9B8
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFBE3BC
[info] ili9341_set_backlight: backlight set to 80%
[info] rg_display_init: Display ready.
[info] rg_alloc: SIZE=12800, CAPS=SPIRAM|8BIT, PTR=3F80001C
[info] rg_gui_set_font_type: Font set to: points=12, scaling=1.00
[info] rg_audio_init: Audio ready. sink='Speaker', samplerate=32000, volume=50
[info] rg_system_init: Panic log found, saving to sdcard...
```

Thoughts?

ducalex, 2022-03-03:
> Unable to set boot app! App not found

This suggests that your partition table is incorrect. If you didn't use odroid-go-firmware to flash the .fw first, then you'll need to [build and flash your own partition table](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html#generating-binary-partition-table). The expected names and sizes can be found in rg_config.py. 

I'm not sure about the sdcard issue, do you know if retro-esp32 or go-play work on the same device with the same sdcard?

Can you provide the full serial log when booting? 

*As a side note: Thanks for making me realize this issue. Requiring odroid-go-firmware isn't ideal, I'll soon improve rg_tool.py to be able to generate its own partition table and raw image that can be flashed with esptool :).*

32teeth, 2022-03-03:
### Resolved.

Flashed with [Multi Firmware](https://github.com/ducalex/odroid-go-multi-firmware)

> As a side note: Requiring odroid-go-firmware is clearly a problem, I'll soon improve rg_tool.py to be able to generate its own partition table and raw image that can be flashed with esptool :).

This can be a future feature

You can mark as closed

closed 2022-03-03 by 32teeth