ducalex/retro-go #7: [Feature Request] - Theming

32teeth, 2022-03-03:
**Is your feature request related to a problem? Please describe.**
As a contributor / user of retro-go, i would like to create themes

There is some SD card check for theme folder in [gui.c](https://github.com/ducalex/retro-go/blob/master/launcher/main/gui.c#L144-L145)
```
// Try SD card, then search the built-ins
sprintf(path, RG_BASE_PATH "/theme/%s", name);    
```

**Describe the solution you'd like**
Short documentation on how to
* Image naming
* Sub folder naming
* Image size / format

**Describe alternatives you've considered**
Sticking things in the theme folder on the sd card and seeing what sticks

**Additional context**
I would like to add additional themes for users to be able to download and put on their sd card as well as creative assets and files associated to the theme...   ...think 'retro pi themes'

ducalex, 2022-03-04:
The gist is that you can copy the files from git:/launcher/images/ to sdcard:/retro-go/theme/ and it will load from there. I'm not very happy with that method, I would prefer to copy the theme to flash instead. I will write proper documentation only once we've settled on a system.

Retro-Go supports some scaling but I recommend respecting the resolution of the original files. You can use any bit depth. Magenta is used for transparency (note: it is converted to 565 so in your editor anything RGB(>=224, 0, >=224) will be transparent).

Can you make a list of everything that you'd want to be themable in retro-go? (list of colors, images, styles, etc). 

And do you think system/dialog color schemes should be tied to launcher theme or be separate?

32teeth, 2022-03-09:
> Wish List

1. subfolders in theme directory (ex: dark, box, tft, comic...)
2. each subfolder can have a config file with theme colors, x,y ordinal position for images (ex: icons:[0,0], banner:[10,100])
3. when entering gui to go through themes, should pick up # of folders and their corresponding names

I have started somewhat on the base code exploring this

ducalex, 2022-03-10:
> I have started somewhat on the base code exploring this

Alright, I'll step back and let you work on theming then!

32teeth, 2022-08-26:
Closing as this is now part of core

closed 2022-08-26 by 32teeth