ducalex/retro-go #9: atari lynx emulator crashes above version 1.25 (g858da)

smutek110, 2022-03-08:
Any build above version 1.25 (g858da) crashes while loading roms when loading roms in atari lynx emulator. In version 1.25 everything works very well.I am using multi-firmware build: 20210226-7ebc009
I attach the log:

[crash.log](https://github.com/ducalex/retro-go/files/8202662/crash.log)
.

ducalex, 2022-03-09:
That crash.log isn't terribly helpful unfortunately. I suggest you try clearing your settings (In the launcher Menu -> Reset settings, or delete /retro-go/retro-go.json).

If that doesn't help please try [1.31.1](https://github.com/ducalex/retro-go/releases/tag/1.31.1), it has improved logging and might produce a more helpful crash.log!

smutek110, 2022-03-11:
I reset settings ant try 1.31.1 
i using board with 8Mb PSRAM AND 16Mb FLASH
[crash.log](https://github.com/ducalex/retro-go/files/8234342/crash.log)
[image]

smutek110, 2022-03-11:
version 1.25 (g858da) WORKS CORRECTLY

ducalex, 2022-03-12:
Thanks that new log is helpful!

The invalid cart messages could indicate that the ROM isn't read correctly from SD Card, but some lynx games are just weird so I'm not sure this is the problem...

The actual crash is caused by a failed memory allocation of (presumably) 7048 bytes. This should never fail when PSRAM is available, so maybe it isn't initializing correctly? Have you checked the serial output during the esp-idf boot process?

smutek110, 2022-03-13:
The problem is the "ttgo t8 v1.7 "has 8Mb Psram. original wrover in odroid has 4Mb psram. Many applications do not work properly for me, e.g. wolf_3d. But the earlier versions of your application worked fine. Psram is detected, but there is a problem with addressing it. The PSRAM die is marked: "PSRAM64H"
[image]
.

ducalex, 2022-03-13:
Thanks! So it's clear PSRAM is the issue but retro-go autodetects 32/64mbit just fine on other boards (even 16 on the esp32s2). I think the issue is caused by retro-go 1.26+ using esp-idf 4.1 because 1.25 was built with 4.0. It probably needs an extra setting or small tweak to make it work...

Do you have the ability to build retro-go yourself? It would be simpler if you could tinker with the PSRAM config and see if something resolves the issue. Otherwise I can make you a build with verbose logging and `CONFIG_SPIRAM_IGNORE_NOTFOUND=n` and maybe it will tell us why it fails :) .

smutek110, 2022-03-13:
Could you compile? I am currently using esp-idf 4.0 for another project. I use linux because I have compilation problems under windows. I would have to prepare another computer with another linux to work on a clean system, because I don't want to break anything before I finish my espMCUME version.Alternatively, if you do not have time, tell me what to change and I will try, only this will take me some time.

Thanks for giving me your precious time

ducalex, 2022-03-14:
I dug into it a little and the SPIRAM init is partly done by the stage 2 boot loader which is part of the multi-firmware. I remember having to fix similar issues with apps not detecting SPIRAM while working on multi-firmware, but nothing as recent as retro-go. Tomorrow I'll install the version you're using and maybe that way I'll be able to replicate the issue...

ducalex, 2022-03-14:
But in the meantime I'm attaching two retro-go builds with logging set to debug and memory test enabled and all that. One is built with esp-idf 4.0 and the other with 4.1. Please try both to see if there's a difference. They will probably abort before reaching retro-go so please capture the serial output directly!

[retro-go_1.31.1-3-SPIRAM.zip](https://github.com/ducalex/retro-go/files/8241008/retro-go_1.31.1-3-SPIRAM.zip)

smutek110, 2022-03-14:
[IDF_4.1.2.txt](https://github.com/ducalex/retro-go/files/8241083/IDF_4.1.2.txt)

smutek110, 2022-03-14:
[IDF_4.1.2.txt](https://github.com/ducalex/retro-go/files/8241097/IDF_4.1.2.txt)

smutek110, 2022-03-14:
[IDF_4.0.4.txt](https://github.com/ducalex/retro-go/files/8241098/IDF_4.0.4.txt)

smutek110, 2022-03-14:
Just as you predicted. Both have failed

ducalex, 2022-03-15:
Unfortunately those logs are incomplete, I think your serial software or terminal limits scrollback to 800 lines. Can you check if there is an option to increase the buffer size? I really need the whole log from the moment you power the device.

smutek110, 2022-03-16:
Here you have logos, if I do not let him auto-read the rom, I will choose it myself
[log.txt](https://github.com/ducalex/retro-go/files/8258065/log.txt)

smutek110, 2022-03-16:
there are some applications that do not work on this board, although the call has PSRAM

ducalex, 2022-03-16:
That new log is very interesting, it says SPIRAM is working just fine... Before going further I think we need to make sure the file is loaded correctly to memory in the first place.

Please try this build: https://github.com/ducalex/retro-go/actions/runs/1995317348 (It should appear in a few minutes, look at the bottom for "artifacts"). It will print something like:
````
[=] Loading '/sd/roms/lnx/mspacman.lnx'...
[=] -> read ok. size=131136, crc32=FD071DD6
[=] Cart name='MSPACMAN', crc32=7DE3783A, bank0=131072, bank1=65536
[=] EEPROM: Type: none 16bit
````
The first CRC32 represents the whole file. Can you check that it matches your file?

You can use this page to get the crc32 of your rom: https://oss.sheetjs.com/js-crc32/

smutek110, 2022-03-17:
Signed    : 662667970
Unsigned  : 662667970
Hex value : 277f82c2
[log1.txt](https://github.com/ducalex/retro-go/files/8286649/log1.txt)

I downloaded two compilations. The first one named: app_a3688_odroid-go.fw works fine for me. second name: app_a3688_mrgc-g32.fw is seen as invalid firmware.
 
The crc checksums are correct

I do not know what you changed in this bulid (a36883d), but now it does not crash and loads every game correctly

smutek110, 2022-03-17:
Signed    : 662667970
Unsigned  : 662667970
Hex value : 277f82c2
[log1.txt](https://github.com/ducalex/retro-go/files/8286649/log1.txt)

I downloaded two compilations. The first one named: app_a3688_odroid-go.fw works fine for me. second name: app_a3688_mrgc-g32.fw is seen as invalid firmware.
 
The crc checksums are correct

I do not know what you changed in this bulid (a36883d), but now it does not crash and loads every atari lynx game correctly

[crash.log](https://github.com/ducalex/retro-go/files/8287318/crash.log)
But snes games no working .While loading the game there is a message: "Memory init failed"

smutek110, 2022-03-17:
But snes games no working .While loading the game there is a message: "Memory init failed"

[crash.log](https://github.com/ducalex/retro-go/files/8287894/crash.log)

smutek110, 2022-03-17:
[SNEScrash.log](https://github.com/ducalex/retro-go/files/8287917/SNEScrash.log)

ducalex, 2022-03-17:
> I do not know what you changed in this bulid ([a36883d](https://github.com/ducalex/retro-go/commit/a36883d9ab3aba929d8f01ab662a9d3d876a99d8)), but now it does not crash and loads every game correctly

My changes made the loading more robust to errors but I'm not convinced it's what fixed your issue...

Can you try the attached build with lynx games again? It's the exact same version as the one you just tried except this one was built on my PC using my usual build environment. I just want to be sure the problem wasn't my toolchain all along...

[retro-go_1.31.1-7-ga3688_odroid-go.zip](https://github.com/ducalex/retro-go/files/8292938/retro-go_1.31.1-7-ga3688_odroid-go.zip)

smutek110, 2022-03-17:
Everything works fine now. Thank you. Because I can still enjoy your retro go to the fullest

smutek110, 2022-03-17:
I wonder what was the cause?

ducalex, 2022-03-18:
Well a36883d9 fixes loading of some types of ROMs that caused bogus memory allocations. But if that is the fix, then I don't know how it could have worked in 1.25... Anyway I'm glad everything works now! 

I wasn't able to replicate the snes issue but I have ideas and will make a new release soon. Please wait for that and then you could open a new issue if the problem persists :) .

smutek110, 2022-03-21:
NOW THE SNES WORKS WELL. DOES NOT WORK ONLY IN app_a3688_odroid-go.fw THAT YOU HAVE GIVEN ME TO CHECK
If it's the same compilation, it must have depend on the environment you compiled in. Compare files under hexedit.

closed 2022-06-07 by ducalex