ducalex/retro-go #11: No sound on SNES emulator in Retro-Go 1.31.1, but sound works on the other emulators

ryan-makes-it, 2022-03-15:
I just received a MRGC G32 ESP32 Retro Gaming Console, and installed Retro-Go 1.31.1 for the SNES emulator.  However, I have no sound from the several SNES games I have tried (F-Zero, Castlevania - Dracula X, Super Mario World, etc).  The NES emulator plays sound just fine.

I have tried changing the "Audio out" option from "Ext DAC" to "Dummy", but that doesn't seem to be helping.  

I'm happy to share debugging logs or whatever else you need - thanks for creating this lovely software package!

smutek110, 2022-03-15:
Snes doesn't have sound support yet. 
Check in the description.

ryan-makes-it, 2022-03-15:
Ah, my apologies, I missed that - thank you!

closed 2022-03-15 by ryan-makes-it

ducalex, 2022-03-15:
I will add a popup to say "SNES is very slow and has no sound" when using it for the first time. I understand that not everybody reads the README :) .

ryan-makes-it, 2022-03-16:
Yea, I apparently stoped at the "Nintendo: NES, Gameboy, Gameboy Color, SNES (very slow)" in the README.md :-/ 

The SNES emulation did not feel slow to me - at least not in F-Zero, which is the game I've played the most.  Maybe in a more demanding game it would feel slower?

In any case, thanks for updating me so quickly about this, and thanks for putting this together.  Now I'm going to be hauling a Game Boy-ish device everywhere with me again...