ducalex/retro-go #12: how do i apply patch

ll3macorn, 2022-04-08:
i cant figure out how to run the sd card patch

ducalex, 2022-04-08:
You need open a terminal in your esp-idf installation ($IDF_PATH) and run something like:
- `patch -p1 < "/path/to/retro-go/tools/patches/panic-hook (esp-idf 4.0 and 4.1).diff"

You can also do it manually. Open `components/driver/sdspi_host.c` from your esp-idf installation then apply each changes.

For example if the patch says:
````
     esp_err_t ret = ESP_OK;
+    spi_device_acquire_bus(s_slots[slot].handle, portMAX_DELAY);
     cs_low(slot);
````

Then you'd locate:
````
    esp_err_t ret = ESP_OK;
    cs_low(slot);
````

And insert the missing line to get:
````
    esp_err_t ret = ESP_OK;
    spi_device_acquire_bus(s_slots[slot].handle, portMAX_DELAY);
    cs_low(slot);
````

ll3macorn, 2022-04-09:
im having this issue now

`@mechbook retro-go % ./rg_tool.py build-fw                      
=== Step: Building ===

Building app 'launcher'
Executing action: app
Running ninja in directory /Users/baileyboylson/Downloads/retro-go/launcher/build
Executing "ninja app"...
[45/112] Building C object esp-idf/ret...es/__idf_retro-go.dir/rg_display.c.obj
FAILED: esp-idf/retro-go/CMakeFiles/__idf_retro-go.dir/rg_display.c.obj 
/Users/baileyboylson/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -I/Users/baileyboylson/Downloads/retro-go/launcher/build/config -I/Users/baileyboylson/Downloads/retro-go/components/retro-go -I/Users/baileyboylson/esp/esp-idf/components/newlib/platform_include -I/Users/baileyboylson/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/baileyboylson/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/Users/baileyboylson/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/baileyboylson/esp/esp-idf/components/freertos/esp_additions/include -I/Users/baileyboylson/esp/esp-idf/components/esp_hw_support/include -I/Users/baileyboylson/esp/esp-idf/components/esp_hw_support/include/soc -I/Users/baileyboylson/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/Users/baileyboylson/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/Users/baileyboylson/esp/esp-idf/components/esp_hw_support/port/esp32/private_include -I/Users/baileyboylson/esp/esp-idf/components/heap/include -I/Users/baileyboylson/esp/esp-idf/components/log/include -I/Users/baileyboylson/esp/esp-idf/components/lwip/include/apps -I/Users/baileyboylson/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/baileyboylson/esp/esp-idf/components/lwip/lwip/src/include -I/Users/baileyboylson/esp/esp-idf/components/lwip/port/esp32/include -I/Users/baileyboylson/esp/esp-idf/components/lwip/port/esp32/include/arch -I/Users/baileyboylson/esp/esp-idf/components/soc/include -I/Users/baileyboylson/esp/esp-idf/components/soc/esp32/. -I/Users/baileyboylson/esp/esp-idf/components/soc/esp32/include -I/Users/baileyboylson/esp/esp-idf/components/hal/esp32/include -I/Users/baileyboylson/esp/esp-idf/components/hal/include -I/Users/baileyboylson/esp/esp-idf/components/hal/platform_port/include -I/Users/baileyboylson/esp/esp-idf/components/esp_rom/include -I/Users/baileyboylson/esp/esp-idf/components/esp_rom/include/esp32 -I/Users/baileyboylson/esp/esp-idf/components/esp_rom/esp32 -I/Users/baileyboylson/esp/esp-idf/components/esp_common/include -I/Users/baileyboylson/esp/esp-idf/components/esp_system/include -I/Users/baileyboylson/esp/esp-idf/components/esp_system/port/soc -I/Users/baileyboylson/esp/esp-idf/components/esp_system/port/include/private -I/Users/baileyboylson/esp/esp-idf/components/xtensa/include -I/Users/baileyboylson/esp/esp-idf/components/xtensa/esp32/include -I/Users/baileyboylson/esp/esp-idf/components/vfs/include -I/Users/baileyboylson/esp/esp-idf/components/esp_wifi/include -I/Users/baileyboylson/esp/esp-idf/components/esp_event/include -I/Users/baileyboylson/esp/esp-idf/components/esp_netif/include -I/Users/baileyboylson/esp/esp-idf/components/esp_eth/include -I/Users/baileyboylson/esp/esp-idf/components/esp_phy/include -I/Users/baileyboylson/esp/esp-idf/components/esp_phy/esp32/include -I/Users/baileyboylson/esp/esp-idf/components/esp_timer/include -I/Users/baileyboylson/esp/esp-idf/components/spi_flash/include -I/Users/baileyboylson/esp/esp-idf/components/fatfs/diskio -I/Users/baileyboylson/esp/esp-idf/components/fatfs/vfs -I/Users/baileyboylson/esp/esp-idf/components/fatfs/src -I/Users/baileyboylson/esp/esp-idf/components/wear_levelling/include -I/Users/baileyboylson/esp/esp-idf/components/sdmmc/include -I/Users/baileyboylson/esp/esp-idf/components/driver/include -I/Users/baileyboylson/esp/esp-idf/components/driver/deprecated -I/Users/baileyboylson/esp/esp-idf/components/driver/esp32/include -I/Users/baileyboylson/esp/esp-idf/components/esp_pm/include -I/Users/baileyboylson/esp/esp-idf/components/esp_ringbuf/include -I/Users/baileyboylson/esp/esp-idf/components/app_update/include -I/Users/baileyboylson/esp/esp-idf/components/bootloader_support/include -I/Users/baileyboylson/esp/esp-idf/components/bootloader_support/bootloader_flash/include -I/Users/baileyboylson/esp/esp-idf/components/esp_adc_cal/include -I/Users/baileyboylson/esp/esp-idf/components/json/cJSON -mlongcalls -Wno-frame-address  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -fmacro-prefix-map=/Users/baileyboylson/Downloads/retro-go/launcher=. -fmacro-prefix-map=/Users/baileyboylson/esp/esp-idf=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0-dev-2393-gfaed2a44aa\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -O2 -DRG_REPLACE_PRINTF_FUNCTIONS -DRG_TARGET_ODROID_GO -MD -MT esp-idf/retro-go/CMakeFiles/__idf_retro-go.dir/rg_display.c.obj -MF esp-idf/retro-go/CMakeFiles/__idf_retro-go.dir/rg_display.c.obj.d -o esp-idf/retro-go/CMakeFiles/__idf_retro-go.dir/rg_display.c.obj -c /Users/baileyboylson/Downloads/retro-go/components/retro-go/rg_display.c
/Users/baileyboylson/Downloads/retro-go/components/retro-go/rg_display.c: In function 'spi_init':
[... cut ...]
  File "/Users/baileyboylson/Downloads/retro-go/./rg_tool.py", line 316, in <module>
    build_app(app, args.target, command == "profile", args.with_netplay)
  File "/Users/baileyboylson/Downloads/retro-go/./rg_tool.py", line 206, in build_app
    subprocess.run("idf.py app", shell=True, check=True, cwd=os.path.join(os.getcwd(), target))
  File "/Users/baileyboylson/.pyenv/versions/3.9.6/lib/python3.9/subprocess.py", line 528, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'idf.py app' returned non-zero exit status 2.
`

ducalex, 2022-04-09:
Which version of esp-idf are you using?

ll3macorn, 2022-04-09:
the latest

closed 2024-05-03 by ducalex