ducalex/retro-go #14: Will there be support for seperate SPI bus for the sd card and spi screen?

Wallnutts, 2022-06-07:
Both odroid go and mrgc seem to share the same spi bus for sd card and spi screen. Would be great to have seperate spi buses for both to support generic esp32 platforms with integrated sd card better as they use different sd card pins. Thanks

ducalex, 2022-06-07:
You can change the SPI pins in the `target/<device>.h` for both the display and the SD Card however there's only one free SPI peripheral unfortunately (I *think* if we slow down the PSRAM then SPI3 becomes available too).

The MRGC uses SDMMC instead of SPI and I don't think those gpios can be changed. 

It's unfortunate that esp-idf doesn't include software SPI; the LCD could have the full hardware bus and the SD could use a software bus.

Wallnutts, 2022-06-07:
Im using a TTGO T8 to try and make a custom ODROID-GO where it has an inbuilt sd card at specific pins. I managed to change some pins and get a SD Card detected through sdmmc but am unable to write to it. Any idea what solutions there might be?

Output from rg_tool.py --port=COM8 monitor:
[image]

Pin Definitions: ( I had to disable the I2C audio pins as there wasnt enough GPIO ) ( target/odroid-go.h)
[image]

SDMMC pin definitions: ( components/rg_storage.c )
[image]

ducalex, 2022-06-07:
If I'm reading [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sdmmc_host.html) correctly then GPIO 14 is used as clock?

If that is the case then RG_GPIO_LCD_BCKL will interfer, you can try changing RG_GPIO_LCD_BCKL to something else.

Otherwise you can try changing SDMMC_FREQ_HIGHSPEED for SDMMC_FREQ_PROBING and see if that works (albeit slowly).

Wallnutts, 2022-06-07:
Managed to reassign GPIO 14 and it works now! Its able to write and read the retro-go/config/retro-go.json file. However, the screen seems to be stuck on the hourglass picture and this is all the monitor logs end at.

[image]

ducalex, 2022-06-07:
After the card is unmounted, retro-go will wait for all buttons to be released before rebooting. It's possible one of your button registers all the time? It basically wants `rg_input_read_gamepad()` to be 0.

This all happens in `shutdown_cleanup()` (rg_system.c) and you can comment or remove any step, nothing in there is actually essential!

Wallnutts, 2022-06-07:
thanks for ur help so far! I have been prototyping on a breadboard and didnt realise that the #define RG_GPIO_GAMEPAD_X          #define RG_GPIO_GAMEPAD_Y were not connected to 10Kohm pullups which was causing it to be triggered. 

Now i have managed to get past rg_system_init() but seem to be stuck with these looping outputs and my screen still shows the hourglass image :
[image]

ducalex, 2022-06-07:
That's a good sign, it means retro-go is running :). That line is shown once per second.

I'm not sure why the launcher isn't drawing the interface. I think it could be further input problems. Check your MENU and OPTION button.

Wallnutts, 2022-06-08:
Been testing different pin configs and made sure none of them interfered but still running into the same issue unfortunately

ducalex, 2022-06-10:
One thing you could try to know for sure if input is the problem or not is:
1. open `rg_input.c` and locate the `rg_input_read_gamepad` function
2. edit it to always return 0 (at the time of writing it means changing line 280 to become `return 0;`)

If retro-go does boot then you could add a printf("gamepad_state = %04X\n", gamepad_state); in that same function to find out which button is the problem.

If retro-go still doesn't boot then it's most likely a bug and I'd be happy to help you track it down. I would start by adding `printf("I'm about to do XYZ\n");` statement between every step in the launcher's app_main() function to see how far it goes.

Wallnutts, 2022-06-11:
I added those parts of the code in and managed to track down which input pins were having issues. All the buttons seem to be working fine now where retro-go boots up properly and is able to run some games i put on. Much appreciated for ur help!

I think the print outputs would be a useful addon to check button inputs especially for others looking to make custom odroid-go like me! Ill post my pin configs here later for future reference. 

I havent tested the audio yet though it seems like setting #define RG_AUDIO_USE_SPEAKER 1 sets GPIO25 and and GPIO26 as L and R audio channels right?

Wallnutts, 2022-06-12:
This is my full pin config for all the buttons, seperate sdmmc slot and lcd display and working audio:

```
// Target definition
#define RG_TARGET_NAME             "ODROID-GO"

// Storage and Settings
#define RG_STORAGE_DRIVER           2   // 1 = SDSPI, 2 = SDMMC, 3 = USB
#define RG_SETTINGS_USE_NVS         0

// Audio
#define RG_AUDIO_USE_SPEAKER        1
//#define RG_AUDIO_USE_EXT_DAC        0

// Video
#define RG_SCREEN_DRIVER            0   // 0 = ILI9341
#define RG_SCREEN_TYPE              0
#define RG_SCREEN_WIDTH             320
#define RG_SCREEN_HEIGHT            240
#define RG_SCREEN_ROTATE            0
#define RG_SCREEN_MARGIN_TOP        0
#define RG_SCREEN_MARGIN_BOTTOM     0
#define RG_SCREEN_MARGIN_LEFT       0
#define RG_SCREEN_MARGIN_RIGHT      0

// Input
#define RG_GAMEPAD_DRIVER           1   // 1 = ODROID-GO, 2 = Serial, 3 = MRGC-IO
#define RG_BATT_ADC_CHANNEL         ADC1_CHANNEL_0
#define RG_BATT_MULTIPLIER          2.0f
#define RG_BATT_VOLT_MIN            3.5f
#define RG_BATT_VOLT_MAX            4.2f

// Status LED
//#define RG_GPIO_LED                 GPIO_NUM_12

// I2C BUS
#define RG_GPIO_I2C_SDA             GPIO_NUM_15
#define RG_GPIO_I2C_SCL             GPIO_NUM_4

// Built-in gamepad
#define RG_GPIO_GAMEPAD_X           ADC1_CHANNEL_6
#define RG_GPIO_GAMEPAD_Y           ADC1_CHANNEL_7
#define RG_GPIO_GAMEPAD_SELECT      GPIO_NUM_12
#define RG_GPIO_GAMEPAD_START       GPIO_NUM_39
#define RG_GPIO_GAMEPAD_A           GPIO_NUM_32
#define RG_GPIO_GAMEPAD_B           GPIO_NUM_33
#define RG_GPIO_GAMEPAD_MENU        GPIO_NUM_27
#define RG_GPIO_GAMEPAD_OPTION      GPIO_NUM_0

// SNES-style gamepad
#define RG_GPIO_GAMEPAD_LATCH       GPIO_NUM_NC
#define RG_GPIO_GAMEPAD_CLOCK       GPIO_NUM_NC
#define RG_GPIO_GAMEPAD_DATA        GPIO_NUM_NC

// SPI Display
#define RG_GPIO_LCD_MISO            GPIO_NUM_NC
#define RG_GPIO_LCD_MOSI            GPIO_NUM_23
#define RG_GPIO_LCD_CLK             GPIO_NUM_18
#define RG_GPIO_LCD_CS              GPIO_NUM_5
#define RG_GPIO_LCD_DC              GPIO_NUM_21
#define RG_GPIO_LCD_BCKL            GPIO_NUM_22

// SPI SD Card
//#define RG_GPIO_SD_MISO             GPIO_NUM_19
//#define RG_GPIO_SD_MOSI             GPIO_NUM_23
///#define RG_GPIO_SD_CLK             GPIO_NUM_18
//#define RG_GPIO_SD_CS               GPIO_NUM_22

// External I2S DAC
#define RG_GPIO_SND_I2S_BCK         GPIO_NUM_NC
#define RG_GPIO_SND_I2S_WS          GPIO_NUM_NC
#define RG_GPIO_SND_I2S_DATA        GPIO_NUM_NC
#define RG_GPIO_SND_AMP_ENABLE      GPIO_NUM_NC
```

closed 2022-06-12 by Wallnutts

ducalex, 2022-06-12:
I've done many changes to help prevent someone else from having the same issue:
- The button state is printed during driver init (normally it should be all 0s, if not then there is an issue)
- Dialogs are drawn before they wait for all key to be released (that way someone would likely suspect an input issue when they're unresponsive)
- Recovery will print a message when it detects a key being held
- Improved misc boot logging

(The only place I can't remove the key loop completely is the reboot process, because certain GPIOs used by buttons are strapping pins for the bootloader and we must ensure they're released).

> I havent tested the audio yet though it seems like setting #define RG_AUDIO_USE_SPEAKER 1 sets GPIO25 and and GPIO26 as L and R audio channels right?

The audio using the built-in DAC is actually in mono, both pins go to the speaker. (This was done to increase resolution).