ducalex/retro-go #15: Self-compiled retro-go crashes constantly with "ROM Loading Failed!"

DisableGraphics, 2022-06-18:
A bit of backstory: my odroid go has a flash error and cannot load emulators bigger than 1 mb. So I decided to compile retro-go after not having used the odroid since september. But, everytime I boot up the odroid (even with B) will show an error message with the text "ROM Loading Failed!" in large friendly letters in the cover.

This is the crash.log file: 
```
Application: gnuboy-go
Version: 1.32.1-37-gdb1d2ae3-dirty
Build date: Jun 18 2022 12:44:15
ESP-IDF: v4.1.3-46-g5ef3614e5f-dirty
Total memory: 267868 + 4194260
Free memory: 199784 + 4181460
Free block: 113804 + 4181460
Stack HWM: 5788
Uptime: 0s (0 ticks)
Panic message: ROM Loading failed!
Panic context: app_main

Log output:
[info] rg_system_init: Welcome! Reset reason: 7
[info] rg_storage_init: SD Card mounted at /sd. driver=1
[info] rg_settings_init: Settings loaded from /sd/retro-go/config/retro-go.json.
[info] rg_input_init: Input ready. driver='GPIO', state=00000000 00000000
[info] rg_display_init: Initialization...
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFCA8F0
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFCB2F4
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFCBCF8
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFCC6FC
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFCD100
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFCDB04
[info] ili9341_set_backlight: backlight set to 80%
[info] rg_display_init: Display ready.
[info] rg_alloc: SIZE=12800, CAPS=SPIRAM|8BIT, PTR=3F80001C
[info] rg_gui_set_font_type: Font set to: points=12, scaling=1.00
[info] rg_audio_init: Audio ready. sink='Speaker', samplerate=32000, volume=50
[info] rtc_time_init: Time is now: Sat, 01 Jan 2000 05:00:00
[info] rtc_time_init: Time loaded from hardcoded
[info] rg_system_init: Retro-Go ready.

[info] rg_alloc: SIZE=46080, CAPS=8BIT, PTR=3FFD0844
[info] rg_alloc: SIZE=46080, CAPS=8BIT, PTR=3FFE436C
[=]  * gnuboy_load_rom: Loading file: ''
[=] !! gnuboy_load_rom: ROM fopen failed

*** PANIC TRACE: ***

End of trace
```
Looking at it, we can see that there's a line that most surely causes the error:

```
[=]  * gnuboy_load_rom: Loading file: ''
```
The thing is, in the retro-go.json configuration, the line 
```
"RomFilePath":	"/sd/roms/gbc/crystal-clear-2.0-baby.gbc",
```
Is present. I've noted that 
```
"StartAction":	0,
```
is declared just before the RomFilePath line, so probably it has something to do with it.

DisableGraphics, 2022-06-18:
Update: If I press ok, it panics and shows a message saying: No boot app found!
And the crash.log changes to this:
```
Application: gnuboy-go
Version: 1.32.1-37-gdb1d2ae3-dirty
Build date: Jun 18 2022 12:44:15
ESP-IDF: v4.1.3-46-g5ef3614e5f-dirty
Total memory: 267868 + 4194260
Free memory: 199784 + 4181460
Free block: 113804 + 4181460
Stack HWM: 5788
Uptime: 0s (0 ticks)
Panic message: Unable to set boot app: App not found!
Panic context: rg_system_set_boot_app

Log output:
[info] rg_system_init: Welcome! Reset reason: 4
[info] rg_storage_init: SD Card mounted at /sd. driver=1
[info] rg_settings_init: Settings loaded from /sd/retro-go/config/retro-go.json.
[info] rg_input_init: Input ready. driver='GPIO', state=00000000 00000000
[info] rg_display_init: Initialization...
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFCA8F0
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFCB2F4
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFCBCF8
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFCC6FC
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFCD100
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FFCDB04
[info] ili9341_set_backlight: backlight set to 80%
[info] rg_display_init: Display ready.
[info] rg_alloc: SIZE=12800, CAPS=SPIRAM|8BIT, PTR=3F80001C
[info] rg_gui_set_font_type: Font set to: points=12, scaling=1.00
[info] rg_audio_init: Audio ready. sink='Speaker', samplerate=32000, volume=50
[info] rg_system_init: Panic log found, saving to sdcard...
[info] rg_system_save_trace: Saving debug trace to '/sd/crash.log'...
[warn] rg_system_init: Aborting: panic!

*** PANIC TRACE: ***

End of trace
```

DisableGraphics, 2022-06-18:
Nevermind, I forgot to add 'launcher' as an option to this build command:
```
./rg_tool.py build-fw gnuboy-go launcher
```

DisableGraphics, 2022-06-18:
Also, the output binary is only 600 kb in size. Great work with this project!

closed 2022-06-18 by DisableGraphics

ducalex, 2022-06-18:
> Nevermind, I forgot to add 'launcher' as an option to this build command:

I agree the instructions aren't clear about that (and even then, it should be the job of the build script). I've updated the build script to automatically include the launcher so that nobody else gets stuck on that same problem!