ducalex/retro-go #16: Flashing into generic ESP32 board

IndiePix, 2022-07-03:
**Board model**
TTGO ESP32 WROVER-B (4MB)
[image]

**Process**
1. Build an image directly  (the 4MB flash wouldn't allow to have the goplay bootloader and install fw from there) with:
`./rg_tool.py build-img`

2. Flashed into the generic ESP32 board succesfully after erasing all its flash previously with:
`esptool.py -b 921600 write_flash 0x0 retro-go.img`

Flashing logs:
```
esptool.py v2.9-dev
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting.......
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 84:0d:8e:e6:c1:b0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 393216 bytes to 239105...
Wrote 393216 bytes (239105 compressed) at 0x00000000 in 21.8 seconds (effective 144.0 kbit/s)...
Hash of data verified.

```
**Problem**
Program is not booting into the esp32. The screen shows blank and no feedback.
From monitoring the esp32 boot we found this error log:
```
I (184) boot: Loaded app from partition at offset 0x10000
I (185) boot: Disabling RNG early entropy source...
E (197) spi_flash: Detected size(4096k) smaller than the size in the binary image header(16384k). Probe failed.
abort() was called at PC 0x40080f78 on core 0
0x40080f78: start_cpu0_default at /home/esp/esp-idf/components/esp32/cpu_start.c:485 (discriminator 15)
```

**Additional info**
We tried to flash smaller builds with only the launcher and emulators individually to see if the 4MB was not enough for the full retro-go build but the problem persists the same.

ducalex, 2022-07-03:
> E (197) spi_flash: Detected size(4096k) smaller than the size in the binary image header(16384k). Probe failed.

Can you try `esptool.py -b 921600 write_flash --flash_size detect 0x0 retro-go.img`? It's possible you'll have to add `--flash_freq 40M` as well, I'm not sure.

If that fails you can try editing `base.sdkconfig` but that's not very convenient:
- change `CONFIG_ESPTOOLPY_FLASHSIZE_4MB=n` to `CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y`
- change `CONFIG_ESPTOOLPY_FLASHSIZE_16MB=y` to `CONFIG_ESPTOOLPY_FLASHSIZE_16MB=n`
- CONFIG_ESPTOOLPY_FLASHSIZE="16MB" to CONFIG_ESPTOOLPY_FLASHSIZE="4MB"
- You'll need to do a clean build after that.

>  We tried to flash smaller builds with only the launcher and emulators individually to see if the 4MB was not enough for the full retro-go build but the problem persists the same.

retro-go no longer fits in 4MB with the new genesis emulator (It almost does, I'm working on it :)). You will need to disable at least one big app (snes9x-go or gwenesis-go or prboom-go).

IndiePix, 2022-07-07:
**Problem is solved.** 
I did all those things previously without realizing I wasn't cleaning the build each time.

**Problem apparently was:**
ESPTOOL (at least in my setup) seems to overwrite the detected flashsize with the given in base.sdkconfig manually. It may be inconvenient to modify the flashsize manually each time but in the end it works.

ducalex, 2022-08-24:
Thanks for sharing the resolution with us!

closed 2022-08-24 by ducalex