ducalex/retro-go #18: Sega Megadrive/Genesis not available?

Trithilon, 2022-07-14:
Can't seem to find Megadrive/Genesis emulator in the menu.
Using Retro Go 1.32.1 on an ODroid Go.
I am sure this is something silly or I am missing something.

jdalmanza, 2022-07-14:
the odroid go cpu is not powerful enough to run the genesis/megadrive emulators. 8-bit consoles are the limit here, with the pc engine/tg-16 being the upper end (technically an 8-bit cpu)

ducalex, 2022-07-14:
Genesis was added after the last release and it's not really ready but you can go to [actions](https://github.com/ducalex/retro-go/actions) > top green item > artifacts to get the most up to date build. For better speed you can try disabling Z80 or YFM in the menu, at the cost of some audio.

Trithilon, 2022-07-15:
I was able to play Genesis/MD roms. The one I really wanted to play was Sokoban. :)
Thank you! Closing this issue.

closed 2022-07-15 by Trithilon