ducalex/retro-go #21: Understanding how to put a retro go project together

stardestro, 2022-08-22:
Hi,

I am less familiar with ESP32 development and want to build your project. I have been able to flash a couple projects that use a ESP8266 either through my raspberry pi or using Visual Studio Code's PlatformIO IDE extension with success. SlimeVR-Tracker-ESP and ChickenManGame. Those projects either had robust documentation for hardware/software or had Youtube videos that showed a lot of the process and can be deduced for a complete understanding. I've been intrigued by some random Youtube videos and hackaday posts that use the ESP32 to run games such as Doom and Mario but a lot of the information is either scattered, missing, or inconsistent. Also I am interested in this project since it uses a screen which would be rather new for me. 

I guess my questions can be summed as follows:
1. I understand that building and flashing is governed by your a python script is there anything I need to install in order for that to run properly? (Sorry I mainly have experience in C/C++, and no serial knowledge)

2. Where can I find information on how to wire up the hardware to get this working? (I have a ESP32-WROVER-E and ILI9341(240x320) screen that has a built in SD card reader)

ducalex, 2022-08-22:
1. The only prerequisite is having a functional [esp-idf environment](https://docs.espressif.com/projects/esp-idf/en/v4.1.3/get-started/index.html), it already contains the python dependencies.

2. You can follow the [ODROID-GO schematics](https://github.com/hardkernel/ODROID-GO/blob/master/Documents/ODROID-GO_REV0.1_20180518.pdf) and you can look at the `components/retro-go/targets/odroid-go.h` file where most pins are defined.

It might be beneficial for you to learn a bit about [building esp-idf projects](https://docs.espressif.com/projects/esp-idf/en/v4.1.3/get-started/index.html#get-started-build) first. For example you can build and flash the launcher without using rg_tool.py at all and focus on that until it works!

ducalex, 2024-05-03:
Issue is almost 2 years old. Hopefully you've resolved your issue by now (or moved on). If not, feel free to re-open!

closed 2024-05-03 by ducalex