ducalex/retro-go #22: Can it compiled on idf 4.4.2？

FantasyGmm, 2022-08-23:
I made a hardware for ESPlay (based on ESP32S3), I ported two ESPlay emulators and it can run normally, I would like to ask if it can be compiled under IDF4.4.2 now, this is the first version of S3 released

I might be able to port it to my hardware if I can compile it

I used the same DAC and audio amplifier as EsPlay, as well as the extended IO chip

The ESP32S3 has more performance and more memory, and I believe it will be better on the emulator than the old 32, and it will work with both SDIO and I2S without DMA errors

FantasyGmm, 2022-08-23:
ODROID-GO Launcher is available now

ducalex, 2022-08-23:
If you want to use .fw format then you'll have to also port [odroid-go-multi-firmware](https://github.com/ducalex/odroid-go-multi-firmware) to your device.

Otherwise use the [img format](https://github.com/ducalex/retro-go#generic-esp32). The first time it must be flashed with esptool and after that rg_tool.py flash will work.

BTW most of the speed gain on the S3 is because of faster PSRAM so make sure to edit base.sdkconfig to get it running at 120Mhz.

FantasyGmm, 2022-08-23:
Thanks, I have successfully logged into FW, I can see all the emulators, but when I start the game sometimes it starts normally, and most of the time it crashes
As for S3 PSRAM, the 120M option is not available in 4.4.2, so it can only work with Flash at 80M

FantasyGmm, 2022-08-23:
When I tried to launch the game it show: Aborting: panic!
```
========================================================
nofrendo-go b8eb9d9-dirty (Aug 24 2022 01:50:04)
 built for: ODROID-GO. aud=0 disp=0 pad=0 sd=0 cfg=0
========================================================

[info] rg_system_init: Welcome! Reset reason: 4
[info] rg_storage_init: SD Card mounted at /sd. driver=2
[info] rg_settings_init: Settings loaded from /sd/retro-go/config/retro-go.json.
[info] rg_i2c_init: I2C driver ready (SDA:10 SCL:11).
[info] rg_input_init: Input ready. driver='ESPLAY-I2C', state=00000000 00000000
[info] rg_display_init: Initialization...
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FCAC10C
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FCACB10
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FCAD514
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FCADF18
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FCAE91C
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FCAF320
[info] lcd_set_backlight: backlight set to 80.00%
[info] rg_display_init: Display ready.
[info] rg_alloc: SIZE=12800, CAPS=SPIRAM|8BIT, PTR=3D8025E8
[info] rg_gui_set_font_type: Font set to: points=12, scaling=1.00
[info] rg_audio_init: Audio ready. sink='Ext DAC', samplerate=32000, volume=50
[warn] rg_system_init: Aborting: panic!
[info] rg_gui_dialog: text_buffer usage = 93
[info] rg_system_set_boot_app: Boot partition set to 16 'launcher'
[info] rg_audio_deinit: Audio terminated. sink='Ext DAC'
[info] rtc_time_save: System time saved to storage.
[info] rg_storage_deinit: SD Card unmounted.
[info] rg_input_deinit: Input terminated.
[info] rg_i2c_deinit: I2C driver terminated.
[info] rg_display_deinit: Display terminated.
```

ducalex, 2022-08-23:
>  As for S3 PSRAM, the 120M option is not available in 4.4.2, so it can only work with Flash at 80M

I didn't know that, that's a bummer :(

ducalex, 2022-08-23:
> When I tried to launch the game it show: Aborting: panic!

That log is what happens AFTER a crash, you should look at what happens just before.

FantasyGmm, 2022-08-23:
When I comment rgsytem-panic judge, I can see the error message. Most emulators crash may  have a null rom path
```
[info] lcd_set_backlight: backlight set to 80.00%
[info] rg_display_init: Display ready.
[info] rg_alloc: SIZE=12800, CAPS=SPIRAM|8BIT, PTR=3D8025E8
[info] rg_gui_set_font_type: Font set to: points=12, scaling=1.00
[info] rg_audio_init: Audio ready. sink='Ext DAC', samplerate=32000, volume=50
[info] rtc_time_init: Time loaded from storage
[info] rtc_time_init: Time is now: Thu Jan  1 00:00:02 1970

[info] rg_system_init: Retro-Go ready.

[info] rg_alloc: SIZE=65280, CAPS=INTERNAL|8BIT, PTR=3FCB533C
[info] rg_alloc: SIZE=65280, CAPS=INTERNAL|8BIT, PTR=3FCC5240
[=] NES: System initialized!
[=] !! ROM: Unable to open file ''

*** RG_PANIC() CALLED IN 'app_main' ***
*** ROM load failed. ***

```

FantasyGmm, 2022-08-23:
It seems to have crashed at delete I2C, so couldn't start emulator after reboot because Panic
```
[info] rg_settings_commit: Saving 4 change(s)...
[info] rg_storage_deinit: SD Card unmounted.
[info] rg_input_deinit: Input terminated.
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4038425c  PS      : 0x00060133  A0      : 0x803844f7  A1      : 0x3fca6190  
0x4038425c: block_set_prev_free at /Users/fantasy/esp/esp-idf-esplay/components/heap/heap_tlsf_block_functions.h:111
 (inlined by) block_mark_as_free at /Users/fantasy/esp/esp-idf-esplay/components/heap/heap_tlsf_block_functions.h:163
 (inlined by) tlsf_free at /Users/fantasy/esp/esp-idf-esplay/components/heap/heap_tlsf.c:873

A2      : 0x3fca73a4  A3      : 0xf30a7d1c  A4      : 0x0000002a  A5      : 0x00000080  
A6      : 0x00000000  A7      : 0x0000002a  A8      : 0x3fca7d1c  A9      : 0xb33ffffc  
A10     : 0xfffffffc  A11     : 0x3fc98874  A12     : 0x3fc98874  A13     : 0x0000cdcd  
A14     : 0x00060123  A15     : 0x00060120  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xf30a7d20  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffff5  

Backtrace:0x40384259:0x3fca61900x403844f4:0x3fca61b0 0x403799b1:0x3fca61d0 0x403847e1:0x3fca61f0 0x4200596d:0x3fca6210 0x420199d6:0x3fca6230 0x420112da:0x3fca6260 0x4200def5:0x3fca6280 0x4200df0b:0x3fca62a0 0x4200dff0:0x3fca62c0 0x42007866:0x3fca62e0 0x42007945:0x3fca64b0 0x4202d326:0x3fca64d0 0x42006889:0x3fca64f0 0x4202ed13:0x3fca65e0 
0x40384259: offset_to_block at /Users/fantasy/esp/esp-idf-esplay/components/heap/heap_tlsf_block_functions.h:134
 (inlined by) block_next at /Users/fantasy/esp/esp-idf-esplay/components/heap/heap_tlsf_block_functions.h:146
 (inlined by) block_link_next at /Users/fantasy/esp/esp-idf-esplay/components/heap/heap_tlsf_block_functions.h:154
 (inlined by) block_mark_as_free at /Users/fantasy/esp/esp-idf-esplay/components/heap/heap_tlsf_block_functions.h:162
 (inlined by) tlsf_free at /Users/fantasy/esp/esp-idf-esplay/components/heap/heap_tlsf.c:873

0x403844f4: multi_heap_free_impl at /Users/fantasy/esp/esp-idf-esplay/components/heap/multi_heap.c:222
 (inlined by) multi_heap_free_impl at /Users/fantasy/esp/esp-idf-esplay/components/heap/multi_heap.c:210

0x403799b1: heap_caps_free at /Users/fantasy/esp/esp-idf-esplay/components/heap/heap_caps.c:340

0x403847e1: free at /Users/fantasy/esp/esp-idf-esplay/components/newlib/heap.c:39

0x4200596d: esp_intr_free at /Users/fantasy/esp/esp-idf-esplay/components/esp_hw_support/intr_alloc.c:693
 (inlined by) esp_intr_free at /Users/fantasy/esp/esp-idf-esplay/components/esp_hw_support/intr_alloc.c:634

0x420199d6: i2c_driver_delete at /Users/fantasy/esp/esp-idf-esplay/components/driver/i2c.c:407 (discriminator 2)

0x420112da: rg_i2c_deinit at /Users/fantasy/clionProject/retro-go/components/retro-go/rg_i2c.c:52

0x4200def5: shutdown_cleanup at /Users/fantasy/clionProject/retro-go/components/retro-go/rg_system.c:661

0x4200df0b: rg_system_restart at /Users/fantasy/clionProject/retro-go/components/retro-go/rg_system.c:685

0x4200dff0: rg_system_start_app at /Users/fantasy/clionProject/retro-go/components/retro-go/rg_system.c:696

0x42007866: application_start at /Users/fantasy/clionProject/retro-go/launcher/main/applications.c:150
 (inlined by) application_show_file_menu at /Users/fantasy/clionProject/retro-go/launcher/main/applications.c:626

0x42007945: event_handler at /Users/fantasy/clionProject/retro-go/launcher/main/applications.c:437

0x4202d326: gui_event at /Users/fantasy/clionProject/retro-go/launcher/main/gui.c:62

0x42006889: retro_loop at /Users/fantasy/clionProject/retro-go/launcher/main/main.c:202
 (inlined by) app_main at /Users/fantasy/clionProject/retro-go/launcher/main/main.c:329

0x4202ed13: main_task at /Users/fantasy/esp/esp-idf-esplay/components/freertos/port/port_common.c:141
```

FantasyGmm, 2022-08-23:
I doubled the Input task so it didn't crash anymore. All the emulators worked fine, including SNES, which got up to 60 frames, very smooth
But when the scene gets more complicated, it still drops to around 40

ducalex, 2022-08-23:
Thank you for the feedback, I guess 4.4 uses more stack to read the ADC (that's really all input task does). 

I've made the change you suggested!

FantasyGmm, 2022-08-24:
I'm sorting out the project. may submit a PR merge. I wrote a kconfig, so that I can easily use menuconfig to select hardware

FantasyGmm, 2022-08-24:
Are there any plans to support GBA? I think the performance of S3 may be enough to run smoothly

ducalex, 2022-08-24:
> Are there any plans to support GBA? I think the performance of S3 may be enough to run smoothly

I tried a few years ago without much success but if you succeed I'd be happy to take your PR :) .

FantasyGmm, 2022-08-26:
I hope so, but I have never ported an emulator. Do you know which GBA emulators are suitable to be ported to MCU

ducalex, 2022-08-26:
I believe this is what I used the last time: https://github.com/libretro/meteor-libretro .

Unfortunately I seem to have deleted the branch containing that work. I got it to start a game but it took a fair amount of time and it was extremely slow (about 1fps).

closed 2022-09-02 by FantasyGmm