ducalex/retro-go #28: Arduino mega chess port

mikeinkl1, 2022-09-11:
There is a port for the m5 stack, which kind of runs ok on the retroesp32 with a few minor changes to the code, but the ui with 3 buttons is clumsy. 
Maybe someone more skilled can do a proper port?
[image]

ducalex, 2024-05-03:
I was hoping someone would implement that but nobody did. So here I am two years later, interested in finally doing it!

Do you have a source for the ODROID-GO port in your image? It would be a good starting point.

mikeinkl1, 2024-08-23:
Great... sorry for the belated resopnse. 
i copied the source and fw here: https://github.com/mikeinkl1/retro-go/tree/master/chess