ducalex/retro-go #29: Pdoom has stack overflow

FantasyGmm, 2022-09-13:
**Describe the bug**
There is a stack overflow, but I don't know which stack it is, but there is no error in the previous firmware, I would like to ask whether there has been any modification recently.

**To Reproduce**
Start the game, it may freeze, TASK_WDT may trigger, or it may prompt for stack overflow, but the image can be displayed

**Additional context**
```
========================================================
prboom-go 1.33-43-g7c49438f-dirty (Sep 13 2022 19:03:20)
 built for: ESPLAY-S3. aud=0 disp=0 pad=0 sd=0 cfg=0
========================================================

[info] rg_system_init: Chip info: model 9 rev0 (2 cores), reset reason: 3
[info] rg_system_init: Internal memory: free=227747, total=279563
[info] rg_system_init: External memory: free=8386191, total=8386191
I (846) gpio: GPIO[40]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (856) gpio: GPIO[41]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (866) gpio: GPIO[39]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
[info] rg_storage_init: Storage mounted at /sd. driver=2
[info] rg_settings_init: Settings loaded from /sd/retro-go/config/retro-go.json.
[info] rg_i2c_init: I2C driver ready (SDA:48 SCL:47).
[info] rg_input_init: Input ready. driver='ESPLAY-I2C', state=00000000 00000000
[info] rg_display_init: Initialization...
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FCB8E88
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FCB988C
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FCBA290
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FCBAC94
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FCBB698
[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=3FCBC09C
[info] lcd_set_backlight: backlight set to 60.00%
[info] rg_display_init: Display ready.
[info] rg_alloc: SIZE=12800, CAPS=SPIRAM|8BIT, PTR=3D8025E8
[info] rg_gui_set_font_type: Font set to: points=12, scaling=1.00
[info] rg_gui_set_theme: Theme set to '(none)'!
I (1306) I2S: DMA Malloc info, datalen=blocksize=2136, dma_buf_count=8
I (1316) I2S: I2S0, MCLK output by GPIO0
[info] rg_audio_init: Audio ready. sink='Ext DAC', samplerate=11025, volume=45
I (1326) esp_image: segment 0: paddr=00100020 vaddr=3c040020 size=1b75ch (112476) map
I (1336) esp_image: segment 1: paddr=0011b784 vaddr=3fc96620 size=022a0h (  8864) 
I (1346) esp_image: segment 2: paddr=0011da2c vaddr=40378000 size=025ech (  9708) 
I (1356) esp_image: segment 3: paddr=00120020 vaddr=42000020 size=3fa60h (260704) map
I (1366) esp_image: segment 4: paddr=0015fa88 vaddr=4037a5ec size=0c028h ( 49192) 
I (1376) esp_image: segment 5: paddr=0016bab8 vaddr=50000000 size=00010h (    16) 
[info] rg_system_set_boot_app: Boot partition set to 16 'launcher'
[info] rtc_time_init: Time loaded from storage
[info] rtc_time_init: Time is now: Thu Sep  8 05:40:51 2022

[info] rg_system_init: Retro-Go ready.

[info] rg_alloc: SIZE=76800, CAPS=INTERNAL|8BIT, PTR=3FCC2950
[=] M_LoadDefaults: Load system defaults.
[=]  failed to read config from: /sd/roms/doom/prboom.cfg
[=] IWAD found: /sd/roms/doom/doom.wad
[=] prboom-go 2.5.0 (built Sep 13 2022), playing: DOOM Registered
PrBoom is released under the GNU General Public license v2.0.
You are welcome to redistribute it under certain conditions.
It comes with ABSOLUTELY NO WARRANTY. See the file COPYING for details.
[=] Save path set to: /sd/retro-go/saves/doom
[=] I_Init: Setting up machine state.
[=] D_InitNetGame: Checking for network game.
[=] W_Init: Init WADfiles.
[=]  added file:/sd/roms/doom/doom.wad (2194 lumps)
[=]  added data:prboom.wad (79 lumps)
[=] V_Init: Setting up video.
[=] V_InitMode: using 8 bit video mode
[=] M_Init: Init miscellaneous info.
[=] R_Init: Init DOOM refresh daemon:
[=] R_InitData: 
[=] R_InitData: Textures
[=] R_InitData: Flats
[=] R_InitData: Sprites
[=] R_InitData: Translucency
[=] R_InitTranMap: Gen from PLAYPAL(0)...
[=] done.
[=] R_InitData: Colourmaps
[=] R_InitLightTables [=] R_InitSkyMap [=] R_InitTranslationsTables [=] R_InitPatches [=] R_InitInterpolation
[=] P_Init: Init Playloop state.
[=] S_Init: Setting up sound.
[=] S_Init: channels=8, sfx=1, music=1
STACK:316, HEAP:1+7257 (0+7168), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:89.14
[=] HU_Init: Setting up heads up display.
[=] ST_Init: Init status bar.
[=] 
[=] mus2mid: Length: 42, Start: 1443, Channels: 11, Instruments: 13.
[info] update_viewport_scaling: 320x240@1.333 => 320x240@1.333 x_pos:0 y_pos:0 x_inc:320 y_inc:240
STACK:316, HEAP:0+6693 (0+6656), BUSY:100.00, FPS:16.00 (SKIP:12, PART:0, FULL:4), BATT:90.86
STACK:316, HEAP:0+6693 (0+6656), BUSY:99.17, FPS:35.00 (SKIP:0, PART:0, FULL:36), BATT:88.86
STACK:316, HEAP:0+6693 (0+6656), BUSY:100.00, FPS:35.00 (SKIP:0, PART:0, FULL:35), BATT:90.57
STACK:316, HEAP:0+6693 (0+6656), BUSY:100.00, FPS:35.00 (SKIP:0, PART:0, FULL:35), BATT:90.86
STACK:316, HEAP:0+6693 (0+6656), BUSY:100.00, FPS:35.00 (SKIP:0, PART:0, FULL:35), BATT:90.86
[=] G_DoPlayDemo: playing demo with doom/doom2 v1.9 compatibility
[=] mus2mid: Length: 24, Start: 9806, Channels: 7, Instruments: 4.
[=] P_GetNodesVersion: using normal BSP nodes
[=] R_PrecacheLevel: pre-cached 29 flats
STACK:316, HEAP:0+6446 (0+6400), BUSY:48.50, FPS:17.00 (SKIP:1, PART:0, FULL:16), BATT:89.43
[=] R_PrecacheLevel: pre-cached 186 textures
STACK:316, HEAP:0+6043 (0+6016), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:89.43
STACK:316, HEAP:0+5701 (0+5632), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:89.43
[=] R_PrecacheLevel: pre-cached 1029 sprites

***ERROR*** A stack overflow in task main has been detected.

Backtrace:0x403798be:0x3fcb59400x4037eee1:0x3fcb5960 0x4038182e:0x3fcb5980 0x4038076d:0x3fcb5a00 0x4037ef98:0x3fcb5a30 0x4037ef8e:0x3fcb5a50 0x4203df82:0x00000000  |<-CORRUPTED
0x403798be: panic_abort at /Users/fantasy/esp/idf-442/components/esp_system/panic.c:402

0x4037eee1: esp_system_abort at /Users/fantasy/esp/idf-442/components/esp_system/esp_system.c:128

0x4038182e: vApplicationStackOverflowHook at /Users/fantasy/esp/idf-442/components/freertos/port/xtensa/port.c:394

0x4038076d: vTaskSwitchContext at /Users/fantasy/esp/idf-442/components/freertos/tasks.c:3505

0x4037ef98: _frxt_dispatch at /Users/fantasy/esp/idf-442/components/freertos/port/xtensa/portasm.S:436

0x4037ef8e: _frxt_int_exit at /Users/fantasy/esp/idf-442/components/freertos/port/xtensa/portasm.S:231

0x4203df82: R_ProjectSprite at /Users/fantasy/CLionProjects/retro-go/prboom-go/components/prboom/r_things.c:543
 (inlined by) R_AddSprites at /Users/fantasy/CLionProjects/retro-go/prboom-go/components/prboom/r_things.c:664
```

ducalex, 2022-09-14:
It still works on the esp32, the worse HWM I can achieve is 1060 which is low but still fine. 

So it probably has to do with you compiling for the S3. The main task stack size is controlled by `CONFIG_ESP_MAIN_TASK_STACK_SIZE`, let me know which value works and I'll update baseS3.sdkconfig (or you can submit a PR directly :)).

FantasyGmm, 2022-09-14:
Both base.cmake and baeS3.sdkconfig need to be modified
Let me solve this problem and reorganize the code to submit a new pr

closed 2022-10-15 by ducalex