ducalex/retro-go #31: Pdoom panics while booting up

mtojek, 2022-09-22:
Hi,

I pulled the latest source code and unfortunately Pdoom panics right while loading:

 ```
STACK:4280, HEAP:3+3220 (1+3200), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:75.60
[=] HU_Init: Setting up heads up display.
E (21716) diskio_sdmmc: sdmmc_read_blocks failed (257)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4010b5e4  PS      : 0x00060f30  A0      : 0x80100cd7  A1      : 0x3ffcecxtensa-esp32-elf-addr2line: 'a.out': No such file
40
A2      : 0x3f851d14  A3      : 0x88cf0fa2  A4      : 0x00000000  A5      : 0x00002022
A6      : 0x00000000  A7      : 0x3ffcec40  A8      : 0x3f80c674  A9      : 0x00000000
A10     : 0x00004d44  A11     : 0x3f811854  A12     : 0x00004d44  A13     : 0x00000001
A14     : 0x00000000  A15     : 0x3f80c67c  SAR     : 0x0000000b  EXCCAUSE: 0x0000001c
EXCVADDR: 0x88cf0fa2  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNTxtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
  : 0x00000000

Backtrace:0x4010b5e1:0x3ffcec400x40100cd4:0x3ffceca0 0x400e82fe:0x3ffcecc0 0x400e3e7c:0x3ffced00 0x400e443a:0x3ffcef50 0x400d6ef5:0x3ffcef70 0x40138702:0x3ffcf010

HU_Init
/Users/x/Documents/Arduino/retro-go/prboom-go/components/prboom/hu_stuff.c:345
D_DoomMainSetup
/Users/x/Documents/Arduino/retro-go/prboom-go/components/prboom/d_main.c:975
D_DoomMain
/Users/x/Documents/Arduino/retro-go/prboom-go/components/prboom/d_main.c:1055
app_main
/Users/x/Documents/Arduino/retro-go/prboom-go/main/main.c:561
main_task
/Users/x/Documents/Arduino/esp-idf-v4.4.1/components/freertos/port/port_common.c:129
```

SD card reader works well in other applications and the launcher. It just complains in Doom.

ducalex, 2022-09-22:
> E (21716) diskio_sdmmc: sdmmc_read_blocks failed (257)

This means it ran out of DMA-capable heap. Normally there is a reserved 
memory pool for that, it is shared by many things (stacks, DMA, SPI, 
etc). Any changes you've made to those might be the cause. Different SDK 
versions also have different needs.

In other words the assumption I made when picking up that value might no 
longer apply to your setup. Have a look at 
`CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL`. You can also check 
`heap_caps_malloc_extmem_enable` that might help, by setting a low value 
like 1024. Just call once after rg_system_init.

---
Alex Duchesne

Links:
------
[1] https://github.com/ducalex/retro-go/issues/31
[2] 
https://github.com/

mtojek, 2022-09-23:
It looks like I fixed it by applying the following [changes](https://github.com/mtojek/retro-go/commit/2989a81310a0cbc7d85f3e33ccf1f80ed94e481e). I'm wondering if we can improve somehow the documentation/logging around this issue to shorten root cause analysis.

I would have to check again which config property was crucial here: `CONFIG_SPIRAM_CACHE_WORKAROUND ` or adjusting `CONFIG_SPIRAM_MALLOC_*`.

ducalex, 2022-09-23:
The cache workaround is enabled in DOOM already but it is true it isn't enabled in esp-idf so who knows.

I'd rather not enable it globally as it affects performance quite a bit, so please do more testing and let me know which setting truly solves the issue :) .

ducalex, 2022-10-01:
I saw the same issue while working on using esp-idf 4.3.3 for releases. A modest bump of 4KB seems sufficient. It's fixed in https://github.com/ducalex/retro-go/commit/5d9b96692617400e2e17a7845f5c251c79202c5a. Feel free to reopen if not working for you!

closed 2022-10-01 by ducalex