ducalex/retro-go #32: Launcher: audio sink issues

mtojek, 2022-09-22:
Hi,

With latest source code I can observe some issues with audio sink.

1. When I come back from the application (OPTIONS button), it just hangs and doesn't start the launcher:

```
[=] ROM: Mapper: 0, PRG:16K, CHR:8K, Flags: H---
[=] MMC: Mapper None (iNES 000)
[=] MMC: PRG-ROM: 2 banks
[=] MMC: PRG-RAM: 1 banks
[=] MMC: CHR-ROM: 1 banks
[=] NES: System type: NES-NTSC (NTSC-U)
[=] NES: System reset (hard)
[info] update_viewport_scaling: 256x224@1.143 => 320x240@1.333 x_pos:0 y_pos:0 x_inc:256 y_inc:224
STACK:5008, HEAP:61+4074 (45+4032), BUSY:37.18, FPS:37.09 (SKIP:1, PART:36, FULL:1), BATT:81.00
[info] rg_gui_dialog: text_buffer usage = 193
STACK:5008, HEAP:59+4074 (45+4032), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:81.80
STACK:5008, HEAP:59+4074 (45+4032), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:81.40
STACK:5008, HEAP:59+4074 (45+4032), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:80.60
STACK:5008, HEAP:59+4074 (45+4032), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:80.80
STACK:5008, HEAP:59+4074 (45+4032), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:80.60
[info] exit_handler: Exit handler called.
[info] rg_system_set_boot_app: Boot partition set to 16 'launcher'
[info] rg_audio_deinit: Audio terminated. sink='Ext DAC'
```

2. When recovery-quitting Pdoom after the other issue, I chose the "back to launcher" and observed the panic:

```
[info] rg_alloc: SIZE=12800, CAPS=SPIRAM|8BIT, PTR=3F802424
[info] rg_gui_set_font_type: Font set to: points=12, scaling=1.00
[info] rg_gui_set_theme: Theme set to '(none)'!
[warn] rg_system_init: Button 00000000 01000000 being held down...
[warn] rg_system_init: Button 00000000 01000000 being held down...
[warn] rg_system_init: Button 00000000 01000000 being held down...
[warn] rg_system_init: Button 00000000 01000000 being held down...
[warn] rg_system_init: Button 00000000 01000000 being held down...
[warn] rg_system_init: Button 00000000 01000000 being held down...
[warn] enter_recovery_mode: Entering recovery mode...
[info] rg_gui_dialog: text_buffer usage = 126
[info] rg_system_set_boot_app: Boot partition set to 16 'launcher'

*** RG_PANIC() CALLED IN 'rg_audio_set_mute' ***
*** Assertion failed: `audio.sink != NULL` : Audio device not ready! ***

abort() was called at PC 0x400dcfef on cxtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
xtensa-esp32-elf-addr2line: 'a.out': No such file
ore 0

Backtrace:0x400818da:0x3ffcebf00x4008758d:0x3ffcec10 0x4008c272:0x3ffcec30 0x400dcfef:0x3ffceca0 0x400d72a6:0x3ffcecc0 0x400d6715:0x3ffcece0 0x400dd3c5:0x3ffced00 0x400dd48e:0x3ffced20 0x400dd4bb:0x3ffced40 0x400dda32:0x3ffced60 0x400d6e7c:0x3ffcef70 0x40138702:0x3ffcf010

??
??:0
spi_flash_chip_gd_get_caps
/Users/x/Documents/Arduino/esp-idf-v4.4.1/components/spi_flash/spi_flash_chip_gd.c:38
_vsnprintf
/Users/x/Documents/Arduino/retro-go/components/retro-go/rg_printf.c:526
application_show_file_menu
/Users/x/Documents/Arduino/retro-go/launcher/main/applications.c:631
scan_folder
/Users/x/Documents/Arduino/retro-go/launcher/main/applications.c:41
_ftoa
/Users/x/Documents/Arduino/retro-go/components/retro-go/rg_printf.c:342
_ftoa
/Users/x/Documents/Arduino/retro-go/components/retro-go/rg_printf.c:350
_ftoa
/Users/x/Documents/Arduino/retro-go/components/retro-go/rg_printf.c:370
rg_settings_set_string
/Users/x/Documents/Arduino/retro-go/components/retro-go/rg_settings.c:155
application_init
/Users/x/Documents/Arduino/retro-go/launcher/main/applications.c:101
??
??:0
```

mtojek, 2022-09-22:
Re 1:

I investigated this issue a bit and it looks like code execution hangs on `fopen(RG_BASE_PATH_CONFIG "/clock.bin", "wb")`.  Is there something wrong with SD card initialization?

ducalex, 2022-09-22:
> *** RG_PANIC() CALLED IN 'rg_audio_set_mute' ***
> *** Assertion failed: `audio.sink != NULL` : Audio device not ready! ***

That part is likely a bug in retro-go and I'll investigate.

But the freeze on fopen might be related to your other ticket where we can see sdmmc errors in the log?

ducalex, 2022-09-22:
> *** RG_PANIC() CALLED IN 'rg_audio_set_mute' ***
> *** Assertion failed: `audio.sink != NULL` : Audio device not ready! ***

That part is likely a bug in retro-go and I'll investigate.

But the freeze on fopen might be related to your other ticket where we can see sdmmc errors in the log? DOOM is harder on the SD Card than most other apps because it does frequent and random accesses. If you did not apply the sdcard patch, or if you're running it too fast, this could be the reason.

mtojek, 2022-09-23:
> But the freeze on fopen might be related to your other ticket where we can see sdmmc errors in the log?

Yes, you're right. The bug disappeared when I applied this [fix](https://github.com/ducalex/retro-go/issues/31#issuecomment-1255921543). Thank you!

> That part is likely a bug in retro-go and I'll investigate.

I will leave it open as it can be easily reproduced.

ducalex, 2022-09-23:
I believe this has now been fixed :) .

commit 3c64fbc4 (2022-09-23): rg_system: Do not call app's event handler in recovery (Fixes #32)

During init process we should wait until AFTER recovery and panic handling before setting the app's event handlers.

The app should get events only once we know the init will go through.
closed 2022-09-23 by ducalex