ducalex/retro-go #35: Start Screen: Auto always loads Carousel on boot

Cralex, 2022-09-28:
**Describe the bug**
Any time the start screen setting is set to auto, the system boots to the carousel when it should've booted to the ROM browser. Only setting it to Browser loads the Browser on boot. Tested on ODG with public release 1.34.

**To Reproduce**
Steps to reproduce the behavior:
1. Opon the options menu and set Start Screen: to Auto
2. Go to the ROM browser and turn off the unit. (It doesn't matter if you play a game first or not.)
3. Turn on the unit again, and the Carousel for the previous system appears.

ducalex, 2022-09-28:
What you describe seems to be the expected behaviour. 

Auto does:
- Show browser when returning from an emulator
- Show carousel on cold boot

I can add more options maybe, what behaviour would you like to see?

Cralex, 2022-09-29:
Well, I _thought_ it would start at the browser from cold boot if you turned it off in the browser, but maybe I’m remembering it wrong and/or had it set to browser this whole time. I can just set it to browser if I’d rather it start in the browser from cold boot. Thanks.

closed 2022-10-10 by Cralex