ducalex/retro-go #45: compile error

mikeinkl1, 2022-10-24:
had to comment this line out in "doomdef.h" to get the latest release to compile:

#define exit(x) rg_system_switch_app(RG_APP_LAUNCHER, RG_APP_LAUNCHER, 0, 0);

ducalex, 2022-10-24:
What was the compile error?

mikeinkl1, 2022-10-25:
`../components/prboom/doomdef.h:47:72:` error: expected declaration specifiers or '...' before numeric constant
 #define exit(x) rg_system_switch_app(RG_APP_LAUNCHER, RG_APP_LAUNCHER, 0, 0);``

mikeinkl1, 2022-10-27:
I tried compiling the dev branch but it hangs before it even gets to doom.
`/retrogo/retro-go-dev/components/retro-go/rg_utils.c:119:15:` error: implicit declaration of function 'realloc' [-Werror=implicit-function-declaration]
    strings = realloc(strings, (strings_count + 1) * sizeof(char*));``

ducalex, 2022-10-27:
It's probably just missing an `#include <stdlib.h>` at the top of rg_utils.c, try adding it (I will push that fix myself later today too).

Which version of esp-idf and OS are you using? Because I wonder why there's so much differences between your setup and mine.

mikeinkl1, 2022-10-28:
I'm using a ubuntu VM and  ESP-IDF v4.1.3-129-g7c0f5e658e-dirty

ducalex, 2022-10-28:
Thanks for the info!

commit e4d642d1 (2022-10-25): DOOM: Fixed compilation with some versions of stdlib.h (Fixes #45)

This replaces calls to exit() by the already present I_SafeExit() which, I assume, was always intended as a wrapper for exit().
closed 2022-11-28 by ducalex