ducalex/retro-go #47: How to build a mrgc fw

mrkellyhuang, 2022-10-25:
Sorry ,this is not a bug.
I want to build mrgc F/W myself. 
I have cloned the project and configured the esp_idf 4.3.  
After running " ./rg_tool.py build-fw" , I got the file :"retro-go_1.35.1-dirty_odroid-go.fw", but it is not  for ESPGBC. 
Which config file should I modify to build the mrgc F/W.
Thank you.

KD-MM2, 2022-10-25:
You need to change the build target here:
https://github.com/ducalex/retro-go/blob/master/rg_tool.py#L20

mrkellyhuang, 2022-10-25:
Thank you for your reply. I will read these two documents carefully. I think I can solve it now. Thank you.

ducalex, 2022-10-25:
Just to clarify, there is also a flag for rg_tool.py, there is no need to edit its source: `./rg_tool.py --target=mrgc-g32 build-fw`

There is also an environment variable that can be used to define the default target instead: `RG_TOOL_TARGET=mrgc-g32`.

mrkellyhuang, 2022-10-26:
thank you Alex, thanks for your coding and improvement.

closed 2022-10-26 by mrkellyhuang