ducalex/retro-go #49: Stack smashing protect error while checking time with NTP

mtojek, 2022-10-30:
Hi,

I've been fighting with this issue for few hours:

```
$ xtensa-esp32-elf-addr2line -pfiaC -e launcher/build/launcher.elf Backtrace:0x400821b5:0x3ffd30a00x400900e1:0x3ffd30c0 0x40097c65:0x3ffd30e0 0x400d3844:0x3ffd3160 0x400e4dba:0x3ffd3180 0x400e4e71:0x3ffd3200 0x4016d742:0x3ffd3290 0x4016d279:0x3ffd32c0 0x4016d33d:0x3ffd3300 0x40093565:0x3ffd3320
0x00000bac: ?? ??:0
0x40097c65: abort at /Users/mtojek/Documents/esp-idf/components/newlib/abort.c:46
0x400d3844: __stack_chk_fail at /Users/mtojek/Documents/esp-idf/components/esp_system/stack_check.c:36
0x400e4dba: rg_network_sync_time at /Users/mtojek/Documents/retro-go/components/retro-go/rg_network.c:160
0x400e4e71: network_event_handler at /Users/mtojek/Documents/retro-go/components/retro-go/rg_network.c:55
 (inlined by) network_event_handler at /Users/mtojek/Documents/retro-go/components/retro-go/rg_network.c:33
0x4016d742: handler_execute at /Users/mtojek/Documents/esp-idf/components/esp_event/esp_event.c:145
0x4016d279: esp_event_loop_run at /Users/mtojek/Documents/esp-idf/components/esp_event/esp_event.c:598 (discriminator 3)
0x4016d33d: esp_event_loop_run_task at /Users/mtojek/Documents/esp-idf/components/esp_event/esp_event.c:115 (discriminator 15)
0x40093565: vPortTaskWrapper at /Users/mtojek/Documents/esp-idf/components/freertos/port/xtensa/port.c:131
```

I found out that `serv_addr` is being corrupted by memcpy. Apparently, in my case `h_length`= 24 (?!) and I couldn't find an explanation why does it happen. To fix it, I had to replace `h_length` with 4 (as IPv4):

```
memcpy(&serv_addr.sin_addr.s_addr, server->h_addr, 4);
```

I tried to print the content of hostent:

```c
    RG_LOGI("addr bc: %s, addr_list: %d, h_length: %d\n", inet_ntoa(server->h_addr), sizeof(server->h_addr_list), server->h_length);

    struct in_addr **inaddr_ptr;
    inaddr_ptr = (struct in_addr **) server->h_addr_list;
    while (*inaddr_ptr != NULL)
    {
       RG_LOGI("addr: %s\n", inet_ntoa(**inaddr_ptr));
       inaddr_ptr++;
    }
```

... and got this:

```
STACK:4936, HEAP:82+3635 (62+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:78.80
STACK:4936, HEAP:82+3635 (62+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:63.40
[info] network_event_handler: Connected! IP: 192.168.0.25, RSSI: -55
[info] rg_network_sync_time: addr bc: 36.1.128.63, addr_list: 4, h_length: 24
[info] rg_network_sync_time: addr: 94.23.94.78
[info] rg_network_sync_time: Received Time: Sun Oct 30 13:54:28 2022, we were 200206ms behind
[info] rg_system_save_time: System time saved to storage.
```

Bonus points: why does ESP32 return 36.1.128.63 (cn)? The other IP (94.23.94.78) is correct for my location (`dig +short pool.ntp.org`).

ducalex, 2022-10-30:
I don't want to hardcode length to 4 but I suppose we could do use the size of the destination instead, eg
````c
memcpy(&serv_addr.sin_addr.s_addr, server->h_addr, sizeof(serv_addr.sin_addr.s_addr));
`````

but that doesn't really resolve the weird length... 24 doesn't seem to match anything.

Any suggestion? maybe another function we can use?

mtojek, 2022-10-30:
> I don't want to hardcode length to 4 but I suppose we could do use the size of the destination instead, eg

👍 

> but that doesn't really resolve the weird length... 24 doesn't seem to match anything.

I looked at man, but I only found references to length = 8 or length = 16 (IPv6). No idea what does 24 mean. Maybe I should report a bug to esp-idf?

ducalex, 2022-10-30:
I looked into esp-idf and it returns an array of this:
````c
typedef struct ip_addr {
  union {
    ip6_addr_t ip6;
    ip4_addr_t ip4;
  } u_addr;
  /** @ref lwip_ip_addr_type */
  u8_t type;
} ip_addr_t;
````

which is indeed 24 bytes. I'm not familiar enough with gethostbyname to know if this is acceptable so I don't know if it's worth you reporting it to esp-idf.

But in any case the address is still at the beginning of the struct, so copying the first 4/16 bytes is correct (done in b08f1ae8cfaac2df1bfd3974e7f1526ac5076066 but I'm sure there's a better way).

closed 2022-10-30 by ducalex

mtojek, 2022-10-30:
Thanks for taking care of this.