ducalex/retro-go #51: rg_input_read_battery: noise to signal ratio

mtojek, 2022-11-01:
Hi,

Today I spent some time examining the battery status. I connect GPIO36 to GND via 10k resistor, and to VCC via 10k resistor. I measured the VCC voltage and it's ~4.787V. Unfortunately I can't get that value reported by retro-go:

```
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:80.20 (4.80V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:80.80 (4.81V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:81.20 (4.81V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:87.60 (4.88V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:87.40 (4.87V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:87.80 (4.88V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:85.60 (4.86V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:97.80 (4.98V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:70.40 (4.70V) <--- !!!!
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:80.80 (4.81V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:88.00 (4.88V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:86.00 (4.86V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:88.20 (4.88V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:88.40 (4.88V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:87.00 (4.87V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:80.60 (4.81V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:100.00 (5.09V) <--- !!!!
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:88.20 (4.88V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:88.80 (4.89V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:80.80 (4.81V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:88.20 (4.88V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:79.60 (4.80V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:86.00 (4.86V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:66.40 (4.66V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:88.00 (4.88V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:87.60 (4.88V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:81.40 (4.81V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:86.40 (4.86V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:86.60 (4.87V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:88.20 (4.88V)
STACK:4892, HEAP:62+3483 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:87.80 (4.88V)
```

I tried:
1. Put extra 104 capacitor and grounded it.
2. Set the proper eFuse value (1177mv). _BTW should this value be configurable via targets?_
3. Increased number of samples to 16.

Values are oscillating around 4.88V, but I'd like to achieve a higher precision.

Do you have any clues how can I improve readings?

Thanks for all suggestions!

ducalex, 2022-11-01:
I think you're near the top of the range with that voltage. Maybe try a 1:3 ratio in your divider instead? Adding a rolling average could help with stability but probably not accuracy. Unfortunately the esp32's ADC is famously bad :( .

As for the vref in target, well it varies per chip so I don't think it's a device-specific thing.

mtojek, 2022-11-01:
Well, with 3 resistors (1:3 ratio), things also don't look promissing:

```
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:83.60 (4.84V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:84.80 (4.85V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:89.90 (4.90V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:84.20 (4.84V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:90.80 (4.91V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:86.90 (4.87V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:89.90 (4.90V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:100.00 (5.08V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:83.90 (4.84V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:90.50 (4.91V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:89.60 (4.90V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:90.20 (4.90V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:82.40 (4.82V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:100.00 (5.19V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:91.10 (4.91V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:93.80 (4.94V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:90.20 (4.90V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:94.70 (4.95V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:89.60 (4.90V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:91.70 (4.92V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:82.40 (4.82V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:90.80 (4.91V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:89.90 (4.90V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:89.00 (4.89V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:91.10 (4.91V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:81.80 (4.82V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:91.70 (4.92V)
STACK:4892, HEAP:63+3635 (48+3392), BUSY:0.00, FPS:0.00 (SKIP:0, PART:0, FULL:0), BATT:90.20 (4.90V)
```

Do you thing that this is an outcome of bad calibration or I should add some filters.

BTW is there any default/official/recommended schematics of the console? It might be a good idea to note some tricks for different platforms.

KD-MM2, 2022-11-01:
Odroid-Go official document:
https://wiki.odroid.com/odroid_go/odroid_go

mikeinkl1, 2022-11-02:
> Hi,
> 
> Today I spent some time examining the battery status. I connect GPIO36 to GND via 10k resistor, and to VCC via 10k resistor. I measured the VCC voltage and it's ~4.787V. Unfortunately I can't get that value reported by retro-go:
> 

2x10k is too low. It'll drain your battery too. Try 2x100k or maybe 2x470k  and a capacitor on the input pin. I use 2x100k like on the original odroid go and it also isn't total stable, but good enough.

ducalex, 2022-11-02:
We could consider adding back a rolling average, or at least re-enable the old averaging (the two commented lines in rg_input_read_battery) but 100mV fluctuation is quite a lot even for the esp32 so there are likely electrical improvements to be made too.

mikeinkl1, 2022-11-02:
Yes, main issue is probably the hardware, and this may be reated the screen flicker and audio noise as well. Rolling average will help, but finding out why the launcher screen flickers but not the menu screen would be better. Does the launcher read the sd card on every cycle perhaps? If so, can that be changed?

ducalex, 2022-11-03:
> Does the launcher read the sd card on every cycle perhaps? 

It doesn't. There used to be a background task but it was disabled when I added wifi. There is currento no sd card access at all when idle. 

If wifi is enabled then that might be the cause. When transmitting it can easily use over 150mA for a few ms, this is a frequent cause of instability in esp32 projects.

mikeinkl1, 2022-11-03:
Yes.. I just unplugged the sd card and the flicker was still there. I dont think it is the wifi either as the flicker already occurred in the previous releases. 
There must be something else that gets called in the launcher loop, but not when the menu is open.   Maybe time to buy an oscilloscope and see whats really going on.

ducalex, 2022-11-03:
The screen is redrawn 10x per second. Other than that I don't think the launcher does anything when idle anymore, it's basically sleeping in a loop.

 You can tune the redrawing by editing launcher/main.c:

````c
        else if (rg_system_timer() >= next_idle_event)
        {
            gui.idle_counter++;
            gui.joystick = 0;
            prev_joystick = 0;
            gui_event(TAB_IDLE, tab);
            next_idle_event = rg_system_timer() + 100000; <---- or you can increase that number for even less fps
            redraw_pending = true; <------------ comment this line
        }
````

But frankly the emulators do a lot more redrawing so it would be weird if that were the issue.

mikeinkl1, 2022-11-04:
yes, that fixed the issue. 
I think what happens is that during the redraw gui_draw_background is called, which calls gui_get_image, which reads (and writes?) to the sd card. Screen and sd card are on the same spi bus which cause the flickering probably. And the audio noise when the SD card is accessed is definitely related to a power or crosstalk issue, which of course cant be fixed in software. 
The emulators dont use the SD card and thats why there is no flicker or noise. 
I cant see any adverse effects of commenting the line out, so you may want to consider changing that in the code.

ducalex, 2022-12-14:
I added averaging to retro-go directly a while ago that might help + the electrical changes you have made.

Is this issue still relevant?

mikeinkl1, 2022-12-16:
I tried compiling 1.36.3 but it hangs here:
 `/home/mike/retrogo/retro-go-1.36.3/components/retro-go/rg_storage.c:338:37: error: 'struct stat' has no member named 'st_mtim'; did you mean 'st_mtime'?
             result->mtime = statbuf.st_mtim.tv_sec;
                                     ^~~~~~~
                                     st_mtime
`

ducalex, 2022-12-16:
The stat and dirent structs are so annoying, it's always a pita to make portable use of it.

It's now fixed in dev branch but you can make the change on your local branch: https://github.com/ducalex/retro-go/commit/cfa842fa65efe2a8b17aa5b1c46cf9a2b66deb1a

mikeinkl1, 2022-12-16:
Thanks.. it compiles now but the noise is pretty much the same. I reduced the refresh rate to 1Hz which makes it hardly noticeable.

ducalex, 2024-02-28:
I've added thresholds to reach before the UI/console is updated, this eliminates the noise issue (well, it hides it).

In other words:
- the percent value will only be updated when the difference is at least 1%
- the voltage value will only be updated when the difference is at least 10mV

This fixes the flicker in the battery icon, it fixes the jumps in the serial console, etc. Note that battery is still being read every second so any large change will reflected immediately, so we're not losing anything.

(First implemented in f6b8df6de1cb6135b8c2d5514114046b07ca6330 which is now merged to master, made configurable in 0e237db07a8708576730c2faf6705297a139eb49 which is not yet merged to master).

In my eyes this solve the issue as best as we possibly can, so I'm closing this issue!

There were a few tangential issues in this thread. If they are still relevant, please open a separate issue :)

closed 2024-02-28 by ducalex