ducalex/retro-go #53: ESP32 Mini Game Box Port

Cralex, 2022-11-01:
As you may have seen on the forums, I was able to build a working firmware image! I'm building on windows 10 with ESP-IDF 4.1 based on version 1.3.3, since this is all what I had set up already. 

Like you suggested, basing it off of the mrgc-g32 target worked for generating an image that the device's bootloader would allow to flash. After tweaking just a few values, I was able to build the launcher and gnuboy. The speaker, screen (with brightness controls), and buttons all work out of the box. The buttons match the vendor's firmware, so from left to right it's Start, Select, Menu, and Power. (You hold Menu/Sound on boot to access the bootloader.) If I enable the option button variable, I imagine the Option button will map itself to Power. (You have to hold down the Power button to shut the unit down, so this'll work fine.)

I'm attaching the modified target file that I used to build the firmware. (With the resolution flipped the right way around.)
[mrgc-g32.zip](https://github.com/ducalex/retro-go/files/9914324/mrgc-g32.zip)

I noticed that current retro-go has made some changes in the target file, so now I'm working on building the release version for this device to see if it'll work and if I can get wifi working. The finer details, such as battery capacity, are unfortunately beyond me.

Regarding getting a serial console going... It appears on my computer as "USB-SERIAL CH340 (COM16)" whether or not Retro-Go is running. So I tried running `rg_tool.py --port=COM16 --target=mrgc-g32 run launcher` while Retro-Go was booted, but it failed to connect. I can give you the exact error message once I'm done trying to build more firmware. :)

ducalex, 2022-11-03:
Regarding serial, it's possible that they haven't wired all the signals and you'd need to manually put the device in flash mode (like holding a button while powering up). This is an issue on the mrgc-g32 too, very annoying. 

But, even if you can't easily flash over serial, you should be able to at least access the log with `rg_tool.py --port=COM16 monitor` or using another program like CoolTerm or putty (COM16 at 115200).

Cralex, 2022-11-03:
OK, I will try that. In the meantime, I was able to get a 4.3X IDF running and used it to successfully build the current version of Retro-Go for the device, including networking. I will try poking around with the serial connection and see if I can learn anything else.

KD-MM2, 2022-11-03:
> Regarding serial, it's possible that they haven't wired all the signals and you'd need to manually put the device in flash mode (like holding a button while powering up). This is an issue on the [...]

@Cralex  You can use the ArduinoIDE for convenience, just select the Port, and click the button. But remember to close the Serial Monitor window before flashing or something else, or you will get "Device is busy" error.

Cralex, 2022-11-07:
@KD-MM2 Thanks for the tip. :)

I got my device up and running again. I think I’m monitoring the wrong thing for the battery level, since it’s stuck at 100 over the serial connection. I let the battery die earlier, and the power LED flickered only moments before the unit shutdown, whereas I’m pretty sure the original firmware gives an earlier warning. If I can find the right entry in the original firmware’s source code that should tell me where to look…

Does anyone have any ideas what I can experiment with to fix the screen colors? Darker colors tend to look almost normal...
[image]
It seems brownish/reddish in some places...
[image]
[image]
[image]
Some grays (the title screen of Link to the Past) are indistinguishable from each other.
[image]

Overall, the screen works very well in all areas other than this. The ILI9341 driver clearly has no problem initing the screen, but I don't know enough about little displays like this to see what the problem could be.

ducalex, 2022-11-07:
That's likely the wrong gamma curve. Maybe you can find the proper ones in the original source code, they're ili commands 0xE0 and 0xE1. Could also be the pixel format (command 0x3A) but I feel like that would be more obvious.

I'm not an expert either, it could be any other missing or incorrect init command, but that's where I'd start :) .

Cralex, 2022-11-07:
> That's likely the wrong gamma curve. Maybe you can find the proper ones in the original source code, they're ili commands 0xE0 and 0xE1. Could also be the pixel format (command 0x3A) but I feel like [...]

GOT IT! I found the display driver in the original source code. The following gamma settings cause the proper colors to appear:
`{0xE0, {0xD0, 0x00, 0x05, 0x0E, 0x15, 0x0D, 0x37, 0x43, 0x47, 0x09, 0x15, 0x12, 0x16, 0x19}, 14},
{0xE1, {0xD0, 0x00, 0x05, 0x0D, 0x0C, 0x06, 0x2D, 0x44, 0x40, 0x0E, 0x1C, 0x18, 0x16, 0x19}, 14},`
Guess we'll need to add a new display type for this unit after all. :) I'll look over the driver and see if anything else needs to be changed for the perfect display experience.

ducalex, 2022-11-10:
I can't comment on what would feel best on this device, but we certainly can add more layout options in SNES if that's the only blocker :).

Cralex, 2022-11-10:
> I can't comment on what would feel best on this device, but we certainly can add more layout options in SNES if that's the only blocker :).

It would be ideal if the two buttons above A and B could be X and Y, but we aren't fortunate enough to have that option. :) Maybe I'll find some other places to ask around for opinions. (Other than the "It'll give me hand cramps!" or "I'll go blind squinting at that tiny screen! that tiny devices always get.) Point is, it _CAN_ run Retro-GO, so I won't quit until it does. :)

Cralex, 2022-11-28:
I'm rebasing development on the shiny new release. 😀 

The build file needed to be updated in order to build properly. (See my pull request.) However, the resulting firmware does not flash. [image][image][image]

My first thought was that the format wasn't correctly set to ESPlay, but so far I haven't found anything that would indicate that. If I find it before anyone else, I'll submit another pull request.

ducalex, 2022-11-29:
No you're right, something went wrong when I merged the branches. It's not respecting the env.py files.

ducalex, 2022-11-29:
It should be fixed now! I guess I'll have to re-issue the release, the format was broken for MRGC-G32 as well. Good thing I hadn't advertised it yet :).

Cralex, 2024-04-10:
Despite the unobtanium nature of this device, someone generously offered to sell me their metal case unit when they saw that my plastic unit had a broken screen. Upon receiving it, I realized that the metal case has a taller view of the screen by ten pixels, which causes some unsightly lines of garbage when playing on a metal unit, that gets masked off on a plastic unit. I have adjusted the screen height and bottom margin to account for this, leaving a nice black bar on the bottom of the screen. Setting the bottom margin to zero, meanwhile, uses the full height of a metal unit but would crop off the bottom of a plastic unit.

For my pull request, I have a line that you uncomment to achieve the maximum screen height depending on which case you have, with the plastic case line uncommented by default as a reasonable compromise. I'd like to make it easy for any metal-case owners out there to use their whole screens, but having a separate image built based on a single variable changing would kind of bug me. But maybe it shouldn't?

ducalex, 2024-04-10:
You're right making two builds would be annoying!

I guess we could add an advanced menu to tweak things like margins at run time? There's no reason why margins couldn't be dynamic. Menu could contains other advanced things like overclocking or even SD/LCD Card frequency. 

I'm not sure where's the best place to put that new menu, though. I don't want things to get even more cluttered.

Cralex, 2024-04-10:
Yeah, it wouldn't be great to clutter things up more, especially since it's only one target (so far) that realistically needs more than one margin setting. Of the few people that I know of that have the GameBox Mini, only a handful of them have ever had a metal case unit.

closed 2025-08-24 by ducalex