ducalex/retro-go #54: Build Error on PRBOOM

32teeth, 2022-11-02:
**Describe the bug**

* did a rebase
*  **run** `python3 rg_tool.py --target retro-esp32 build-img`
*  **error** `components/retro-go/config.h:32:25: error: expected declaration specifiers or '...' before string constant
 #define RG_APP_LAUNCHER "launcher"` 
* **error** `prboom-go/components/prboom/doomdef.h:47:72: error: expected declaration specifiers or '...' before numeric constant #define exit(x) rg_system_switch_app(RG_APP_LAUNCHER, RG_APP_LAUNCHER, 0, 0);`

Also tried full clean

32teeth, 2022-11-02:
Or...  ...where can i comment out prboom in `build-img all`

32teeth, 2022-11-02:
Thanks @mikeinkl1 
I simply just commented it out since i did not want it in my build ;)

ducalex, 2022-11-03:
Duplicate of #45

closed 2022-11-03 by ducalex