ducalex/retro-go #57: Issue with ESPlay micro

monsieurPiGG, 2022-11-03:
retro-go (mrgc-g32 last version) installs and launch in ESPlay micro but display remains white.
We can argue it installs since retro-go folder are created in the sd card.
ESPlay should be a clone of odroid-go.
Here are the specs :
ESP32 WROVER, Dual-core processor with Integrated 16MB Flash + 8MB PSRAM
Integrated WIFI and Bluetooth 4 BLE
2,4" ILI9341 TFT Panel
Micro SD slot connected to SDMMC Host in 1 Line Mode for saving GPIO pin
Integrated I2S DAC via UDA1334A
Is it a known issu Edith other odroid go clones ? Is there a workaround ? Thanks !

Cralex, 2022-11-05:
I noticed that there's an esplay target, but for some reason it's not included in the main releases. (Maybe it's not done?) I don't have an esplay micro of my own, but maybe this firmware I built will work for you. 
[retro-go-1.35.1-esplay-s3.zip](https://github.com/ducalex/retro-go/files/9944625/retro-go-1.35.1-esplay-s3.zip)

monsieurPiGG, 2022-11-05:
Thanks ! Unfortunately it doesn't flash, There is an header mismatch error and erase error.

monsieurPiGG, 2022-11-07:
Edit : the .img flash but it's stuck with a white screen, as for mrgc-g32 release...
Indeed, I can see an esplay target in rg_tool.py...

ducalex, 2022-11-07:
Esplay micro isn't supported. No reason it couldn't be, but someone will have to write the target file :) . The mrgc target is a good starting point, from what I can tell it's already a close match in terms of hardware and pinout. Maybe borrowing the lcd init sequence from the esplay micro's code would be enough, I don't know.

Cralex, 2022-11-10:
@monsieurPiGG Do you have a build environment set up? If I have time, maybe I can look at the source code for one of your device's firmwares and see if there are any clues there to get your screen working.

monsieurPiGG, 2022-11-28:
Hi ! Sorry for the delay !
@Cralex, not yet but it's my first step for sure !

Cralex, 2023-01-12:
I have an ESPlay Micro V2 on order. Time permitting, I'll be able to attempt building and testing Retro-Go ports for it once it finally arrives. 🚀

monsieurPiGG, 2023-01-18:
Sorry I didn't,t have time to try something...
That's sound good ! I guess display init sequence is the same for both v1 and v2. I had a v2 running the same firmware...

Cralex, 2023-01-19:
That is an important clue, though. I noticed, of the multiple firmwares available for the ESPlay Micro, none of it is labeled V1 or V2. If they are in fact all capable of running the same firmware, that gives us a lot of working code to base a Retro-Go port off of and I might be justified in naming the target just ESPlay Micro. 

@ducalex I noticed that it's possible to specify shoulder buttons in Retro-Go targets. Since this device actually has some, are the emulators set up to take advantage of them?

Cralex, 2023-01-22:
Sorry to bug, @ducalex , but I'm having some trouble building a compatible firmware file that doesn't do the "header mismatch error" thing. Also, the firmware files from https://github.com/pebri86/esplay-micro-firmware-collections don't seem to work either. I'm attaching some firmware that came on the device's card. Could you possibly check it to see what format we should aim for? Thanks.

[firmware.zip](https://github.com/ducalex/retro-go/files/10474772/firmware.zip)

ducalex, 2023-01-24:
> @ducalex I noticed that it's possible to specify shoulder buttons in Retro-Go targets. Since this device actually has some, are the emulators set up to take advantage of them?

Currently no but rg_input supports them in most drivers I think, so it's only needed to be added on the emulator side. Feel free to send a PR or just tell me what mapping should be added where :).

> I'm attaching some firmware that came on the device's card. Could you possibly check it to see what format we should aim for? Thanks.

Those .fw files seem to match the esplay format:
- 22 bytes magic "ESPLAY_FIRMWARE_V00_01"
- 40 bytes name "Retro-Emulation"
- 8256 bytes icon 
- data
- crc32

I don't see anything obvious as to why those are accepted as valid and the ones produced by retro-go (in esplay mode) are not...

monsieurPiGG, 2023-01-25:
> Perhaps worst of all, my screen came crooked and was mounted with some really sticky adhesive foam.

Indeed. You can lift the screen up with a thin blade. I had a strong double sided tape, the one used for carpets.
Overall, if you want something that feel stronger and have access to a 3D printer, I've designed this tiny case for v2 esplay micro board : https://www.thingiverse.com/thing:5592683

That's strange that the mrgc-g32 retro go .fw doesn't have the header mismatch error and seems to install since retro go folders are created in sd card.

Cralex, 2023-01-25:
@monsieurPiGG That's a cute case! I don't have access to a 3D printer but I might have to order one made at some point.

Regarding Retro-Go folders, you mentioned that you tried flashing the .img file that I uploaded before and that succeeded, but there was nothing on the screen, right? I am guessing that Retro-Go was able to boot at that time (just without the proper screen configuration) and that's when the folders were created.

I'll keep trying and see if I come up with anything. This device is unusual among the other ESP32 handhelds I've used in that it displays an animated splash screen before starting the installed firmware or boot loader. No idea how that works or if it matters. I'll probably try to flash it over USB or something and see if I can at least get it running.

monsieurPiGG, 2023-01-25:
> you mentioned that you tried flashing the .img file that I uploaded before and that succeeded, but there was nothing on the screen, right?
I meant the mrgc-g32 release of retro go. This one seems to launch, but with a display issue. Unfortunately, the .fw and the .img files you provided earlier gave the header mismatch error.

Cralex, 2023-01-25:
Progress, but ugh, it was a bear getting here.

[image]

(I'm partly writing this so I don't forget how on earth I got here.) Since I can't get it to accept .fw files, I tried flashing the .img file I compiled instead using esptool. It kept failing to connect and giving me a `Invalid head of packet (0x45)` error, so I went over to where the base firmware is (https://github.com/pebri86/esplay-base-firmware/issues) to flash that, just to see if I could. Same problem. I was about to give up when I tried pressing the little reset button on the back of the board while it was trying to connect. Finally, I have some means of testing Retro-Go and also restoring the stock bootloader when needed. The screen works, but it's rotated wrong and inverted(?!) but I know what to poke with to fix that. Just need to take a break first...

monsieurPiGG, 2023-01-27:
@Cralex it's amazing ! You're almost there !

ducalex, 2023-01-27:
> Edit: Since I actually know very little about how esp32 works under the hood, what defines whether or not a .fw needs to be in ESPlay format or not, the hardware or the bootloader, @ducalex ?

It is the flashing utility that resides in the factory partition (the app that starts when you power up holding a specific button). It's called odroid-go-firmware on the ODROID GO.

> If for some reason, I am never able to figure out how to load Retro-Go in .fw format in the stock bootloader, would porting and flashing your Odroid Multi-Firmware bootloader neutralize the format problem? 

Yes it would, but surely finding the .fw format would be easier!

The .fw format can be known from two sources:
- The source of their odroid-go-firmware's version/fork (I don't know which came first...)
- The source of their mkfw tool (I couldn't locate it, but it's called in the .sh file you showed)

Alternatively if someone can provide me a .fw that IS recognized by the firmware and a .fw that ISN'T but SHOULD BE recognized, I might be able to find the issue. It has to be something minor!

---

I can't explain why your device rejects the mrgc-g32 builds I published, but have you double checked that your own builds are indeed in the esplay format? Maybe your target isn't configured correctly and retro-go makes the .fw in the odroid-go format?

Cralex, 2023-01-28:
I'm very nearly done with the screen. It fills the display properly and everything looks great, but everything is still flipped across the y-axis. (So, right-side up but mirrored.) I tried changing the value of `#define RG_SCREEN_ROTATE` but it doesn't seem to change anything. I have no idea what I'm missing. Mind taking a look? My screen driver is [here](https://github.com/Cralex/retro-go/commit/cfcd09d805aea3396d0877a8fa911297e11a3274) and the reference I used to write it from the stock fw is [here](https://github.com/pebri86/esplay-retro-emulation/blob/cf68ea80ea34987592b4fb1ba650a074ea85b839/esplay-sdk/esplay-components/esplay-hal/src/ili9341.c). And here's the current state of the target: [esplay-micro.zip](https://github.com/ducalex/retro-go/files/10526564/esplay-micro.zip)

I looked at the resulting firmware in a hex editor and it definitely showing as an esplay firmware. Might looking at the various [emulator firmware](https://github.com/pebri86/esplay-micro-firmware-collections/commits/master/esplay-retro-emu.fw) versions for the Esplay Micro line help? The "update 2.0" version is an exact match for the one on the card and is valid, while the "new release" one is for the V1 and is not valid. (Other changes were made as well, I'm sure.)

Still do do: verify sound works, finish mapping buttons, make sure networking works... Also flashing the .img never seems to allow me to actually launch an emulator. That's not normal, right? I'm only building with retro-core and launcher for now. Haven't checked for serial output yet.

ducalex, 2023-01-28:
> I'm very nearly done with the screen. It fills the display properly and everything looks great, but everything is still flipped across the y-axis. (So, right-side up but mirrored.) 

Take a look at the 0x36 command/register, it controls scanning direction. It's called "Memory Access Control" in the [datasheet](https://cdn-shop.adafruit.com/datasheets/ILI9341.pdf).

Can't really help with your other questions but I imagine obtaining debug output would be a good place to start :).

monsieurPiGG, 2023-01-28:
I don't know if it could help, but I've ever use ILI9341 display with rasp pi. In raspbian (config.txt), display_lcd_rotate can take that arguments : 
- 1 for 90 degrees clockwise
- 2 for 180 degrees clockwise
- 0x10000 for horizontal flip
- 0x20000 for vertical flip

monsieurPiGG, 2023-01-28:
> Take a look at the 0x36 command/register, it controls scanning direction. It's called "Memory Access Control" in the [datasheet](https://cdn-shop.adafruit.com/datasheets/ILI9341.pdf).

Can this be helpful ?
https://github.com/Bodmer/TFT_eSPI/issues/544

Cralex, 2023-01-29:
Thanks @ducalex for pointing me in the right direction, and @monsieurPiGG for the legwork. We have a fully functional screen, ready to be merged. 😁

[image]

nod3011, 2023-01-31:
> Thanks @ducalex for pointing me in the right direction, and @monsieurPiGG for the legwork. We have a fully functional screen, ready to be merged. 😁
> 
> [image]
> 
> Edit: Probably gonna call it a night. I tinkered around a little now that I have a legible display. Here's where I'm at.
> 
> * Networking: Seems to work fine. I was able to create an AP and transfer a file over. Need to test client mode. (Had problems, see last bullet point.)
> * Buttons: Most of the main buttons work with the target file that I based off of the g32 except for Menu, L, and R. They seem to be connected a different way from the rest. Need to figure out how they're connected and how to express it properly in the target file. Edit: Additional buttons are connected via a PCF8574 I2C GPIO, but I don't know how to address them yet. I found what appears to be a [list](https://github.com/pebri86/esplay-retro-emulation/blob/master/esplay-sdk/esplay-components/esplay-hal/include/pin_definitions.h) of the buttons as GPIO, but trying to map them that way caused the function I was trying to map (Menu button) to be called when I pressed up and down.
> * Fw format: I flashed the stock bootloader again and, just for fun, tried flashing the g32 firmware. Still didn't work, but I got more that just the header match error. When I tried flashing it anyway, I got checksum mismatch error (it was gone so fast that I had to film in slow motion to read it) and partition count error, which is where the bootloader hangs.
> * Launching emulators: ~As stated above, the launcher restarts when I try to start an emulator. I got serial output, but didn't see anything that looked like "something's wrong, so I'll just restart the launcher". (Having trouble copying it out of what I was using to monitor it, or else I'd post the output.) **But**, I'm starting to wonder if my sd card is faulty and might even be causing the problem. I noticed a lot of retro-go creating directories in the serial output that should've been there already, and it doesn't always reliably appear in a computer or the unit itself. I imagine retro-go wouldn't know what to do if the files it just created aren't there when it goes to read them... I'll try with a known good card and see if anything changes. After sleep.~ Retro-Go didn't like the janky card the unit came with. Now that it's not losing the files it creates, the thing boots games now.

This look great.

ducalex, 2023-02-01:
> Fw format: I flashed the stock bootloader again and, just for fun, tried flashing the g32 firmware. Still didn't work, but I got more that just the header match error. When I tried flashing it [...]

That's progress at least!

I built a .fw with our mkfw.py and their mkfw.c[1] then compared the result and they were identical. So I'm tempted to say our .fw format is fine. An SD Card issue (driver or card) perhaps?

1. https://github.com/pebri86/esplay-base-firmware/tree/master/tools/mkfw

monsieurPiGG, 2023-02-01:
> An SD Card issue (driver or card) perhaps?

Since it has problems launching emulators from roms stored in the supplied sd card, I'm also tempted to point the sd card.
@Cralex does it flash .fw from bootloader since you've changed your sd ?

Cralex, 2023-02-02:
>  An SD Card issue (driver or card) perhaps?

> does it flash .fw from bootloader since you've changed your sd ?

Well, I _thought_ I'd tried flashing from the bootloader after I switched cards, or maybe I just did a backup of what was already on the card without checking the validity of my build. I flashed the bootloader and copied over a fresh build of retro-go... It works. Icon appears, it flashes, it boots. Oy! Sorry all for the wild goose chase. In addition to simply working, I also tested the headphone audio (can people solder a speaker to this?), the wifi again, and the battery. Battery needs more tweaking, but everything else works fine. 

I'm taking another look at the buttons. I'm using the I2C driver, and I noticed that past a certain point the buttons start to repeat themselves without hitting Menu, L, and R. I'm going to keep trying and see if maybe the new buttons can be mapped by passing by the first repeats. I'll post my results when I have them.

Cralex, 2023-02-02:
Okay, now that the self-inflicted flashing problem is solved, I see no reason not to put out an [alpha release](https://github.com/Cralex/retro-go/releases/tag/esplay-micro-alpha-1) for ESPlay Micro owners to test out. See usage and installation notes in the link. I also made a pull request with all the necessary files to build the firmware in case anyone wants to help me work on it.

nod3011, 2023-02-02:
> Okay, now that the self-inflicted flashing problem is solved, I see no reason not to put out an [alpha release](https://github.com/Cralex/retro-go/releases/tag/esplay-micro-alpha-1) for ESPlay Micro [...]

I've try this alpha version. look like the [select] button and [menu] button is the same. This might need some fix.

monsieurPiGG, 2023-02-02:
Amazing !

Point 3 in the release :
> Turn your unit off and back on again. This should get you access to the bootloader.

It's not necessary. Just hold menu while booting and the device boots the boot loader.

> look like the [select] button and [menu] button is the same. This might need some fix.

@nod3011, it's normal and intended by @Cralex in order to make it playable.

> can people solder a speaker to this?

I have a speaker soldered to it and it works. Sound glitch's with snes but I guess it's a known issue.

It's amazing ! I'm impressed by the snes performance, still not 100% speed but good enough to play some games (with patience !).

Cralex, 2023-03-10:
After a nice long rest, I'm back to look at the buttons, this time with moderately more research. There's a diagram of the unit [here](https://github.com/pebri86/esplay_micro_hardware/blob/master/docs/new_esplay.pdf). Note the PCF8574 GPIO extender chip that appears to be used to wire some of the buttons. There's a little guide on how to use the thing [here](https://www.instructables.com/PCF8574-GPIO-Extender-With-Arduino-and-NodeMCU/). 

As little experience as I have with ESP32, I have even less with Arduino. But it seems as though someone wanting to use this with Arduino would need to add some libraries. Does it look like Retro-Go would need to do this as well? I glanced over the stock firmware and it doesn't seem as though they do. There also seems to be a section in the guide that hints that the addresses for the added pins should start at 0x20, but I don't know how to apply the knowledge. This is all rather above my head, but I really want to improve the port for this device.

ducalex, 2023-03-11:
In retro-go you can use rg_i2c_* functions to talk to the chip. The equivalent to what they do [here](https://github.com/pebri86/esplay-retro-emulation/blob/master/esplay-sdk/esplay-components/esplay-hal/src/gamepad.c#L33-L63) would be `rg_i2c_read_byte(0x20, -1)`.

So essentially in `rg_input_init` you'd need to call `rg_i2c_init();` and then in `gamepad_read` you'd do `uint8_t buttons = rg_i2c_read_byte(0x20, -1);`.

Cralex, 2023-10-28:
I've been working for a few hours on the button mapping and could use a break and/or another set of (human) eyes. 
Here's what I added to rg_input.c: 
Under `static inline uint32_t gamepad_read(void)`
```
#elif RG_GAMEPAD_DRIVER == 7  // ESPlay Micro V2
    uint8_t buttons = rg_i2c_read_byte(0x20, -1);
        for (size_t i = 0; i < keymap_size; ++i)
    {
        if (buttons & (1 << i))
            state |= keymap[i].key;
```
Under `void rg_input_init(void)`
```
#elif RG_GAMEPAD_DRIVER == 7 // ESPlay Micro V2
    const char *driver = "I2C_BYTE";
    rg_i2c_init();
```
It builds, but always boots to Recovery Mode regardless of how I've set the button mapping, so either I'm not expressing the buttons correctly or my additions to rg_input.c are inherently buggy. I eventually stumbled on a repo containing some simple tests for the unit [here](https://github.com/Makerfabs/ESP32_PLAY), including a helpful [button test](https://github.com/Makerfabs/ESP32_PLAY/tree/main/example/button_test) for reference. The repo mentions that only the L, R and MENU keys are directly connected to the IO pins, while the other keys are connected through the PCF8574.

But yeah, need sleep. I'll think on this more tomorrow.

ducalex, 2023-10-29:
I think it should look more like:

````c
    uint8_t buttons = rg_i2c_read_byte(0x20, -1);
        for (size_t i = 0; i < keymap_size; ++i)
    {
        if ((buttons & keymap[i].src) == keymap[i].src)
            state |= keymap[i].key;
````

RG_GAMEPAD_MAP is a table where each entry has:
- .src: This is the bitmask that represents the button in hardware
- .key: The retro-go key to which this bit maps

You could ignore the RG_GAMEPAD_MAP until you get things working, it can get a bit confusing. Another reason to ignore RG_GAMEPAD_MAP for now is that it isn't super flexible, it can't express two sources of inputs (gpio+i2c extender) nicely. We'll have to fix that.

Just to get things working you could just do something like that:
````c
    uint8_t buttons = rg_i2c_read_byte(0x20, -1);
    if (buttons & (1 << 0)) 
        state |= RG_KEY_UP;
    if (buttons & (1 << 1)) 
        state |= RG_KEY_DOWN;
    if (!gpio_get_level(MENU GPIO))
        state |= RG_KEY_MENU;
    if (!gpio_get_level(L GPIO))
        state |= RG_KEY_L;
````

Takiiiiiiii, 2024-03-26:
Not exactly sure why, but this doesn't seem to work with ESPLAY Micro V1 (along with other boards based on it). Have any of you been able to get it to work on a V1 board?

V1 - ESP32 WROVER, Dual-core processor with Integrated 4MB Flash + 4MB PSRAM
V2 - ESP32 WROVER, Dual-core processor with Integrated 16MB Flash + 8MB PSRAM

The two V1 boards that I have just loop: 

`rst:0x10 (RTCWDT_RTC_RESET),boot:0x3b (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57
`

Solved by reflashing the bootloader with JP1 connected to ground. 

@ducalex It might be worth renaming the release to V2 unless someone with a V1 confirms it works.

ducalex, 2024-03-26:
I don't have any of those devices so we'll wait for @Cralex to see if he knows what's going on before I make any change.

But that sounds like maybe retro-go doesn't fit in the flash and, for whatever reason, the firmware/bootloader tries to flash it anyway? Based on that log, it would end at 0x410000 (1.40 is 3.75MB). Which is likely past the 4MB flash.

We have several options to make retro-go smaller:
- Disable networking: maybe 400KB saved
- Remove SNES: 512KB saved
- Remove Genesis: 960KB saved

We could also instruct users to flash a (non-reduced) .img version instead. But it's perhaps less convenient.

Let me know what you all think is the best route to take and I'll see how I can integrate those needs in my next release.

Takiiiiiiii, 2024-03-26:
Darn. I might swap out my Wrover module for an 8MB one. I was looking forward to testing those two systems. Losing networking isn't a huge issue.

Cralex, 2024-03-26:
Yeah, I also don't have a V1 board, and I'm guessing (I don't know off the top of my head. Maybe the MRGC GameBox Mini?) that all the ports I've been working on so far have had enough flash storage to install a full build of Retro-Go using the bootloader's flashing feature.

Renaming the port to specify a V2 target is certainly one option, although I'm curious to know how well it runs Retro-Go. Were you able to flash a Retro-Go .img on your V1 directly, without using the bootloader? Like Ducalex said, building it without some of the extra components (networking, SNES, Genesis) should hopefully result in an image small enough to flash from the bootloader.

I am rather spoiled by the networking features, but I can compile a copy without it if you would like to test it out.

ducalex, 2024-03-27:
> Darn. I might swap out my Wrover module for an 8MB one. I was looking forward to testing those two systems. Losing networking isn't a huge issue.

If I'm being honest you're not missing much, neither is running very well. 

Attached build is as small as I could pack things, can you test it to confirm it's just a size issue? Nothing was cut, I've just moved snes9x to retro-core.

[retro-go_1.41-dirty_esplay-micro.zip](https://github.com/ducalex/retro-go/files/14765892/retro-go_1.41-dirty_esplay-micro.zip)

Takiiiiiiii, 2024-03-27:
> > Darn. I might swap out my Wrover module for an 8MB one. I was looking forward to testing those two systems. Losing networking isn't a huge issue.
> 
> If I'm being honest you're not missing much, neither is running very well.
> 
> Attached build is as small as I could pack things, can you test it to confirm it's just a size issue? Nothing was cut, I've just moved snes9x to retro-core.
> 
> [retro-go_1.41-dirty_esplay-micro.zip](https://github.com/ducalex/retro-go/files/14765892/retro-go_1.41-dirty_esplay-micro.zip)
> 
> If it still doesn't work for other reasons then I'm ok with making it clear retro-go is v2 only!

I'll try and update you with the results. The strange thing is this person was able to get retro-go working on a 4MB V1 derivative based on their serial output: https://github.com/ducalex/retro-go/issues/91#issuecomment-1739379236

Takiiiiiiii, 2024-03-27:
> Were you able to flash a Retro-Go .img on your V1 directly, without using the bootloader?

I am unsure of how to do this. Do you use Flash_Download_Tool/esptool?

ducalex, 2024-03-27:
> To other users, be aware that this may not work for V1 derivative boards. The one that I have cannot boot this dirty build, but a real ESPLAY Micro can.

I'm not sure this is a retro-go issue (at least based on the logs you provided). In the first log, boot fails because the bootloader expected to find a 16MB flash chip but found a 4MB instead.

This is likely because the bootloader was configured for 16MB flash at build time, and you didn't override that during flashing.

If flashed correctly, this line:
> I (50) boot: SPI Flash Size : 16MB

should become:
> I (50) boot: SPI Flash Size : 4MB

I think for esptool you just have to use `--flash_size 4MB`. In the gui tool there's a size selector and also a checkbox to uncheck I think.