ducalex/retro-go #59: Help Needed: Multiple Errors on MacOS Venture

32teeth, 2022-11-04:
**Describe the bug**
* Upgraded to Ventura
* Using ESP IDF 4.1
* Running `python3 rg_tool.py build launcher` produces multiple errors in compiling

```
# example errors (only copy pasted a few to illustrate)
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/emulators.c.obj 

error: implicit declaration of function 'rg_system_get_path'; did you mean 'rg_system_get_app'? [-Werror=implicit-function-declaration]
     crc_cache_path = rg_system_get_path(NULL, RG_PATH_CACHE_FILE, "crc32.bin");

error: 'tab_t' {aka 'struct tab_s'} has no member named 'is_empty'
     tab->is_empty = items_count == 0;

error: 'KEY_PRESS_B' undeclared (first use in this function)
     else if (event == KEY_PRESS_B)

error: too many arguments to function 'gui_save_config'
```

32teeth, 2022-11-04:
Issue might be MacOS Ventura based
```
os.path.join(os.getcwd(), app))
```

ducalex, 2022-11-04:
I don't think that's related to your OS. I think you're using a mishmash of different versions of retro-go and the launcher.

Can't really help you if that's what you want to do, you'll have to adapt your code. Or stick with older retro-go (If I had to guess, KEY_PRESS_B was renamed at the beginning of 2022).

32teeth, 2022-11-04:
Lol, okay I'll check. It's a rebase from current commit

32teeth, 2022-11-04:
> Closing, 

somehow i ended up on an old commit that showed up as current even after fetch in MacOS Ventura update

closed 2022-11-04 by 32teeth