ducalex/retro-go #65: GBC - Pokemon Trading Card Game 1 and 2 fail to run

Cralex, 2022-11-13:
**Describe the bug**
On 1.35.1, I can't get Pokemon Trading Card Game (CRC32 81069D53) and Pokemon Trading Card Game 2 English Patched (CRC32 D880DFC1) to run. If starting from a new game, both boot to a black screen, without locking up the system or triggering a crash. If starting from a previous save state, they'll run until the game needs to load something and then softlock. This happens on both the public release on OdroidGo and on my Game Box Mini port in development.

Cralex, 2022-11-15:
Worth noting that my development environment matches the code at the time of the last release, so any later changes to the GBC emulator isn't reflected in my copy of the Game Box Mini firmware.

ducalex, 2022-11-15:
What is the last retro-go version you remember working correctly with this game? (if any)

Cralex, 2022-11-15:
I'll try some old versions and get back to you. They definitely worked before, since I have some save preview images of the games running properly.

ducalex, 2022-11-15:
Thanks for tracking down the exact versions! Fixed in 76a4be91ad186a341acd59e1f3acb17667728c63.

closed 2022-11-15 by ducalex