ducalex/retro-go #67: Not connecting to WiFi

jobitjoseph, 2022-11-23:
**Describe the bug**
A clear and concise description of what the bug is.
Not connecting to WiFi. Created the wifi.json file and added the wifi SSID and password, But not connecting WiFi.
Device ESP32 wrover 16Mb
**To Reproduce**
Steps to reproduce the behavior:
1. Created the wifi.json file
2. added the wifi SSID and password
3. But not connecting WiFi. In system menu network status not configured

mikeinkl1, 2022-11-24:
Did you compile it with wifi enabled?

ducalex, 2024-05-03:
It's been almost two years with no response, I'm going to assume that your problem has been fixed or that you've moved to another project.

Feel free to re-open this issue if you need further help.

closed 2024-05-03 by ducalex