ducalex/retro-go #70: Can retro-go be used for education?

YobeZhou, 2022-11-30:
Hi ducalex
We are an embedded education and training company located in Shenzhen, China. www.100ask.net is our official website homepage. Recently, in order to better use esp32 to carry out related project courses, we saw your game console comprehensive project retro-go,
retro-go is a great and amazing project, very impressive after experimenting! We want to let more people know about him through education. Our plan is:

1. We will maintain and support our DShanMCU-Mio (Mio) (ESP32S3-R8 16MB Flash) development board game console version for a long time
2. Add more features to retro-go, such as multiple input devices can be used at the same time, only need to configure in the settings, without recompiling and burning
3. Introduce comprehensive documentation and tutorials, so that more people can better understand and use retro-go
4. There will also be systematic video tutorials, so that more people can learn about esp32 and retro-go
5. Release the LVGL GUI version, this is mainly for teaching, if you want we can merge branches at any time!

Our main contributions to LVGL:
https://github.com/lvgl/lvgl/pull/3601 
https://github.com/lvgl/lvgl/pull/3408 

Some demonstrations of our retro-go-based results:
https://youtu.be/Z7cNesAQYAo 
https://youtu.be/hTvfZPf3qiw 

We hope that after getting your approval and authorization suggestions, we will launch the PCB integrated board and the above 1-3 points in the next step. Before that, I would like to know your opinions and suggestions in order to better carry out this project. Very Looking forward to your reply!

Wish everything is well! YobeZhou

ducalex, 2022-11-30:
Hey! 

Yes you may use retro-go for any purpose you wish, including education and commercial. Parts of retro-go are GPLv2 so it's best to consider the whole project as such. The main condition it puts on you is to provide source if someone asks for it but you are not required to keep the source publicly on github if you don't want to.

Having the multiple input support configurable at runtime is something I wanted to do for a long time so I'm definitely interested in accepting a PR when/if you want to share!

Thanks for showing your project, it makes me happy to see retro-go used like that :).

YobeZhou, 2022-12-01:
> Hey!
>
> Yes you may use retro-go for any purpose you wish, including education and commercial. Parts of retro-go are GPLv2 so it's best to consider the whole project as such. The main condition it [...]

Thank you very much for your reply and support! We are happy to disclose all details and will share our achievements to improve retro-go!

YobeZhou, 2023-03-30:
@ducalex  Hi! I can't wait to share our progress with you!

Version 1: [https://youtu.be/hTvfZPf3qiw](https://youtu.be/hTvfZPf3qiw)
Version 2: [https://youtu.be/Z7cNesAQYAo](https://youtu.be/Z7cNesAQYAo)
3rd version: [https://youtu.be/PSaZW6c0juk](https://youtu.be/PSaZW6c0juk)

I am currently preparing the 3d printed shell for the third version, after which it will be sold as an educational development board, and we are also preparing a detailed tutorial!

ducalex, 2023-03-31:
> 3rd version: https://youtu.be/PSaZW6c0juk

It's looking good and thanks for uploading in 4K, it makes it easy to analyze the circuitry :).

Are you using esp-idf 5.0? I think we're still missing some fixes to build correctly with it but it would be important for full S3 support...

YobeZhou, 2023-04-03:
Currently using esp-idf 4.4.2, we are trying to use esp-idf 5.0. Retro-Yao-Mio is not intended for practical product purposes, it is positioned as a learning function board, but I will try to make it more perfect. I will submit a PR to this warehouse if I have any new progress in the future.

PS: We are very happy to share our achievements with you and would like to send you a set of Retro-Yao-Mio hardware, of course, all the costs will be borne by us.  If possible, please send your receipt information to this email address: team100ask@outlook.com

Looking forward to your reply:)

closed 2023-05-04 by YobeZhou