ducalex/retro-go #71: 1.36.2 SNES Still Crashes when switching mapping

Cralex, 2022-12-08:
Just installed the public release 1.36.2 on my Odroid Go, and the crash still happens when trying to open the menu to switch controls in SNES. 
[crash.log](https://github.com/ducalex/retro-go/files/10182378/crash.log)

Cralex, 2022-12-08:
I let them know on the boards.

It's an improvement, but it needs one more tweak to be ready. It is possible to access the menu now and flip between the options, but any attempt to close the menu just resets it with the cursor back at the top. I am guessing that the control setting is only saved when the menu is closed, so shutting off the unit does not preserve any changes you make.

Disclaimer: I left my ODG at home when I went to work, so I am only able to test with my GBM. (Thanks to the new access point option. :) ) But I'm guessing the behavior would be consistent.

ducalex, 2022-12-09:
Alright that's a bummer but thanks for letting them know! It seems someone made a PR to fix that but I'll wait for when I'm able to test it myself this time :p.

jdcanado, 2022-12-09:
Hi @ducalex, i changed my PR to FIX in the method on the rg_gui.c file! I tested in my board and working!!! Thanks!

ducalex, 2022-12-11:
Merged :)

closed 2022-12-11 by ducalex