ducalex/retro-go #74: Launcher - Incorrect Date Displayed in UI

Cralex, 2022-12-26:
**Describe the bug**
Some time ago, I noticed that both of my Retro-Go devices were showing a date of December 14th in the about section of the menu. Connecting to WiFi sets the time as normal, but the date isn't updating. I finally got around to looking in the web file browser, and the correct date seems to be appearing there.

**Screenshots**
[image]

[image]

ddrsoul, 2022-12-26:
I can't check right now, but I believe that this is date when firmware was create.

Cralex, 2022-12-26:
Oh, that explains it. I am looking in the "about" section after all.

closed 2022-12-26 by Cralex