ducalex/retro-go #76: SNES - APU causes softlock after defeating Dark World boss in Zelda Link to the Past

Cralex, 2023-01-11:
**Describe the bug**
After beating a boss in the Dark World in Zelda Link to the Past for SNES, unexpected behavior occurs when grabbing the crystal that appears. If the APU is on, the game locks up. If the APU is off, gameplay continues, albeit skipping a delay that occurs normally.

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on the APU if it isn't on already.
2. Load the included save state, which takes place immediately after beating a boss in the Dark World. 
3. Collect the Heart Container and Crystal that appear.
4. If the APU is on, the game will lock up until it is turned off.
5. If the APU is off, or has been off at any point since loading the save state, the fanfare delay mentioned below is skipped, leading directly into the cutscene and regular gameplay afterwards.

**Expected behavior**
When the crystal is collected, it's supposed to pause gameplay for about ten seconds to play a musical fanfare, then a cutscene begins where the maiden in the crystal starts talking. 

**Additional context**
Tested on OdroidGo and GameBox Mini with both the vanilla game (777AAC2F) and the popular Redux hack version 10.1 (DC40F93E).

[Legend of Zelda, The - A Link to the Past save.zip](https://github.com/ducalex/retro-go/files/10396265/Legend.of.Zelda.The.-.A.Link.to.the.Past.save.zip)

ducalex, 2023-01-12:
APU toggling does cause issues in many games, especially when combined with save states. I've now removed the APU toggle and I explain why in e9646a9099dea0257a4043f8ca9d7797acd6f181.

I don't know if that will fix your issue but to troubleshoot it correctly you'd have to replay the game from a fresh start using the latest commit :( .

Cralex, 2023-01-12:
The issue occurs when APU is on while the save state is being loaded, and is left on until a crystal is collected. I normally play with it off in order to maximize performance, but at least a time or two I've accidentally had it on for some reason and worried that I'd have to redo a dungeon until I remember to check the APU.

Would Reset->Soft Reset be enough for me to test the latest commit? Doing this resets the game while preserving in-game save data.

ducalex, 2023-01-13:
> Would Reset->Soft Reset be enough for me to test the latest commit? Doing this resets the game while preserving in-game save data.

I think that's a good idea, the odds of SRAM being corrupted by bogus APU state seems low to me.

But I'm not 100% sure the snes emulator implements soft reset.

Cralex, 2023-01-17:
I built the dev branch the other day and tried the reset thing. It skipped the fanfare and I was no longer able to replicate the softlock no matter what I did. ...but just to be sure, I'm doing a new playthrough without relying on any save states from the stable retro-go version. I'll report back when I get around to beating a dark world boss.

ducalex, 2023-01-24:
Thanks for the effort!

Cralex, 2023-01-28:
Okay, looks good. I can't reproduce the softlock using the beta firmware.

closed 2023-01-28 by Cralex