ducalex/retro-go #80: .img Building Showing this error

TheTest21, 2023-02-18:
So, im trying to port doom to the m5stickC Plus and while building the .img for the flashing I got this error

 **sudo python rg_tool.py build-fw launcher prboom-go 
[sudo] password for thetest21:             
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
IDF_PATH is not defined. Are you running inside esp-idf environment?**

I have install esp-idf and requirements and even just building it as it is doesn't work

ducalex, 2023-02-19:
You need to be setup the esp-idf environment before building. If you already did, then it's likely that using sudo is your issue (you don't need sudo).

Here's the doc: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html#step-4-set-up-the-environment-variables

ducalex, 2024-05-03:
Hello! It's been over a year so I'm going to assume you've fixed your problem or moved to another project.

Feel free to re-open this issue if you still need help.

closed 2024-05-03 by ducalex