ducalex/retro-go #81: Mortal Kombat 3 Europe rom on GB crash on level loading

philectro, 2023-03-14:
When loading a level most of the time the game crash there.

1 run the game
2 select and player and a level of difficulty
wait for the time juste after the line of symbols(game code)
and voila, crash

**Expected behavior**
no crash on level loading

If i can help in a way i will be happy to.

Good day and thanks for all the things you do for the community.

ducalex, 2023-03-16:
Can you tell me your current retro-go version and do you know if the game worked on past versions?

philectro, 2023-03-16:
Hi Ducalex, thanks for your time.
Yes this crash existed already for months or years on the portage of gnuboy on esp32.
I have installed the last build from here just before testing on my odroid-go

ducalex, 2023-03-20:
Alright, it's not crashing for me but I do get a white screen and nothing happens. 

I've investigated and it's the same issue as in #40, so it's not an easy fix unfortunately :( .

ducalex, 2023-08-21:
I believe I have fixed it but I want to make sure I didn't break anything else before releasing.

Please try https://github.com/ducalex/retro-go/actions/runs/5928759465 and report any issues, especially in other games!

philectro, 2023-09-13:
I ducalex i read your update weeks ago but still didn't test it. I will test it soon and come back.
Thanks for your effort!

philectro, 2023-09-13:
Ok, i tried several roms and obviously the mk3 one.
All games tested work but this artifact doesn't solve this issue.