ducalex/retro-go #82: Can't figure out the OGO header pinout for external DAC

ZirconiaCubed3, 2023-03-28:
I can't figure out which pins in the Odroid Go header are supposed to be connected to which contacts on my audio jack. I looked at the official pinout to eliminate the ground and power pins. I also did some testing and process of elimination, and while I did, I found that a few of the pins outputted very staticy (is that a real word?) version of the desired sound. Is there a definitive 'pin that it outputs audio on' or do I have to get the special DAC hat instead?

ddrsoul, 2023-03-29:
Hello!
I made DAC hat to connect headphones thanks to this topic on odroid forum:
https://forum.odroid.com/viewtopic.php?t=31853

you should use "headphones only" variant with 3.3V power supply.

ODROID-GO -> DAC
---------    ---
 1 GND       GND
 2 
 3 IO12      LRCLK
 4 IO15      OUT
 5 IO4       SCLK
 6 P3V3      P5V0 (for headphone use only)

Those 1....6 pin numbers are according to official odroid documentation:
[image]
and are located on the upper side of odroid go

If I remember correctly - this pinout above was OK to use.
I had similar PCM5102A board but it was designed to have external 5VDC supply, so I shorted one of resistors that divided voltage from 5 VDC to 3.3VDC to the "un-mute" XSMT signal.
That is what I got:
https://imgur.com/IT4QYg1

ext DAC gives very nice sound (much better than that from the internal speaker). odroid turns into good mp3 player if you connect your device to separate active speaker (such as JBL Charge or Go)

ducalex, 2023-08-01:
I've updated the README to have a link to the best DAC mod resource I could find (mentioned in ddrsoul's link) and also embedded the PINOUT table. Feel free to re-open if questions remain.

closed 2023-08-01 by ducalex