ducalex/retro-go #83: Crash after updating to Retro-Go 1.38 or 1.37.47

explit7, 2023-03-31:
After Installation of 1.38 (or 1.37.47 FW on my Odroid-Go) the system is crashed on stratup.
Installing back 1.37 FW - working without any Problems.

**To Reproduce**
Download 1.38 FW  (retro-go_1.38_odroid-go.fw) or 1.37.47 FW (retro-go_gd334e_odroid-go.fw).
Update Odroid-Go to the 1.38 or 1.37.47 FW
Crash on StartUp

**Expected behavior**
Normal Start of RetroGo
[crash.log](https://github.com/ducalex/retro-go/files/11117012/crash.log)

**Additional context**
Add any other context about the problem here.

explit7, 2023-03-31:
Can you upload new .fw files with the fix please ?
I have Problems to setup ESP32 environment on Debian-Testing.
Thanks.

ducalex, 2023-03-31:
It's not tested yet so I won't upload new .fw, but if you want to try it anyway they are available from the build bot.

Go to https://github.com/ducalex/retro-go/actions , click the commit you want to try and scroll down to *Artifacts*.

explit7, 2023-03-31:
Thanks, i tested the dev build and it is working. Doesn't crash anymore.

ducalex, 2023-03-31:
Thanks for testing! The crash was probably caused because you have a file without extension in one of your roms folder and a recent change didn't account for that.

I'll release 1.38.1 in a few days with the fix (after I review the rest of our code for similar issues).

closed 2023-03-31 by ducalex