ducalex/retro-go #85: Error with esp_http_client_read on esp-idf 4.4

32teeth, 2023-04-08:
**Describe the bug**

Came across an issue on attempting to build after full clean. This seems to be new

Command
`python3 rg_tool.py --target retro-esp32 build launcher `

Error
```
error: implicit declaration of function 'esp_http_client_read_response'; did you mean 'esp_http_client_read'? [-Werror=implicit-function-declaration]
     int len = esp_http_client_read_response(req->client, buffer, buffer_len);
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
               esp_http_client_read
cc1: some warnings being treated as errors
[602/656] Building C object esp-idf/retro-go/CMakeFiles/__idf_retro-go.dir/rg_display.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

ducalex, 2023-04-09:
I'm not sure what could be the problem, it seems to be available in esp-idf 4.4: https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32/api-reference/protocols/esp_http_client.html#_CPPv429esp_http_client_read_response24esp_http_client_handle_tPci

Since the compiler recommends esp_http_client_read you could try it. Replace the error line by:
````c
    size_t len = 0;
    while (len < buffer_len) {
        int data_read = esp_http_client_read(req->client, buffer + len, buffer_len - len);
        if (data_read <= 0)
            break;
        len += data_read;
    }
````

But I don't see why that would work and esp_http_client_read_response doesn't.

32teeth, 2023-04-10:
I will try again with a clean 4.4, maybe my paths where conflicting

32teeth, 2023-04-21:
> Resolved

Issue was with my local paths