ducalex/retro-go #87: [Feature] - ST7701S SPI LCD Driver

32teeth, 2023-04-24:
Hey @ducalex 

Wondering if we could implement support for [4" 480x480 Wide Temperature IPS TFT Display with ST7701S driver IC](https://www.winstar.com.tw/products/tft-lcd/ips-tft/st7701s.html), looking through the white papers doesn't seem anything blocking.

* Can the ESP32 support 480x480
* Any side effects?

I have ordered a kit from [AliExpress](https://www.aliexpress.us/item/3256803783122458.html?gatewayAdapt=glo2usa&_randl_shipto=US) be it through the 3.4" version

Looking for your guidance / feedback / thoughts?

mikeinkl1, 2023-04-28:
Cant see why it should not work, but 480x480 spi will be slow. The datasheet doesnt say what the max clock frequency is but  see https://github.com/juj/fbcp-ili9341 for whats possible in terms of fps.

32teeth, 2023-04-28:
I'll wait for my AliExpress order to come in (i got the one with the breakout board) and i'll see what i can do

ducalex, 2023-05-01:
It will use more internal memory and it will definitely be slower to blit but if it can run at 80Mhz it will help.

The scaling code will probably be possible to optimize a lot too, offsetting some of the speed impact. Because you can fit more integer scales in 480x480 (eg gameboy could do 3x and leave very empty space) so you could skip the bilinear filtering and temporary buffers and floats math and whatnot.