ducalex/retro-go #88: Files with too long a name do not show in launcher and/or are corrupted in the web UI.

netbusiness, 2023-05-25:
Don't know when this issue started happening. I had to revert to 1.33 before the files started appearing in the launcher again.

Examples:
"Arle no Bouken - Mahou no Jewel (Japan) (SGB Enhanced) (GB Compatible)[tr en].gbc"
"Pokemon - Yellow Version - Special Pikachu Edition (USA, Europe) (GBC,SGB Enhanced).gb"
"Legend of Zelda, The - Link's Awakening DX (USA, Europe) (SGB Enhanced) (GB Compatible).gbc"
"Meitantei Conan - Kigantou Hihou Densetsu (Japan) (SGB Enhanced) (GB Compatible)[tr en].gbc"

Problem does not seem to be limited to GB/GBC games.

ducalex, 2023-05-25:
File names that are too long will now be skipped instead of being truncated (which I agree was weird). I've also increased the size limit from 74 to 90 which is about as high as we can reasonably go right now.

We can keep this issue open until I or someone else come back to add proper support for arbitrary length filenames (hopefully with no performance impact).

ducalex, 2024-02-27:
This should be fully fixed as of 3427ade981a5fd7e6ce32aeba6f26c5d19439382 and in the current 1.40 release.

Though names displayed in screen will still be truncated, there's no good way to fix that.

Feel free to re-open if still relevant!