ducalex/retro-go #90: Numerical Battery Readout in Debug

Cralex, 2023-09-10:
As far as I'm aware, it's possible to see the battery level in two places: over serial when connected to power and the graphical icon. I'd like to see it added to the Debug menu as well if possible. I've got two (sluggish) ports in progress and I feel that it would be helpful to be able to watch a numerical battery estimate go down over time in the menu somewhere, just to help me know if I've got the battery gauge configured correctly. (Over serial, there's a good chance it would be fully charged anyway, or I might not have it set up right and it'd show a false 0 or 100 that never changes.)

ducalex, 2023-09-11:
Makes sense to me and I'm guessing it's as simple as calling `rg_input_read_battery` in `rg_gui_debug_menu`. 

Do you want to do it? Or I can do it if you prefer to skip the whole PR process but I won't be able to test it right now.

Kelly97582044, 2023-10-03:
in rg_input.c , Line 40- line 43 , so the data[4] is battery value, but, when you connect usb cable , data[4] is always 100. if you want to see the actual value, peel off the usb cable, cut the red wire(USB 5V).

Kelly97582044, 2023-10-03:
in rg_input.c , Line 40- line 43 , so the data[4] is battery value, but, when you connect usb cable , data[4] is always 100. give me your email, i will send you the sch and pcb file, and tell you how to get the battery value through usb cable.

Cralex, 2023-10-27:
Added in my pull request. This will cause the percentage and voltage to appear in the debug menu. The menu doesn't refresh over time, so you'd need to check back if you're looking for something to change.
[image]
I had some help from AI to put this in, so it could probably be nicer with a human touch, but it's a start and it works.

ducalex, 2023-10-29:
Yeah live update is possible but pretty complicated with our dialog system :(

Looking good, I left a comment on the PR just to double check something and then we can merge it.

ducalex, 2024-01-17:
This was completed in PR#92.

closed 2024-01-17 by ducalex