ducalex/retro-go #93: MRGC and Gamebox battery voltage

Kelly97582044, 2023-11-05:
Hi ducalex：

  MRGC battery voltage  V = 3 * data[4] * 3.3 / 256
there are two 14500 battery, so one 14500 battery voltage is : V / 2

//==============================================
game box(mini GBA) battery voltage  V = 256 * 3.3 / ( 2 * data[4] ) =  128 * 3.3 /  data[4] 
if  data[4]==255 ,the battery is charging

ducalex, 2023-11-06:
If I understand you correctly you're suggesting we put this in our mrgc-g32/config.h:

````c
#define RG_BATTERY_CALC_VOLTAGE(raw) (3 * (raw) * 3.3f / 256)
````

?

Kelly97582044, 2023-11-07:
yes, if the voltage value is correct. my english is not good ,can not write so many words,but i think you can understand.

ducalex, 2024-02-29:
I've added your changes in 53717b767e27218951f8dcd776e0aa759e3eb071. 

Thanks!

closed 2024-02-29 by ducalex