ducalex/retro-go #95: Does this filmware work with esp32-s3?

hello-world12, 2023-12-22:
I am having issues trying to get this filmware to function on my esp32-s3 for some time. Has anyone made this filmware work with esp32-s3?

ducalex, 2024-01-01:
It can be built for esp32-s3 (we have a target called ESPLAY-S3 and it used to build and work fine to my knowledge) but I think the current tree doesn't build with esp-idf 5.0+ so that might be a problem. 

The changes needed are probably fairly minor, though.

queenkjuul, 2024-01-12:
I have been thus far unable to build the esplay-s3 target on ubuntu 23.10, from the main branch or the 1.38.1 tag. 

I will keep playing with it, but I'm pretty new to C and ESP32 programming so I'm not really sure I'm understanding what's wrong

My command line is `./rg_tool.py --target esplay-s3 build-img` so I'm not sure why it's complaining about odroid-go

```
/home/qkj/esp32doom/retro-go/components/retro-go/rg_storage.c: In function 'rg_storage_init':
/home/qkj/esp32doom/retro-go/components/retro-go/targets/odroid-go/config.h:79:37: error: 'GPIO_NUM_22' undeclared (first use in this function); did you mean 'GPIO_NUM_42'?
 #define RG_GPIO_SDSPI_CS            GPIO_NUM_22
                                     ^~~~~~~~~~~
/home/qkj/esp32doom/retro-go/components/retro-go/rg_storage.c:84:27: note: in expansion of macro 'RG_GPIO_SDSPI_CS'
     slot_config.gpio_cs = RG_GPIO_SDSPI_CS;
                           ^~~~~~~~~~~~~~~~
/home/qkj/esp32doom/retro-go/components/retro-go/targets/odroid-go/config.h:79:37: note: each undeclared identifier is reported only once for each function it appears in
 #define RG_GPIO_SDSPI_CS            GPIO_NUM_22
                                     ^~~~~~~~~~~
/home/qkj/esp32doom/retro-go/components/retro-go/rg_storage.c:84:27: note: in expansion of macro 'RG_GPIO_SDSPI_CS'
     slot_config.gpio_cs = RG_GPIO_SDSPI_CS;
                           ^~~~~~~~~~~~~~~~
/home/qkj/esp32doom/retro-go/components/retro-go/targets/odroid-go/config.h:77:37: error: 'GPIO_NUM_23' undeclared (first use in this function); did you mean 'GPIO_NUM_43'?
 #define RG_GPIO_SDSPI_MOSI          GPIO_NUM_23
                                     ^~~~~~~~~~~
/home/qkj/esp32doom/retro-go/components/retro-go/rg_storage.c:86:24: note: in expansion of macro 'RG_GPIO_SDSPI_MOSI'
         .mosi_io_num = RG_GPIO_SDSPI_MOSI,
                        ^~~~~~~~~~~~~~~~~~
In file included from /home/qkj/esp32doom/retro-go/components/retro-go/config.h:17,
                 from /home/qkj/esp32doom/retro-go/components/retro-go/rg_system.h:12,
                 from /home/qkj/esp32doom/retro-go/components/retro-go/rg_display.c:1:
/home/qkj/esp32doom/retro-go/components/retro-go/rg_display.c: In function 'spi_init':
/home/qkj/esp32doom/retro-go/components/retro-go/targets/odroid-go/config.h:68:37: error: 'GPIO_NUM_23' undeclared (first use in this function); did you mean 'GPIO_NUM_43'?
 #define RG_GPIO_LCD_MOSI            GPIO_NUM_23
                                     ^~~~~~~~~~~
/home/qkj/esp32doom/retro-go/components/retro-go/rg_display.c:142:24: note: in expansion of macro 'RG_GPIO_LCD_MOSI'
         .mosi_io_num = RG_GPIO_LCD_MOSI,
                        ^~~~~~~~~~~~~~~~
/home/qkj/esp32doom/retro-go/components/retro-go/targets/odroid-go/config.h:68:37: note: each undeclared identifier is reported only once for each function it appears in
 #define RG_GPIO_LCD_MOSI            GPIO_NUM_23
                                     ^~~~~~~~~~~
/home/qkj/esp32doom/retro-go/components/retro-go/rg_display.c:142:24: note: in expansion of macro 'RG_GPIO_LCD_MOSI'
         .mosi_io_num = RG_GPIO_LCD_MOSI,
                        ^~~~~~~~~~~~~~~~
/home/qkj/esp32doom/retro-go/components/retro-go/rg_audio.c: In function 'rg_audio_init':
/home/qkj/esp32doom/retro-go/components/retro-go/rg_audio.c:97:53: error: 'I2S_MODE_DAC_BUILT_IN' undeclared (first use in this function); did you mean 'I2S_MODE_MASTER'?
             .mode = I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN,
                                                     ^~~~~~~~~~~~~~~~~~~~~
                                                     I2S_MODE_MASTER
/home/qkj/esp32doom/retro-go/components/retro-go/rg_audio.c:97:53: note: each undeclared identifier is reported only once for each function it appears in
make[3]: *** [esp-idf/retro-go/CMakeFiles/__idf_retro-go.dir/build.make:188: esp-idf/retro-go/CMakeFiles/__idf_retro-go.dir/rg_storage.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
/home/qkj/esp32doom/retro-go/components/retro-go/rg_audio.c:107:19: error: implicit declaration of function 'i2s_set_dac_mode'; did you mean 'i2s_set_clk'? [-Werror=implicit-function-declaration]
             ret = i2s_set_dac_mode(RG_AUDIO_USE_INT_DAC);
                   ^~~~~~~~~~~~~~~~
                   i2s_set_clk
make[3]: *** [esp-idf/retro-go/CMakeFiles/__idf_retro-go.dir/build.make:90: esp-idf/retro-go/CMakeFiles/__idf_retro-go.dir/rg_display.c.obj] Error 1
/home/qkj/esp32doom/retro-go/components/retro-go/rg_audio.c: In function 'rg_audio_deinit':
/home/qkj/esp32doom/retro-go/components/retro-go/rg_audio.c:193:36: error: 'I2S_DAC_CHANNEL_RIGHT_EN' undeclared (first use in this function); did you mean 'I2S_CHANNEL_FMT_RIGHT_LEFT'?
         if (RG_AUDIO_USE_INT_DAC & I2S_DAC_CHANNEL_RIGHT_EN)
                                    ^~~~~~~~~~~~~~~~~~~~~~~~
                                    I2S_CHANNEL_FMT_RIGHT_LEFT
/home/qkj/esp32doom/retro-go/components/retro-go/rg_audio.c:194:13: error: implicit declaration of function 'dac_output_disable'; did you mean 'esp_intr_disable'? [-Werror=implicit-function-declaration]
             dac_output_disable(DAC_CHANNEL_1);
             ^~~~~~~~~~~~~~~~~~
             esp_intr_disable
/home/qkj/esp32doom/retro-go/components/retro-go/rg_audio.c:194:32: error: 'DAC_CHANNEL_1' undeclared (first use in this function); did you mean 'I2S_CHANNEL_MONO'?
             dac_output_disable(DAC_CHANNEL_1);
[... cut ...]
/home/qkj/esp32doom/retro-go/components/retro-go/rg_audio.c:196:32: error: 'DAC_CHANNEL_2' undeclared (first use in this function); did you mean 'I2S_CHANNEL_MONO'?
             dac_output_disable(DAC_CHANNEL_2);
                                ^~~~~~~~~~~~~
                                I2S_CHANNEL_MONO
/home/qkj/esp32doom/retro-go/components/retro-go/rg_audio.c:197:9: error: implicit declaration of function 'dac_i2s_disable'; did you mean 'esp_intr_disable'? [-Werror=implicit-function-declaration]
         dac_i2s_disable();
         ^~~~~~~~~~~~~~~
         esp_intr_disable
```

queenkjuul, 2024-01-12:
I tried using a docker container, installing esp-idf v4.3 per the esp-idf docs, then building retro-go per these docs. 

I get similar but different errors, this time about undeclared functions called `RTC_SLEEP_PD_*`

ducalex, 2024-01-12:
There was indeed an issue with esplay-s3 not being respected, it's now fixed in the current dev branch.

But to get a successful S3 build you will need esp-idf 5.0 (I think 4.4 maybe had early S3 support too?).

Following those steps in esp-idf 5.0 I get a successful build:

1. Pull my recent changes and switch to the dev branch
2. In `components/retro-go/CMakeLists.txt` near the top you need to comment the first COMPONENT_REQUIRES line and uncomment the second one.
3. Build with `rg_tool.py --target=esplay-s3 --no-networking launcher`

ducalex, 2024-01-24:
We'll track this issue at https://github.com/ducalex/retro-go/issues/98

closed 2024-01-24 by ducalex