ducalex/retro-go #96: Game Boy Advance Emulator

davidmonterocrespo24, 2024-01-04:
If the SNES can be emulated, I think it would also be possible to emulate the game boy advance.
Isn't the emulator for the GBA in development?

ducalex, 2024-05-03:
I've tried (and failed porting GBA emulators to the ESP32. Those that ran were nowhere near full speed. So I gave up. But if someone wants to try the challenge, I can confirm that I would indeed accept to merge GBA support into retro-go!

davidmonterocrespo24, 2024-05-03:
> I've tried (and failed porting GBA emulators to the ESP32. Those that ran were nowhere near full speed. So I gave up. But if someone wants to try the challenge, I can confirm that I would indeed accept to merge GBA support into retro-go!

Hello
do you have the repository of the emulator you tried to make?

ducalex, 2024-05-03:
Unfortunately I deleted all those branches.

But the most successful attempt was with derivatives of https://emulation.gametechwiki.com/index.php/GpSP (I don't remember which one).

I also did get mGBA to run but it was too slow and it took a long time to reduce memory use enough. I wish I had kept it!

davidmonterocrespo24, 2024-05-08:
> Unfortunately I deleted all those branches.
>
> But the most successful attempt was with derivatives of https://emulation.gametechwiki.com/index.php/GpSP (I don't remember which one).
>
> I also did get [...]

It would be good to resume the project. If you create a repository capable of uploading some code and adding more people.
If you give me some basic instructions on how to start the migration, I can manage it.

ducalex, 2024-07-30:
I've made a quick port of gbSP-libretro today: https://github.com/ducalex/retro-go/commits/gba/

It can play games but not very well. I don't know yet if I'll spend more time on it... But it's there, as a reference for anyone who wants it :)

davidmonterocrespo24, 2024-08-02:
> I've made a quick port of gbSP-libretro today: https://github.com/ducalex/retro-go/commits/gba/
> 
> It can play games but not very well. I don't know yet if I'll spend more time on it... But it's there, as a reference for anyone who wants it :)
> 
> Edit: There will be a downloadable build here in a few moments: https://github.com/ducalex/retro-go/actions/workflows/ci.yml?query=branch%3Agba (click top link, then artifacts at bottom)

Hi, great job!! amazing!!
Do you have any way to use a retro-go emulator on windows? How do you test it? Using the esp32 physically?

ducalex, 2024-08-02:
>  Do you have any way to use a retro-go emulator on windows? How do you test it? Using the esp32 physically?

At the moment I only test on real hardware but there is some SDL2 support in retro-go.

In the past I have successfully built retro-go in Cygwin, MSYS2, and even TCC (though that won't work for gbsp). WSL2 would probably work too. No clue about VC++. These days I like to use https://github.com/skeeto/w64devkit/ with upstream SDL2.

But SDL2 support is very likely currently broken and I have no instructions or build script to offer unfortunately.

There's a ticket with some more details: https://github.com/ducalex/retro-go/issues/118

If you can get SDL2 in better shape, PRs welcome :)

ducalex, 2025-01-26:
@the-star-sea Development on the game and watch is quite different from the esp32 and I do not have a lot of experience with it. Their fork has also diverged quite a lot from mine, so it's best to see the two projects as being entirely different.

Maybe you can check their discord, I'm sure they'd be willing to guide you: https://github.com/sylverb/game-and-watch-retro-go?tab=readme-ov-file#discord-support-and-discussion

Good luck!

ducalex, 2025-08-30:
I've added esp32-s3 overclock support. Right now I don't have access to an s3 device capable of running retro-go so I don't know if it will work correctly (I've tested the overclock on a basic esp32s3 module. I know the cpu speed goes up but I don't know if it will break SPI or audio or be stable). However it maxes out at 290Mhz. I don't know how useful it is.

Also fixed the debug menu.

DynaMight1124, 2025-08-30:
Hi

I've tested the latest dev build, theres a bit to unpack, I havent done hours of testing but feel I have enough for some meaningful feedback! Debug menu issue is fixed, thanks. 

SNES sound off bug isnt totally fixed, well not in the same way as MD anyway. It feels like the sound is still being emulated as the gameplay speed now stays the same whether sound is on or off, rather than before when you would see a large increase in speed when the sound was off... just too fast! :) 

Overclocking does work on my S3 device, the menu (I assume its only meant to be in the debug menu? I couldnt see it in the emulator menus like it was in the prev build?) only goes upto 272Mhz, step 2, after it repeats 272 so I didnt go any higher, maybe step 3 was 290 but just didnt show it? I didnt test that as only just noticed what you said above about 290Mhz being the max.

Overclock does bring some meaningful speed, well in GBA it does help but not enough to get anywhere near full speed but gets you somewhat better speeds, turning off sound does help a little too. Mario Kart Advanced, 272Mhz and sound off, gets about 50%, up from about 30% with stock clock and sound.

Now overclocking did massively help SNES, much more than I thought it would after testing GBA but 272Mhz WITH sound, gets it very close to fully playable, theres some small sound stutters but feels very good, I think I could probably play it as is without too many issues, first time I could probably say for SNES in RetroGo. Heres a quick video, sound was quite quiet as I had it set too low.

https://photos.app.goo.gl/KBh6Hd9VQVyhqYTN6

I can do some more extensive testing with overclocking but it does help, if theres no massive issues, feels like maybe it could be an option in the emulator menu? Maybe one that saves (per emulator) but has a failsafe reset option incase it does cause a crash. The options in the emulator menu could be fairly safe with some more advanced ones in the debug menu, that dont save.

ducalex, 2025-08-31:
> SNES sound off bug isnt totally fixed, well not in the same way as MD anyway. It feels like the sound is still being emulated as the gameplay speed now stays the same whether sound is on or off, [...]

Wasn't that the goal? To cap emulation speed to 100%? Or are you saying it's now slower than before, as in it slows down below-100% emulation too?

> Overclocking does work on my S3 device, the menu (I assume its only meant to be in the debug menu? I couldnt see it in the emulator menus like it was in the prev build?) 

It's hidden if you built a RELEASE. I've removed the release check for now, it should always be there now. I'll decide if I want to keep the overclock there before I do the next release.

> only goes upto 272Mhz, step 2, after it repeats 272 so I didnt go any higher, maybe step 3 was 290 but just didnt show it? I didnt test that as only just noticed what you said above about 290Mhz being the max.

>  On my side, the overclocking went up to 279/280 mhz so that might differ based on the ESP32-S3 chip revision ?

The number shown in the menu is a value that is added to `DIV_7_0` register's boot value. That default value may vary depending on your crystal speed, chip revision, sdkconfig, PLL config, etc. So the effects of the OC are hard to predict with such naive code. The speed shown in the menu is the value measured, so it's always the "real" speed. For the esp32s3 board I tested I could go 230-290, but YMMV. On esp32 side it seems more predictable in being steps of 20Mhz, giving 140-360 (but again it depends on the crystal speed on your board)

It's definitely possible to do smarter/more robust overclocking but I'd need to do a lot more testing and reverse engineering. And for even more speed, it's possible to increase the voltage/current to the cores. I will never do that in a release, but it might be fun to experiment if we can get the esp32 to 400Mhz.

> https://photos.app.goo.gl/KBh6Hd9VQVyhqYTN6

Wow the frameskip even looks reasonable, this is amazing!

DynaMight1124, 2025-08-31:
> Wasn't that the goal? To cap emulation speed to 100%? Or are you saying it's now slower than before, as in it slows down below-100% emulation too?

Sorry I probably didnt explain properly. When I tested before with sound off I was getting over 100% speed so got a fairly big bump with sound disabled, after your fix, I would get roughly the same speed with sound enabled and disabled (about 60-80% with no overclock), I was kinda hoping for a good solid 90-100% with no overclock and no sound, similar to MD.

Overclock did put a bit of a spanner in the works as I would probably choose a slight OC and sound but still not sure if your intended frame lock fix worked fully. 

> It's hidden if you built a RELEASE. I've removed the release check for now, it should always be there now. I'll decide if I want to keep the overclock there before I do the next release.

Ah right yes, it was a release build. Thats probably why. If you do decide to keep it, it doesnt need to be in the main options menu but maybe in the 'Options/Emulator Options' options menu

> Wow the frameskip even looks reasonable, this is amazing!

Yeah played really well, I didnt expect that much of a bump to be honest. the GBA bump didnt feel like much so hopes werent high.

I have played around a little more today with overclocking, since I was a sega kid, I aways sway towards the MegaDrive so naturally tested that with OC too, it does get a bump but no-where near as much, gets a similar bump to the GBA so not sure if thats related to the recent changes made to SNES between you and rapha-tech.

Also overclocking does seem to bring back some issues with running too fast when sound disabled, both MD and SNES run over 100% if sound is disabled and overclocked enabled. Maybe SNES and MD would run faster than 100% with sound if they could! 

I'm not trying to be nitpicky, please tell me if I am :) If its something you're not fussed about I'm totally happy with that :)

DynaMight1124, 2025-09-03:
All testing was done on an S3, the OG ESP32 devices I have all require slight disassembly and an external programmer so tend to only get tested/updated when needed because...hassle! 

I did see the commit about the performance hit. Bit of a shame but as long as the emulators other than SNES/MD/GBA still run at full speed then I guess its not big loss, so the S3 doesnt have this bug? I only ever noticed a minor speed improvement on OG vs S3, just a few frames extra on SNES/MD but never enough to truly make it a must upgrade however I guess this would widen the gap.

ducalex, 2025-09-06:
The S3 doesn't have this bug, neither does the og ESP32 rev3. Enabling the fix doesn't seem to affect SNES that much, surprisingly.

> the OG ESP32 devices I have all require slight disassembly and an external programmer so tend to only get tested/updated when needed because...hassle!

I'm continuing work on the updater, do not despair!

csy19900206, 2025-11-28:
In master branch , i find this config in sdkconfig file ,
CONFIG_ESPTOOLPY_FLASHMODE="dio" 
Espressif ESP32S3 WROOM 1 Can Run in QIO
Can you try it? The configuration has indeed helped me improve performance in other projects before and 
Read this comments is of any help for now ? 

```
Move Instructions in Flash to PSRAM[](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/external-ram.html#move-instructions-in-flash-to-psram)
The [CONFIG_SPIRAM_FETCH_INSTRUCTIONS](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/kconfig-reference.html#config-spiram-fetch-instructions) option allows the flash sections (for instructions) to be placed in PSRAM..text

By enabling the [CONFIG_SPIRAM_FETCH_INSTRUCTIONS](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/kconfig-reference.html#config-spiram-fetch-instructions) option,

Instructions from the sections of flash are moved into PSRAM on system startup..text

The corresponding virtual memory range of those instructions will also be re-mapped to PSRAM.
```
----------------------
GBA support！！！:-)