ducalex/retro-go #97: PSRAM Required? or is there a way to build an image for running on a ESP32-wroom-32?

Akrobate, 2024-01-10:
Hello,

I am facing the following problem:
[image]

My esp32 is an esp32-wroom-32 and don't have external ram. Is there a way to build an image that would not require the usage of external ram?

Thank you

ducalex, 2024-01-11:
In `base.sdkconfig` you can delete the entire `SPI RAM config` section, esp-idf should then default to a no PSRAM build. Make sure to run `rg_tool.py clean` after that.

The launcher and some emulators (using small roms) used to work without external RAM but I haven't tested in a very long time, it's quite likely you'll end up with a blue screen due to memory allocation failure...

If that happens you'll have to tweak the code to disable features and save memory. Or you can try to build older retro-go versions!

ducalex, 2024-01-17:
I did some testing to see where the panics occur and I don't think they'll be super easy to fix... I managed to get the launcher to run by doing the attached changes, but no emulator.

````diff
diff --git a/components/retro-go/targets/esplay-s3/sdkconfig b/components/retro-go/targets/esplay-s3/sdkconfig
index a99e607f..d507d18e 100644
--- a/components/retro-go/targets/esplay-s3/sdkconfig
+++ b/components/retro-go/targets/esplay-s3/sdkconfig
@@ -70,7 +70,7 @@ CONFIG_NEWLIB_NANO_FORMAT=y
 #
 # SPI RAM config
 #
-CONFIG_ESP32_SPIRAM_SUPPORT=y
+CONFIG_ESP32_SPIRAM_SUPPORT=n
 CONFIG_SPIRAM_BOOT_INIT=y
 CONFIG_SPIRAM_IGNORE_NOTFOUND=y
 CONFIG_SPIRAM_USE_MEMMAP=n
diff --git a/launcher/main/applications.c b/launcher/main/applications.c
index a4e0bbd6..73030af1 100644
--- a/launcher/main/applications.c
+++ b/launcher/main/applications.c
@@ -142,7 +142,7 @@ static void application_start(retro_file_t *file, int load_state)
 
 static void crc_cache_init(void)
 {
-    crc_cache = calloc(1, sizeof(*crc_cache));
+    crc_cache = NULL;
     if (!crc_cache)
     {
         RG_LOGE("Failed to allocate crc_cache!\n");
diff --git a/launcher/main/gui.c b/launcher/main/gui.c
index 0e8cad5f..8dc8c4ed 100644
--- a/launcher/main/gui.c
+++ b/launcher/main/gui.c
@@ -49,7 +49,7 @@ void gui_init(void)
     gui.browse = gui.start_screen == START_SCREEN_BROWSER ||
                  (gui.start_screen == START_SCREEN_AUTO && rg_system_get_app()->bootType == RG_RST_RESTART);
     gui_update_theme();
-    rg_gui_set_buffered(true);
+    // rg_gui_set_buffered(true);
 }
 
 void gui_event(gui_event_t event, tab_t *tab)
@@ -120,6 +120,8 @@ const rg_image_t *gui_get_image(const char *type, const char *subtype)
 {
     char name[64];
 
+    return NULL;
+
     if (subtype && *subtype)
         sprintf(name, "%s_%s.png", type, subtype);
     else
````

If you want an old version, it seems 1.28 would be a good candidate, it was just before the launcher redesign with images.
Unfortunately it won't build for esp-idf 5.0 without changes, so it might not be any easier for you in the end.

ducalex, 2024-01-17:
My bad I confused two issues, you don't need esp-idf 5.0. You can go back as far as you want in retro-go's history then! But you might have to use the obsolete esp-idf 3.3 because at one point it didn't build on 4.x :).

ducalex, 2024-02-29:
I'm closing this issue, the short answer is that yes PSRAM is required.

I've also added an alert in 6baf173afd5a5b63d632c99bd105c644a943f446 that pops up if external ram isn't detected.

Eventually it might be interesting to see if implementing a low memory mode would be feasible :) . For example run the launcher in text mode and run emulators in single buffer mode.

closed 2024-02-29 by ducalex