ducalex/retro-go #102: Blinking screen in launcher

Akrobate, 2024-03-12:
Hello!

I worked on the last remaining problem I encounter:

Blinking screen in launcher. I've found what causes this problem. the blink occurs while screen refreshes. (I do not encounter this problem in emulators, everything works properly event in full screen update mode)

So the problem is here:

[image]

To fix it and get the same behaviour as before I just commented this line and added `gui_redraw();` at the end of `gui_load_preview` function in gui.c file.

[image]

Firstly I thought it would be dirty (using redraw in gui.c) , but I've seen you already done it in the same file
[image]

What do you think about it?

Also I don't really see why do we need to refresh every 100ms while no changes are done...

Akrobate, 2024-03-14:
> In 1.40 I updated the launcher to use the same display update function as the games, have you checked if it resolved your issue?

Yesterday I resynchronized my fork with your dev, and rebased all my branches, everything works fine but the problem with screen is still here.

> we can't remove the idle timer redraw

Yes I see. Would it be reasonable to keep updating the idle each 100ms with gui_draw in load_preview, but adding an other timer to redraw idle every much rarely: something like 5secs or even 10secs ?

> Would reducing the frequency help? Maybe 200ms? 500ms?
So I tried it, and it blinks at 1/200ms freq.

What I'm investigating is that it seems beeing correlated with data transert. So I have two pathes I'll explore:
* My problem could be because of some powersupply voltage falls. So I'll test to power my device with an LiPo battery to eliminate this hypothesis. I've putted a bunch of capacitors to prevent voltage variations, but it doesn't have any effect.
* Simply trying an other screen. It could be a problem with this cheap device. I've bought it on amazon for something like 15€ it was one of the cheapest ones. Today I received an other SPI model with the same driver. I just need to read a little about backlight management. The current one have a BLK pin that I pluggued directly on the esp32S3, but this one have a LED pin, so I suppose I'll need to put a transistor for beeing able to plug it on BLK pin and having the light management from retro-go ui.

Otherwise I'm working on the hardware casing for retro-go if you have any suggestions or ideas let me know =)
https://github.com/Akrobate/retro-go-hardware

ducalex, 2024-03-15:
> Yes I see. Would it be reasonable to keep updating the idle each 100ms with gui_draw in load_preview, but adding an other timer to redraw idle much rarely: something like 5secs or even 10secs ?

Agreed, we can definitely be smarter about updates!

> So I tried it, and it blinks at 1/200ms freq.

You know, now that I think about it, that's a bit weird... Because in the launcher, the content changes only maybe once per minute (clock) when it's idle, right? And since 1.40 partial updates apply to the launcher as well. 

In other words, there should be 0 SPI transfers when the launcher is idle. So that can't be the cause of your flicker...

To confirm this, I've pushed a commit that tracks all display and sdcard transfers if you want to confirm on your side: https://github.com/ducalex/retro-go/commit/c742caeb9a03ecad770f982aabfceaa77ca8a3b5 (it's not in the main/dev branches yet, not sure if useful to keep).

I think it might still be a power issue, but I'm not sure why it coincides with updates if no transfers occur. Maybe because both cores do some work for a few milliseconds? Would it really affect power draw that much?

> Otherwise I'm working on the hardware casing for retro-go if you have any suggestions or ideas let me know =)
https://github.com/Akrobate/retro-go-hardware

I like that you're using openscad :)

ducalex, 2024-03-15:
I linked the wrong commit (missing sdcard tracking), this is the correct patch: https://github.com/ducalex/retro-go/commit/e31645af3200b6c1e091fa7e6216a73945a074c5

Akrobate, 2024-03-15:
Hey! Thank you for your answer. I think I get confused yesterday while testing. With the last last update of develop there are no screen blinking any more.

And you are perfectly right, there no more data transfering when in idle:

[image]

So I have removed my "fix" for screen blinking.

Now I have a little screen blinks (much less visible) in games when a lot of elements changes. So I'll confirm it with external power supply and then with another screen.

It could totaly be a power supply issue, because I have noticed that when I put the volume to 100%, each time a note is playing, screen blinks. on 5% Volume no blinking.

I'll keep you informed of my progresses =)

Tem4a, 2024-03-15:
Hello. I'll join the conversation. 

I had the same problem with blinking and beeper in the launcher (and emulator snes) as in issue 50 - here https://github.com/ducalex/retro-go/issues/50#issuecomment-1304421322 .
In version 1.40 this problem is no longer there (but snes still runs with sound glitches ). 
By the way, do you have snes running normally? without sound glitches? 

> Now I have a little screen blinks (much less visible) in games when a lot of elements changes. So I'll confirm it with external power supply and then with another screen.
>
> It could totaly be a power [...]

I had the same problem with the screen blinking in time with the music. It's a hardware problem with the sound module.
What sound module are you using? 
Probably not enough resistor rating.

Akrobate, 2024-03-15:
Hello @Tem4a ! Welcome =)

> By the way, do you have snes running normally? without sound glitches?

Actually I haven't tested the snes emulator at all as it mentionned as slow running. So I supposed it's not really working emulation for now. Is it slow? or is it playable? 

> I had the same problem with the screen blinking in time with the music. It's a hardware problem with the sound module. What sound module are you using?

I'm using this one:
[image]

I'm not sure it is module problem. I think it just consuming to much current when working loud.

Also I do not observe any sounds problems in launcher. Spreaker remain quiet in launcher.

Tem4a, 2024-03-15:
> Actually I haven't tested the snes emulator at all as it mentionned as slow running. So I supposed it's not really working emulation for now. Is it slow? or is it playable?

I also have a homemade console (in progress :) ) and snes is not very playable due to slowness, sound glitches, and non-functional controls in some games. 
Although in version 1.28 the controls worked better than in subsequent versions. @ducalex , have you paid attention to this? 

Here is a video, you can see how the screen flashes in time with the music :

https://github.com/ducalex/retro-go/assets/5213785/1b48f8d7-5c2c-4e5f-914c-7915f37844dc

Do you have something similar? 

I use a PAM8302 module .  schematic -> https://cdn-learn.adafruit.com/assets/assets/000/112/257/original/components_schem.png?1654276084
I solved the problem by changing resistors R1 and R2 to 100k as indicated in the official odroid schematic ( https://github.com/hardkernel/ODROID-GO/blob/master/Documents/ODROID-GO_REV0.1_20180518.pdf ) but I connect as a speaker to pins 25 and 26. 

With your module I understand you are using EXT-Pin ( from https://wiki.odroid.com/odroid_go/odroid_go#odroid-go_gpio_pin_mapping ) and unfortunately I don't know how to help with that :( 
Do you have possibilities to connect a regular audio module?

Akrobate, 2024-03-16:
> I use a PAM8302 module 
As I understand its an amplifier. That means you are using internal dac of your ESP Device. That's why putting 100k resistors on audio "lines" helps. You are limiting the current that way and I suppose this way the amplifier is less saturated.

On my side my module is an external dac. ESP32S3 do not have any internal dac, so for me it's only the way to get sound. I'm not worring about this part cause I'm pretty much sure it's a power supplie issues. It could be some little voltage falls, or simply some noise because of the speaker.

I have to test my device with a 3A DC-DC module. I'll tell you results whenever it's done =)

Tem4a, 2024-03-16:
> ESP32S3 do not have any internal dac

ahh, yes, i see. 

> I'll tell you results whenever it's done =)

okay, I hope you can solve this problem :)

ducalex, 2024-03-16:
> Although in version 1.28 the controls worked better than in subsequent versions. @ducalex , have you paid attention to this?

I don't remember what (if anything) changed about controls, that was quite long ago. As for audio I definitely did something at some point because in many games the sound goes crazy after a few minutes, hinting at internal memory/state corruption of some kind.

Anyway feel free to open a separate issue if you want to discuss improving SNES controls (or any aspects of it really)!

Tem4a, 2024-03-16:
> I don't remember what (if anything) changed about controls .... Anyway feel free to open a separate issue if you want to discuss improving SNES controls

ok, I'll take the time to test the versions between 1.28 and 1.29 and find the commit after which I started having control problems. 

> I suspect SNES might run better on all your custom devices where the ESP32-S3 can run PSRAM at 120Mhz!

I don't know much about hardware, maybe you can help. 
I'm using this esp module : https://www.lilygo.cc/products/t7-mini32-v1-5?variant=42302663065781 , the description says only PSRAM 8MB, but there is no frequency (although in Chipset it says ESP32 240MHz Xtensa). 

1. how can I find out what is the frequency of PSRAM ? or 240 is it ? 
2. do I need to make any changes in sdkconfig and compile the build myself to use the full power of the module? or the build from the release adapts itself to the platform and uses everything it can?

Akrobate, 2024-03-17:
@Tem4a 

> Here is a video, you can see how the screen flashes in time with the music

I see on your video you uses a screen I bought also for testing. I suppose this one do have a LED pin for backlight. How did you pluggued it to your esp? With a transistor like a 2N2222?

Tem4a, 2024-03-17:
@Akrobate 

> LED pin for backlight

you mean LED pin, right ?
[image]

this pin is connected to GPIO 14 according to official mapping 
https://wiki.odroid.com/odroid_go/odroid_go#odroid-go_gpio_pin_mapping

here is the module wiki 
http://www.lcdwiki.com/3.2inch_SPI_Module_ILI9341_SKU:MSP3218
and according to the schematic ( http://www.lcdwiki.com/res/MSP3218/MSP3218-3.2-SPI.pdf ) there is already a transistor on this pin. 

[image]

BUT!! if you use 3.3v supply - you need closed the J1 jumper.

Akrobate, 2024-03-17:
@Tem4a Thank you for the wiki: http://www.lcdwiki.com really usefull!!! On your device do you use 5V or 3.3 to power the screen?

Tem4a, 2024-03-17:
@Akrobate I use 3.3v directly from 3v3 pin of ESP. 
and closed the J1 jumper

ducalex, 2024-03-18:
> 1. how can I find out what is the frequency of PSRAM ? or 240 is it ?
> 
>     2. do I need to make any changes in sdkconfig and compile the build myself to use the full power of the module?

Sorry I wasn't clear, only the ESP32-S3 can use 120MHz RAM (and octal RAM, together giving quite a boost vs the 80Mhz quad of the ESP32), your module is the same as in the ODROID-GO, "just" a regular ESP32.

As for `sdkconfig.defaults` changes, they're [well documented by espressif](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-guides/flash_psram_config.html) but what I usually do as a first step is:

Build the launcher with rg_tool.py. Enter the launcher directory, make a  copy of `sdkconfig`. Then run `idf menuconfig` and change the options I need. Then do a diff between the new `sdkconfig` and the one I just backed up. That diff can then be placed in the target's `sdkconfig.defaults` so that it's applied to all apps and all future builds.

Tem4a, 2024-03-22:
@ducalex 

> SNES .... the emulator is mostly a useless tech demo.

is it possible to complete the snes emulator to a full working state? 
or is the problem not in the complexity of the code, but in insufficiently powerful hardware?

ducalex, 2024-03-22:
I still believe full speed SNES emulation on the ESP32 is possible. But I don't know if it will be with snes9x or by me. I spent tons of hours trying so many tweaks and rewrites but I never could achieve full speed with a reasonable frameskip. 

Overclocking might help, though.

ducalex, 2024-08-22:
I think the original issue has been resolved so I will close it. Feel free to re-open if I'm wrong, or open another issue to discuss any other tangential topics!

closed 2024-08-22 by ducalex