ducalex/retro-go #103: Why update screen mode (FULL vs Partial) has been removed?

Akrobate, 2024-03-13:
Hello,

Today I've tested the last dev version. Everything is working fine.

I noticed that the update of screen mode has been removed from the menu. Is it intentionnal? Why this choice? I suppose partial is used by default?

thank you =)

ducalex, 2024-03-14:
Hey! I wrote extensive messages for those changes in the following commits:

- Commit where I removed: https://github.com/ducalex/retro-go/commit/4396bb73d6e955b1df1c9f910937d729bd9b42ff
- Commit where I rewrote the partial update system: https://github.com/ducalex/retro-go/commit/a44530c98fff03e5d391c7f52a54af842ef51629

Let me know if you have questions!

Akrobate, 2024-03-14:
Hi! Thank you for your answer. I haven't yet read the code, but your comments are clear. I understand the reasons of the change.

And keeping the option adding your new screen update as a possible choice doesn't make sense? Someting like "Partial v2"

ducalex, 2024-03-15:
> And keeping the option adding your new screen update as a possible choice doesn't make sense? Someting like "Partial v2"

I don't think it's worth having an option because the new code "just works". The hashing adds some overhead (on core 1, which is underutilised) but I've done extensive measuring and I haven't found any instance where partial code was slower than always doing full frame. I've also tested all combinations of scaling and filtering and the image is identical with or without partial enabled. 

There is a measurement in the debug menu called "Blit time". For comparison a full update takes ~36ms to send to the screen. So, as long as that number is less than 36ms, it means that partial frame is helping or at least not hindering.

So at this point an option wouldn't benefit the user and it would waste my time because on every change I make I'd have to test partial+filtering1, full+filtering1, partial+filtering2, full+filtering2, partial, full, partial+scaling+filtering, full+scaling+filtering, etc.

All that being said! if you truly don't want partial, I can add a `#define` in the target file to disable it at compile time :)

Akrobate, 2024-03-15:
No no you're right it would be a waste of time. I close this issue, it was just by curiosity. Your new implementation of screen update works fine