ducalex/retro-go #109: Prboom(esp32-s3): External DAC crashes when system monitor task is enabled.

Sunyoup, 2024-05-04:
## Describe the bug ##
To use external DAC(like PCM5102a module), the **rg_sysmon** task should be disabled.

## To Reproduce ##
Steps to reproduce the behavior:
1. Target: esp32s3-devkit-c
2. Version: 1.41-2
3. Construct test circuit using external DAC like PCM5102a module.
4. No need to change the wiring in config.h (BCK:41, WS:42, DATA:40).
5. Prepare SD card with WAD file.
6. Build, flash & start.
7. See the serial debug output.

## Expected behavior ##
**"doom_sound" task create fails.** --> System crash blue screen appears on the LCD.

## Additional context ##
In **components/retro-go/rg_system.c** file, disabling **"rg_sysmon"** task solves the problem, giving clear sound from DAC:
..... (rg_system.c) .....
**//rg_task_create("rg_sysmon", &system_monitor_task**, NULL, 3 * 1024, RG_TASK_PRIORITY_5, -1); <-- Block this line using **"//"**.
.....

ducalex, 2024-05-04:
The rg_sysmon task is fairly benign, I don't think the real fix would be to disable it. But thanks for testing, it gives me clues as to what's going on!

Can you share a log output of when it crashes?

I'd like to know why the doom_sound task creation fails. Is it out of slots? Shouldn't be... Or maybe it's out of internal RAM?

Sunyoup, 2024-05-05:
It should be noted that I'm using ESP-IDF v4.4.1.
I've tried ESP-IDF v5.2.1(most recent at the moment) but
not working at all, with many warnings during build.

Now I'll move on to some soldering work.
I'm thinking about attaching PAM8403 amplifier module next to the PCM5102a module.

(To ducalex:) Thank you for sharing your great work of developing this multiple game platform,
which is exactly what I've searched for !

Here I paste the log output for failure case:
===============================================================
[0m[34m[debug] STACK:4724, HEAP:228+7655 (220+7424), BUSY:0%, FPS:0 (0+0+0), BATT:3302
[0m[34m[debug] STACK:4724, HEAP:228+7655 (220+7424), BUSY:9%, FPS:108 (98+9+0), BATT:3302
[0m[34m[debug] STACK:4724, HEAP:228+7655 (220+7424), BUSY:8%, FPS:110 (101+9+0), BATT:2498
[0m[34m[debug] STACK:4724, HEAP:228+7655 (220+7424), BUSY:8%, FPS:109 (100+10+0), BATT:2498
[0m[info] application_init: Initializing application 'DOOM' (prboom-go)
[0m[34m[debug] STACK:4724, HEAP:228+7505 (220+7424), BUSY:8%, FPS:109 (99+9+1), BATT:3120
[0m[34m[debug] STACK:4724, HEAP:228+7455 (220+7296), BUSY:26%, FPS:32886 (32878+8+0), BATT:3120
[0m[34m[debug] STACK:4724, HEAP:228+7455 (220+7296), BUSY:19%, FPS:91 (82+9+0), BATT:2458
[0m[34m[debug] STACK:4724, HEAP:228+7453 (220+7296), BUSY:4%, FPS:43 (41+2+0), BATT:2458
[0m[34m[debug] STACK:4724, HEAP:228+7453 (220+7296), BUSY:0%, FPS:50 (50+0+0), BATT:2488
[0m[info] update_viewport_scaling: 320x240@1.333 => 320x240@1.333 left:0 top:0 step_x:1.00 step_y:1.00
[0m[info] crc_cache_save: Saving cache
[0m[info] rg_system_switch_app: Switching to app prboom-go (doom)
[0m[33m[warn] rg_system_restart: Restarting system!
[0m[info] rg_audio_deinit: Audio terminated. sink='Ext DAC'
[0m[info] rg_system_save_time: System time saved to storage.
[0m[info] rg_storage_deinit: Storage unmounted.
[0m[info] rg_input_deinit: Input terminated.
[0m[info] rg_display_deinit: Display terminated.
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375594
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x14a4
load:0x403b6000,len:0xe08
load:0x403ba000,len:0x304c
entry 0x403b6264
[0;32mI (29) boot: ESP-IDF v4.4.1 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 02:59:43[0m
[0;32mI (29) boot: chip revision: 0[0m
[0;32mI (31) qio_mode: Enabling QIO for flash chip WinBond[0m
[0;32mI (36) boot.esp32s3: Boot SPI Speed : 80MHz[0m
[0;32mI (41) boot.esp32s3: SPI Mode       : QIO[0m
[0;32mI (46) boot.esp32s3: SPI Flash Size : 16MB[0m
[0;32mI (51) boot: Enabling RNG early entropy source...[0m
[0;32mI (56) boot: Partition Table:[0m
[0;32mI (60) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (67) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (82) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (89) boot:  3 launcher         OTA app          00 10 00010000 000e0000[0m
[0;32mI (97) boot:  4 prboom-go        OTA app          00 11 000f0000 000c0000[0m
[0;32mI (104) boot: End of partition table[0m
[0;32mI (109) esp_image: segment 0: paddr=000f0020 vaddr=3c070020 size=33cfch (212220) map[0m
[0;32mI (149) esp_image: segment 1: paddr=00123d24 vaddr=3fc921c0 size=0a85ch ( 43100) load[0m
[0;32mI (157) esp_image: segment 2: paddr=0012e588 vaddr=40374000 size=01a90h (  6800) load[0m
[0;32mI (159) esp_image: segment 3: paddr=00130020 vaddr=42000020 size=6e208h (451080) map[0m
[0;32mI (232) esp_image: segment 4: paddr=0019e230 vaddr=40375a90 size=0c72ch ( 50988) load[0m
[0;32mI (242) esp_image: segment 5: paddr=001aa964 vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (250) boot: Loaded app from partition at offset 0xf0000[0m
[0;32mI (250) boot: Disabling RNG early entropy source...[0m
[0;33mW (262) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.[0m

========================================================
prboom-go 1.41-2-g9b53f5e7-dirty (May  6 2024 02:58:55)
 built for: ESP32S3-DEVKIT-C. type: dev
[... 104 lines cut ...]

========================================================
launcher 1.41-2-g9b53f5e7-dirty (May  6 2024 02:58:07)
 built for: ESP32S3-DEVKIT-C. type: dev
========================================================

[info] rg_storage_init: Storage mounted at /sd. driver=1
[0m[info] rg_input_init: Initializing ADC1 gamepad driver...
[0m[info] rg_input_init: Initializing GPIO gamepad driver...
[0m[info] rg_input_init: Initializing ADC1 battery driver...
[0m[info] rg_input_init: Input ready. state=00000000 00000000
[0m[31m[error] rg_system_init: Recoverying from panic!
[0m[info] rg_display_init: Initialization...
[0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fca0640
[0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fca1044
[0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fca1a48
[0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fca244c
[0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fca2e50
[0m[info] lcd_set_backlight: backlight set to 80%
[0m[info] rg_display_init: Display ready.
[0m[info] rg_alloc: SIZE=11520, CAPS=SPIRAM|8BIT, PTR=0x3d80a44c
[0m[info] rg_gui_set_font: Font set to: points=12, scaling=1.00
[0m[info] rg_gui_set_theme: Using built-in theme!

===============================================================

ducalex, 2024-05-05:
So it's clearly running out of memory. I'm not quite sure why it doesn't crash when the monitor task is disabled, though. Surely the 3KB it uses wouldn't make a difference.

Have you modified the sdkconfig? Things like `CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL` are supposed to prevent this from happening. Also I don't see a stack trace in your log but since we have `CONFIG_ESP32_PANIC_PRINT_REBOOT=y` there should be one.

---
>  I've tried ESP-IDF v5.2.1(most recent at the moment) but not working at all, with many warnings during build.

Thanks I'll look into it tonight! I know I got 5.2 to work with the esp32 a while ago but I guess I never tested building for the esp32-s3.

ducalex, 2024-05-05:
I confirm that it does build with esp-idf 5.2.1 on my side when running `rg_tool.py --target=esp32s3-devkit-c build-fw`. There are lots of deprecation and format warnings but they can be ignored. I'm not sure why it fails on your side.

I have not tested the produced build on real hardware, though.

Sunyoup, 2024-05-05:
Sorry I mean that the build job is done with warnings but not running. I'll
keep using IDF v4.4.x and esp32s3 n16r8.

Sunyoup, 2024-05-05:
And I did not modify sdkconfig.

Sunyoup, 2024-05-05:
I'm making img, not fw, and flashing it using esptool.

Btw, I'm using ili9341 tft. Is it possible to use st7796 instead?

ducalex, 2024-05-06:
I see two possible solutions, then:

1. We can initialize the sound task before initializing DOOM, which is what fills the memory so much
2. We can force small allocations to be made to external memory, but it might impact performance a little

The simplest would be the # 2, you can try by adding just two lines to `prboom-go/main/main.c`:

````diff
diff --git a/prboom-go/main/main.c b/prboom-go/main/main.c
index 2c199ab1..0c7ac3e2 100644
--- a/prboom-go/main/main.c
+++ b/prboom-go/main/main.c
@@ -41,6 +41,7 @@
 #include <midifile.h>
 #include <oplplayer.h>
 #include <rg_system.h>
+#include <esp_heap_caps.h>

 // 22050 reduces perf by almost 15% but 11025 sounds awful on the G32...
 #ifdef RG_TARGET_MRGC_G32
@@ -568,6 +569,8 @@ void app_main()

     rg_display_clear(C_BLACK);

+    heap_caps_malloc_extmem_enable(1024);
+
     Z_Init();
     D_DoomMain();
 }
````

ducalex, 2024-05-06:
You can try different numbers than 1024 too, to find the balance between performance and stability. By default it is set to 32768 (it is the value of `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL` in the sdkconfig).

ducalex, 2024-05-06:
> Btw, I'm using ili9341 tft. Is it possible to use st7796 instead?

I think they are similar drivers but not identical. At the very least the init sequence and some commands will need to be changed.. PR welcome if you can make it work :)

Sunyoup, 2024-05-06:
I tried your solution and found that the number of **512** is the best until now.

I tried 1024,2048,4096,8012,16384,32768.
In case of 1024, sound task is created and the demo is running,
but after a few seconds the system crashes.
Other values result in immediate crashes.

So I'll keep watching on with the value of **512**.

But I'll spend a couple of weeks working on soldering on a perfboard.

Here is the serial output of 1024 case:
------------------------------------------

[0m[34m[debug] STACK:4724, HEAP:228+7655 (220+7424), BUSY:8%, FPS:109 (100+9+0), BATT:2606
[0m[34m[debug] STACK:4724, HEAP:228+7655 (220+7424), BUSY:8%, FPS:109 (100+9+0), BATT:2912
[0m[34m[debug] STACK:4724, HEAP:228+7655 (220+7424), BUSY:8%, FPS:109 (100+9+0), BATT:2912
[0m[34m[debug] STACK:4724, HEAP:228+7655 (220+7424), BUSY:8%, FPS:109 (100+9+0), BATT:2848
[0m[info] application_init: Initializing application 'DOOM' (prboom-go)
[0m[34m[debug] STACK:4724, HEAP:228+7455 (220+7296), BUSY:26%, FPS:21434 (21424+9+1), BATT:2848
[0m[34m[debug] STACK:4724, HEAP:228+7453 (220+7296), BUSY:2%, FPS:47 (46+1+0), BATT:2928
[0m[info] update_viewport_scaling: 320x240@1.333 => 320x240@1.333 left:0 top:0 step_x:1.00 step_y:1.00
[0m[info] crc_cache_save: Saving cache
[0m[info] rg_system_switch_app: Switching to app prboom-go (doom)
[0m[33m[warn] rg_system_restart: Restarting system!
[0m[info] rg_audio_deinit: Audio terminated. sink='Ext DAC'
[0m[info] rg_system_save_time: System time saved to storage.
[0m[info] rg_storage_deinit: Storage unmounted.
[0m[info] rg_input_deinit: Input terminated.
[0m[info] rg_display_deinit: Display terminated.
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375594
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x14a4
load:0x403b6000,len:0xe08
load:0x403ba000,len:0x304c
entry 0x403b6264
[0;32mI (29) boot: ESP-IDF v4.4.1 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 02:59:43[0m
[0;32mI (29) boot: chip revision: 0[0m
[0;32mI (31) qio_mode: Enabling QIO for flash chip WinBond[0m
[0;32mI (36) boot.esp32s3: Boot SPI Speed : 80MHz[0m
[0;32mI (41) boot.esp32s3: SPI Mode       : QIO[0m
[0;32mI (46) boot.esp32s3: SPI Flash Size : 16MB[0m
[0;32mI (51) boot: Enabling RNG early entropy source...[0m
[0;32mI (56) boot: Partition Table:[0m
[0;32mI (60) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (67) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (74) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (82) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (89) boot:  3 launcher         OTA app          00 10 00010000 000e0000[0m
[0;32mI (97) boot:  4 prboom-go        OTA app          00 11 000f0000 000c0000[0m
[0;32mI (104) boot: End of partition table[0m
[0;32mI (109) esp_image: segment 0: paddr=000f0020 vaddr=3c070020 size=33cfch (212220) map[0m
[0;32mI (149) esp_image: segment 1: paddr=00123d24 vaddr=3fc921c0 size=0a85ch ( 43100) load[0m
[0;32mI (157) esp_image: segment 2: paddr=0012e588 vaddr=40374000 size=01a90h (  6800) load[0m
[0;32mI (159) esp_image: segment 3: paddr=00130020 vaddr=42000020 size=6e210h (451088) map[0m
[0;32mI (232) esp_image: segment 4: paddr=0019e238 vaddr=40375a90 size=0c72ch ( 50988) load[0m
[0;32mI (242) esp_image: segment 5: paddr=001aa96c vaddr=50000000 size=00010h (    16) load[0m
[0;32mI (250) boot: Loaded app from partition at offset 0xf0000[0m
[0;32mI (250) boot: Disabling RNG early entropy source...[0m
[0;33mW (262) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.[0m

========================================================
prboom-go 1.41-2-g9b53f5e7-dirty (May  6 2024 02:58:55)
 built for: ESP32S3-DEVKIT-C. type: dev
========================================================
[... 126 lines cut ...]
[0;32mI (282) boot: Disabling RNG early entropy source...[0m
[0;33mW (295) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.[0m

========================================================
launcher 1.41-2-g9b53f5e7-dirty (May  6 2024 02:58:07)
 built for: ESP32S3-DEVKIT-C. type: dev
========================================================

[info] rg_storage_init: Storage mounted at /sd. driver=1
[0m[info] rg_input_init: Initializing ADC1 gamepad driver...
[0m[info] rg_input_init: Initializing GPIO gamepad driver...
[0m[info] rg_input_init: Initializing ADC1 battery driver...
[0m[info] rg_input_init: Input ready. state=00000000 00000000
[0m[31m[error] rg_system_init: Recoverying from panic!
[0m[info] rg_display_init: Initialization...
[0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fca0640
[0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fca1044
[0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fca1a48
[0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fca244c
[0m[info] rg_alloc: SIZE=2560, CAPS=DMA|8BIT, PTR=0x3fca2e50
[0m[info] lcd_set_backlight: backlight set to 80%
[0m[info] rg_display_init: Display ready.
[0m[info] rg_alloc: SIZE=11520, CAPS=SPIRAM|8BIT, PTR=0x3d80a44c
[0m[info] rg_gui_set_font: Font set to: points=12, scaling=1.00
[0m[info] rg_gui_set_theme: Using built-in theme!

ducalex, 2024-05-06:
Thanks for testing! When I'm back to the project I'll do some profiling to see if anything would benefit to remain in internal memory, and force the rest to prefer external.

It should resolve the issue nicely and not affect much, since almost all of DOOM's memory is already external.

Good luck with all your soldering!

ducalex, 2024-05-11:
I've made the change, DOOM now prefers external RAM over internal (note: it's on the dev branch). I'm also in contact with Espressif to improve that DMA situation so that SD Card keep working even in low mem situation.

I'm closing this issue because it should be resolved. But feel free to reopen or open a new one if your problem isn't entirely fixed!

closed 2024-05-11 by ducalex

Sunyoup, 2024-05-11:
@ducalex: Thanks a lot and good luck !