ducalex/retro-go #111: panic with some roms

dilixi, 2024-05-18:
## Describe the bug ##
I compile the firmware with esp32s3 and it can work well with some roms.In some case ,i just get the system panic message from the lcd. 

## To Reproduce ##
I placed this rom at sdcard,just open it,the panic will display soon.
[nes_roms.zip](https://github.com/ducalex/retro-go/files/15358923/nes_roms.zip)

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403795b2
[0;33m0x403795b2: xt_highint4 at D:\Espressif\frameworks\esp-idf-v5.1.2\components\esp_system\port\soc\esp32s3\highint_hdl.S:77[0m
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fce3818,len:0x1534
load:0x403c9700,len:0x4
load:0x403c9704,len:0xd88
load:0x403cc700,len:0x2ff4
entry 0x403c9924
[0;32mI (31) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time May 16 2024 22:20:39[0m
[0;32mI (31) boot: Multicore bootloader[0m
[0;32mI (35) boot: chip revision: v0.2[0m
[0;32mI (39) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (44) boot.esp32s3: Boot SPI Speed : 80MHz[0m
[0;32mI (49) boot.esp32s3: SPI Mode       : QIO[0m
[0;32mI (54) boot.esp32s3: SPI Flash Size : 16MB[0m
[0;32mI (58) boot: Enabling RNG early entropy source...[0m
[0;32mI (64) boot: Partition Table:[0m
[0;32mI (67) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (75) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (82) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (90) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (97) boot:  3 launcher         OTA app          00 10 00010000 00100000[0m
[0;32mI (104) boot:  4 retro-core       OTA app          00 11 00110000 000c0000[0m
[0;32mI (112) boot: End of partition table[0m
[0;32mI (116) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=38e18h (232984) map[0m
[0;32mI (149) esp_image: segment 1: paddr=00048e40 vaddr=3fc9d600 size=0481ch ( 18460) load[0m
[0;32mI (151) esp_image: segment 2: paddr=0004d664 vaddr=40378000 size=029b4h ( 10676) load[0m
[0;32mI (156) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=a8484h (689284) map[0m
[0;32mI (234) esp_image: segment 4: paddr=000f84ac vaddr=4037a9b4 size=12bb8h ( 76728) load[0m
[0;32mI (254) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (254) boot: Disabling RNG early entropy source...[0m
[0;32mI (265) cpu_start: Multicore app[0m
[0;32mI (266) flash HPM: Enabling flash high speed mode by dummy[0m
[0;32mI (266) MSPI Timing: Flash timing tuning index: 2[0m
[0;32mI (270) octal_psram: vendor id    : 0x0d (AP)[0m
[0;32mI (275) octal_psram: dev id       : 0x02 (generation 3)[0m
[0;32mI (281) octal_psram: density      : 0x03 (64 Mbit)[0m
[0;32mI (286) octal_psram: good-die     : 0x01 (Pass)[0m
[0;32mI (291) octal_psram: Latency      : 0x01 (Fixed)[0m
[0;32mI (297) octal_psram: VCC          : 0x01 (3V)[0m
[0;32mI (302) octal_psram: SRF          : 0x01 (Fast Refresh)[0m
[0;32mI (307) octal_psram: BurstType    : 0x01 (Hybrid Wrap)[0m
[0;32mI (313) octal_psram: BurstLen     : 0x01 (32 Byte)[0m
[0;32mI (319) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)[0m
[0;32mI (325) octal_psram: DriveStrength: 0x00 (1/1)[0m
[0;32mI (338) MSPI Timing: PSRAM timing tuning index: 2[0m
[0;32mI (338) esp_psram: Found 8MB PSRAM device[0m
[0;32mI (340) esp_psram: Speed: 120MHz[0m
[0;32mI (344) cpu_start: Pro cpu up.[0m
[0;32mI (348) cpu_start: Starting app cpu, entry point is 0x4037913c[0m
[0;33m0x4037913c: call_start_cpu1 at D:\Espressif\frameworks\esp-idf-v5.1.2\components\esp_system\port\cpu_start.c:184 (discriminator 3)[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (577) esp_psram: SPI SRAM memory test OK[0m
[0;32mI (587) cpu_start: Pro cpu start user code[0m
[0;32mI (587) cpu_start: cpu freq: 240000000 Hz[0m
[0;32mI (587) cpu_start: Application information:[0m
[0;32mI (590) cpu_start: Project name:     launcher[0m
[0;32mI (595) cpu_start: App version:      1[0m
[... 394 lines cut ...]
[0;33m0x4201bdff: ppu_renderoam at E:\retro-core\components\nofrendo\nes\ppu.c:543
  inlined by 0x0: ppu_renderline at E:\retro-core\components\nofrendo\nes\ppu.c:685[0m
A2      : 0x00000001  A3      : 0x00000008  A4      : 0x3fca7454  A5      : 0x00000008
A6      : 0x00000008  A7      : 0x3fca7354  A8      : 0x3fca7454  A9      : 0x00000008
A10     : 0x00000000  A11     : 0x00000100  A12     : 0x000000e0  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x000000c0  SAR     : 0x00000007  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000008  LBEG    : 0x42007e10  LEND    : 0x42007e2d  LCOUNT  : 0x00000000
[0;33m0x42007e10: build_palette at E:\retro-core\main\main_nes.c:65 (discriminator 3)[0m
[0;33m0x42007e2d: build_palette at E:\retro-core\main\main_nes.c:69[0m

Backtrace: 0x4201bdfc:0x3fcaf0b0 0x4201b2f3:0x3fcaf120 0x420080ee:0x3fcaf140 0x42007350:0x3fcaf280 0x4207705c:0x3fcaf2a0
[0;33m0x4201bdfc: ppu_renderoam at E:\retro-core\components\nofrendo\nes\ppu.c:543
  inlined by 0x0: ppu_renderline at E:\chenqiang\my_best_things\esp32s3\esplay\output\retro-go_test\retro-core\components\nofrendo\nes\ppu.c:685[0m
[0;33m0x4201b2f3: nes_emulate at E:\retro-core\components\nofrendo\nes\nes.c:43[0m
[0;33m0x420080ee: nes_main at E:\retro-core\main\main_nes.c:236[0m
[0;33m0x42007350: app_main at E:\retro-core\main\main.c:16[0m
[0;33m0x4207705c: main_task at D:\Espressif\frameworks\esp-idf-v5.1.2\components\freertos\app_startup.c:208 (discriminator 13)[0m

ELF file SHA256: 9f329de26983a570

Rebooting...

[image]

what should i do with this panic?

ducalex, 2024-05-18:
I've just pushed a fix to the dev branch for the crash (thanks for the log!).

Unfortunately `Snow Brothers (USA)` seems to be broken for unrelated reasons (I might look into that later).

dilixi, 2024-05-19:
> I've just pushed a fix to the dev branch for the crash (thanks for the log!).
> 
> Unfortunately `Snow Brothers (USA)` seems to be broken for unrelated reasons (I might look into that later).

You are so amazing, Chip 'n Dale Rescue Rangers 2 (USA) can work well now,
it's doesn't a matter with Snow Brothers (USA),because i find a new rom can work well.
[image]

closed 2024-05-19 by dilixi