ducalex/retro-go #125: Build workflow: Patching of IDF does not work

Jason2866, 2024-06-30:
Current CI runs show
```
#10 [4/5] RUN cd /opt/esp/idf && 	patch --ignore-whitespace -p1 -i "/app/tools/patches/panic-hook (esp-idf 4.2 and 4.3).diff" && 	patch --ignore-whitespace -p1 -i "/app/tools/patches/sdcard-fix (esp-idf 4.2 and 4.3).diff"
#10 1.483 patching file components/esp_system/panic.c
#10 1.484 patching file components/driver/sdspi_host.c
#10 1.484 patch unexpectedly ends in middle of line
#10 1.484 Hunk #1 succeeded at 458 with fuzz 1.
#10 DONE 1.5s
```
probably because IDF code parts have changed.

ducalex, 2024-06-30:
Are you sure it does not work? Your log says it succeeded. I think the warning is because the patch file is missing a trailing newline.

Jason2866, 2024-06-30:
Thx, fixed with last commits.

closed 2024-06-30 by Jason2866