ducalex/retro-go #145: Upgraded hardware

daevor-dev, 2024-07-27:
Crossposting here, from the odroid forums.
I was wondering if something like this display https://www.aliexpress.com/item/1005004974870929.html , could be an option for an upgraded version of the original odroid-go. It has the ESP32-S3 and a couple of different lcds, namely the TN version with 480x270 and an IPS version with 840x480. If retro-go worked with something like this and the performance was improved then adding the audio, buttons and charging circuit shouldn't be to hard to do. There's even larger versions of these with 5'' and 7'' display, but I'm unsure about the performance.

ducalex, 2024-07-28:
Anecdotally, the performance of the ESP32-S3 is quite similar to the ESP32 when it comes to retro-go. Some emulators (and DOOM) get a decent boost when running the PSRAM at 80Mhz, though.

The main issue with higher resolution screens is refresh rate. I don't know how this particular screen is connected to the ESP32-S3, but if it's SPI and it can't tolerate 80Mhz then you'll end up with pretty low FPS.

netbusiness, 2024-08-12:
Have you seen the ESP32-P4? It has more performance all around, a MIPI-DSI module, and some kind of "2D graphics accelerator"