ducalex/retro-go #146: ESP32 S3 can't build image

Samighz, 2024-07-28:
Hi 
I'm trying to build retro-go for ESP32 S3 N16R8 ( 16Mo of flash, 8Mo of PSRAM) but without success with idf 4.4, 5.3 and now 5.2.
It's my first day with ESP-Idf ! I'm a novice with it. I use Arduino platform for esp32.

I try with ( on windows 11) : "python ./rg_tool.py release launcher retro-core --target=esp32s3-devkit-c"
It give me  .fw "retro-go_1.42-6-g74079-dirty_esp32s3-devkit-c.fw" 
but i cant get the image file 

**Task failed: Command '['gen_esp32part.py', 'partitions.csv', 'partitions.bin']' returned non-zero exit status 1.**

I need advices please.

I post the last part of my output :

  ```
    |     ^~~~~~~~~~~~
F:/retro-go/retro-core/components/nofrendo/mappers/map020.c:160:35: note: format string is defined here
  160 |     MESSAGE_INFO("FDS write at %04X: %02X\n", address, value);
      |                                ~~~^
      |                                   |
      |                                   unsigned int
      |                                %04lX
F:/retro-go/retro-core/components/nofrendo/mappers/map020.c: In function 'fds_sound_read':
F:/retro-go/retro-core/components/nofrendo/mappers/map020.c:247:18: warning: format '%X' expects argument of type 'unsigned int', but argument 4 has type 'uint32' {aka 'long unsigned int'} [-Wformat=]
  247 |     MESSAGE_INFO("FDS sound read at %04X\n", address);
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~
      |                                              |
      |                                              uint32 {aka long unsigned int}
F:/retro-go/retro-core/components/nofrendo/nes/utils.h:5:68: note: in definition of macro 'LOG_PRINTF'
    5 | #define LOG_PRINTF(level, x...) rg_system_log(RG_LOG_PRINTF, NULL, x)
      |                                                                    ^
F:/retro-go/retro-core/components/nofrendo/mappers/map020.c:247:5: note: in expansion of macro 'MESSAGE_INFO'
  247 |     MESSAGE_INFO("FDS sound read at %04X\n", address);
      |     ^~~~~~~~~~~~
F:/retro-go/retro-core/components/nofrendo/mappers/map020.c:247:40: note: format string is defined here
  247 |     MESSAGE_INFO("FDS sound read at %04X\n", address);
      |                                     ~~~^
      |                                        |
      |                                        unsigned int
      |                                     %04lX
F:/retro-go/retro-core/components/nofrendo/mappers/map020.c: In function 'fds_sound_write':
F:/retro-go/retro-core/components/nofrendo/mappers/map020.c:253:18: warning: format '%X' expects argument of type 'unsigned int', but argument 4 has type 'uint32' {aka 'long unsigned int'} [-Wformat=]
  253 |     MESSAGE_INFO("FDS sound write at %04X: %02X\n", address, value);
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~
      |                                                     |
      |                                                     uint32 {aka long unsigned int}
F:/retro-go/retro-core/components/nofrendo/nes/utils.h:5:68: note: in definition of macro 'LOG_PRINTF'
    5 | #define LOG_PRINTF(level, x...) rg_system_log(RG_LOG_PRINTF, NULL, x)
      |                                                                    ^
F:/retro-go/retro-core/components/nofrendo/mappers/map020.c:253:5: note: in expansion of macro 'MESSAGE_INFO'
  253 |     MESSAGE_INFO("FDS sound write at %04X: %02X\n", address, value);
      |     ^~~~~~~~~~~~
F:/retro-go/retro-core/components/nofrendo/mappers/map020.c:253:41: note: format string is defined here
  253 |     MESSAGE_INFO("FDS sound write at %04X: %02X\n", address, value);
      |                                      ~~~^
      |                                         |
      |                                         unsigned int
      |                                      %04lX
F:/retro-go/retro-core/components/nofrendo/mappers/map020.c: In function 'fds_wave_read':
F:/retro-go/retro-core/components/nofrendo/mappers/map020.c:258:18: warning: format '%X' expects argument of type 'unsigned int', but argument 4 has type 'uint32' {aka 'long unsigned int'} [-Wformat=]
  258 |     MESSAGE_INFO("FDS wave read at %04X\n", address);
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~
      |                                             |
      |                                             uint32 {aka long unsigned int}
F:/retro-go/retro-core/components/nofrendo/nes/utils.h:5:68: note: in definition of macro 'LOG_PRINTF'
    5 | #define LOG_PRINTF(level, x...) rg_system_log(RG_LOG_PRINTF, NULL, x)
      |                                                                    ^
F:/retro-go/retro-core/components/nofrendo/mappers/map020.c:258:5: note: in expansion of macro 'MESSAGE_INFO'
  258 |     MESSAGE_INFO("FDS wave read at %04X\n", address);
      |     ^~~~~~~~~~~~
[... 63 lines cut ...]

=== Step: Packing ===

Building firmware with: launcher retro-core

Running command: tools\mkfw.py retro-go_1.42-6-g74079-dirty_esp32s3-devkit-c.fw Retro-Go 1.42-6-g74079-dirty assets/icon.raw --esplay 0 0 917504 launcher launcher\build\launcher.bin 0 0 917504 retro-core retro-core\build\retro-core.bin
[0]: type=0, subtype=16, size=917504 (55% used), label=launcher
[1]: type=0, subtype=17, size=1048576 (94% used), label=retro-core
 > WARNING: Partition smaller than file (+74272 bytes), increasing size to 1048576

Total size: 1.875 MB

File 'retro-go_1.42-6-g74079-dirty_esp32s3-devkit-c.fw' successfully created.

=== Step: Packing ===

Building image with: launcher retro-core

Generating partition table...
Running command: gen_esp32part.py partitions.csv partitions.bin
'gen_esp32part.py' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

Task failed: Command '['gen_esp32part.py', 'partitions.csv', 'partitions.bin']' returned non-zero exit status 1.
```

ducalex, 2024-07-28:
What happens when you try to run `gen_esp32part.py partitions.csv partitions.bin` in the terminal? 

Does it still say command not found?

What if you do `python ${env:IDF_PATH}\components\partition_table\gen_esp32part.py` ?

Samighz, 2024-07-28:
> What if you do `python ${env:IDF_PATH}\components\partition_table\gen_esp32part.py` ?

```
usage: gen_esp32part.py [-h] [--flash-size [{1MB,2MB,4MB,8MB,16MB,32MB,64MB,128MB}]] [--disable-md5sum] [--no-verify]
                        [--verify] [--quiet] [--offset OFFSET] [--secure [{v1,v2}]]
                        [--extra-partition-subtypes [EXTRA_PARTITION_SUBTYPES ...]]
                        input [output]
gen_esp32part.py: error: the following arguments are required: input
```

Samighz, 2024-07-28:
> What happens when you try to run `gen_esp32part.py partitions.csv partitions.bin` in the terminal?
> 
> Does it still say command not found?

```
usage: gen_esp32part.py [-h] [--flash-size [{1MB,2MB,4MB,8MB,16MB,32MB,64MB,128MB}]] [--disable-md5sum] [--no-verify]
                        [--verify] [--quiet] [--offset OFFSET] [--secure [{v1,v2}]]
                        [--extra-partition-subtypes [EXTRA_PARTITION_SUBTYPES ...]]
                        input [output]
gen_esp32part.py: error: argument input: can't open 'input_partitions.csv': [Errno 2] No such file or directory: 'input_partitions.csv'

```

Samighz, 2024-07-29:
Now it works
I added the IDF_PATH to the build_image and flash_app functions in rg_tool.py. 
On Windows, it's %IDF_PATH% ( ${env:IDF_PATH} is for Linux? ).

 run(["%IDF_PATH%\components\partition_table\gen_esp32part.py", "partitions.csv", "partitions.bin"])

Without this its impossible to run gen_esp32part.py !

Thank you all

Now i will try the flash

ducalex, 2024-07-29:
Looking at the various export.bat/export.ps1/export.sh it seems like maybe the way you start esp-idf might not add the path to the environment.

I guess there's no harm in specifying the full paths where needed in rg_tool.py.

ducalex, 2024-07-29:
Done in https://github.com/ducalex/retro-go/commit/7ceefba72a3eb0beef66f44056b7d000299ae8bf and https://github.com/ducalex/retro-go/commit/36f4016045fa2ed787f15bdd2775e7ba9ee5a3b7.

Feel free to re-open if the new rg_tool.py doesn't work for you!

closed 2024-07-29 by ducalex