ducalex/retro-go #153: Doom

D3a7Hm1N3, 2024-09-02:
How can i install Doom

ducalex, 2024-09-02:
You need to place your WAD file in `/roms/doom`.  

You can find WADs online: https://duckduckgo.com/?q=doom+wad+file&t=ffab&ia=web

You can also use freedoom: https://freedoom.github.io/download.html

D3a7Hm1N3, 2024-09-03:
Where can i put the Engine

ducalex, 2024-09-03:
The engine is built-in.

D3a7Hm1N3, 2024-09-04:
Can you recommend one or make youre port for retro go because it was the best one ive tried and could you maybe make an mp3 player in retro go so i dont have to switch every time

ducalex, 2024-09-04:
I can recommend the shareware version: https://distro.ibiblio.org/slitaz/sources/packages/d/doom1.wad

> could you maybe make an mp3 player in retro go so i dont have to switch every time

I have no plans for an mp3 player but maybe in the future!

D3a7Hm1N3, 2024-09-04:
You could ask someone who knows or has an mp3 player for odroid go and take his code and implant it in